Koenkk/zigbee-herdsman #11: the coordinator doesn't respond to "Match Description Request"

msadeghz, 2019-07-24:
Hi @Koenkk , 
I have a new zigbee device. it sends Match Description Request to the coordinator during join procedure. but the coordinator doesn't respond to it.
I use the latest version of CC2531 firmware and zigbee2mqtt. you can see the issue in below image:
[image]
the normal condition should be like below:
[image]

msadeghz, 2019-07-25:
do you mead I use zigbee-herdsman instead of zigbee-shephered? 
the Match Description Request is sent for cluster-id 6
[image]

Koenkk, 2019-07-25:
If you are on zigbee2mqtt 1.5.1 you are using zigbee-herdman (fork of zigbee-shepherd which im currently refactoring). 

If you want to solve it now you can try to do the required changes yourself, if not, I will look into it once refactor is completed.

msadeghz, 2019-07-25:
OK

msadeghz, 2019-08-04:
Hi again
[image]

the "match descriptor request" is like above. (infact the device wants to check the coordinator supports profile, input cluster and output clusters or not)
where can I handle the request? in the zigbee-shepherd? when I run `DEBUG=* npm start`, I can't see any thing when the node sends the "match descriptor request".

msadeghz, 2019-08-08:
the issue is resolved.
I create an application in the coordinator (like ciepp) which have "ssIasZone" in server and client side and then my coordinator responds the "Match Descriptor Request" finely.

closed 2019-08-08 by msadeghz