Koenkk/zigbee-herdsman #17: the Coordinator doesn't send "Transport Key" message

msadeghz, 2019-08-13:
Hi friends,
I have an standard HA light switch. when it wants to join, after association response, the coordinator doesn't transport security key, so the join procedure isn't continued. I try in multiple times, in one time it works and the device joined. but now it doesn't join again.
you can see the zigbee log below:
[3-gangs switch log.zip](https://github.com/Koenkk/zigbee2mqtt/files/3495667/3-gangs.switch.log.zip)

thincamel, 2019-08-16:
I have the same problem.
ZC CC2530 + fireware 0608
ZE CC2530
[ZC donot send transport key.pcapng.zip](https://github.com/Koenkk/zigbee-herdsman/files/3507911/ZC.donot.send.transport.key.pcapng.zip)

Koenkk, 2019-08-22:
I doubt if this problem can be solved from the zigbee-herdsman side, isn't this a firmware problem? Perhaps you could try to ask on the Texas Instruments forum?

closed 2019-10-28 by stale[bot]