Koenkk/zigbee-herdsman #22: Friends of Hue Smart Switch

mabels, 2019-09-03:
Hi,

i have these:

https://new.abb.com/low-voltage/products/residential-product/remote-control/friends-of-hue-smart-switch

and it seams not supported in zigbee2mqtt, that i will try in the next step.  

I started to figure out how it works.
First I found it is not able to pair at all. 
Than I made a PacketSniffing on zigbee, and found out that the device sends Broadcast if it is pressed and released. 
Yesterday i dived into the sourcecode to do an "Proove of Concept" how to support that kind of device. But after hours of reading i figured out that I knew not enough about the CC2531 API how to get broadcast packets at all.
It would be very helpful if somebody could give me a hint how to capture broadcast packets with the "zigbee-herdsman"?
I will attach the packet trace later today.

thx in advance

meno

Koenkk, 2019-09-03:
These are green power devices which are currently not supported by zigbee-herdsman, note that if you want to experiment with this, you need z-stack 3 (z-stack 1.2 doesn't support green power).

I've also done some investigations into green power but didn't manage to configure it correctly yet.

mabels, 2019-09-03:
Many thx for comeback so fast.
I saw your tries. I will continue after i tried to build z-stack 3 from source, and read abit in the sources to get a better understanding how this zigbee api works. For me most of the terms are new and that has to change first.

mabels, 2019-09-03:
Hi i set the pan_id to 0xffff and got this, if i press the button:

```
  serialport/bindings/poller received "readable" +22s
  serialport/binding-abstract read +22s
  serialport/bindings/unixRead Starting read +22s
  serialport/bindings/unixRead Finished read 26 bytes +1ms
  serialport/stream binding.read finished +22s
  unpi:parser <-- [254,21,85,3,0,100,146,112,1,100,146,112,1,0,0,0,0,0,2,1,95,0,0,0,13,18] +22s
  unpi:parser --- parseNext [254,21,85,3,0,100,146,112,1,100,146,112,1,0,0,0,0,0,2,1,95,0,0,0,13,18] +0ms
  unpi:parser --> parsed 21 - 2 - 21 - 3 - [0,100,146,112,1,100,146,112,1,0,0,0,0,0,2,1,95,0,0,0,13] - 18 +1ms
  cc-znp Frame {
  cc-znp   type: 'AREQ',
  cc-znp   subsystem: 21,
  cc-znp   commandID: 3,
  cc-znp   data:
  cc-znp    [ 0, 100, 146, 112, 1, 100, 146, 112, 1, 0, 0, 0, 0, 0, 2, 1, 95, 0, 0, 0, 13 ],
  cc-znp   length: 21,
  cc-znp   fcs: 18,
  cc-znp   csum: 18,
  cc-znp   sof: 254,
  cc-znp   len: 21,
  cc-znp   subsys: 'GREENPOWER',
  cc-znp   cmd: 'secReq',
  cc-znp   payload:
  cc-znp    { applicationID: 0,
  cc-znp      srcID: 24154724,
  cc-znp      gdpIeeeAddr: '0x0000000001709264',
  cc-znp      endpoint: 0,
  cc-znp      gpdfSecurityLevel: 2,
  cc-znp      gpdfSecurityFrameCounter: 1,
  cc-znp      dgpStubHandle: 95 } } +22s
  unpi:parser --- parseNext [] +2ms
  serialport/stream _read reading +3ms
  serialport/binding-abstract read +4ms
  serialport/bindings/unixRead Starting read +3ms
  cc-znp:AREQ <-- GREENPOWER:secReq, { applicationID: 0, srcID: 24154724, gdpIeeeAddr: '0x0000000001709264', endpoint: 0, gpdfSecurityLevel: 2, gpdfSecurityFrameCounter: 1, dgpStubHandle: 95 } +22s
  zigbee-shepherd:msgHdlr IND <-- GREENPOWER:secReq +22s
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +5ms
  serialport/bindings/poller received "readable" +0ms
  serialport/binding-abstract read +1ms
  serialport/bindings/unixRead Starting read +1ms
  serialport/bindings/unixRead Finished read 26 bytes +0ms
  serialport/stream binding.read finished +1ms
  unpi:parser <-- [254,21,85,3,0,100,146,112,1,100,146,112,1,0,0,0,0,0,2,1,95,0,0,0,14,17] +1ms
  unpi:parser --- parseNext [254,21,85,3,0,100,146,112,1,100,146,112,1,0,0,0,0,0,2,1,95,0,0,0,14,17] +0ms
  unpi:parser --> parsed 21 - 2 - 21 - 3 - [0,100,146,112,1,100,146,112,1,0,0,0,0,0,2,1,95,0,0,0,14] - 17 +1ms
  cc-znp Frame {
  cc-znp   type: 'AREQ',
  cc-znp   subsystem: 21,
  cc-znp   commandID: 3,
  cc-znp   data:
  cc-znp    [ 0, 100, 146, 112, 1, 100, 146, 112, 1, 0, 0, 0, 0, 0, 2, 1, 95, 0, 0, 0, 14 ],
  cc-znp   length: 21,
  cc-znp   fcs: 17,
  cc-znp   csum: 17,
  cc-znp   sof: 254,
  cc-znp   len: 21,
  cc-znp   subsys: 'GREENPOWER',
  cc-znp   cmd: 'secReq',
  cc-znp   payload:
  cc-znp    { applicationID: 0,
  cc-znp      srcID: 24154724,
  cc-znp      gdpIeeeAddr: '0x0000000001709264',
  cc-znp      endpoint: 0,
  cc-znp      gpdfSecurityLevel: 2,
  cc-znp      gpdfSecurityFrameCounter: 1,
  cc-znp      dgpStubHandle: 95 } } +3ms
  unpi:parser --- parseNext [] +1ms
[... 122 lines cut ...]
  cc-znp   length: 21,
  cc-znp   fcs: 50,
  cc-znp   csum: 50,
  cc-znp   sof: 254,
  cc-znp   len: 21,
  cc-znp   subsys: 'GREENPOWER',
  cc-znp   cmd: 'secReq',
  cc-znp   payload:
  cc-znp    { applicationID: 0,
  cc-znp      srcID: 24154724,
  cc-znp      gdpIeeeAddr: '0x0000000001709264',
  cc-znp      endpoint: 0,
  cc-znp      gpdfSecurityLevel: 2,
  cc-znp      gpdfSecurityFrameCounter: 1,
  cc-znp      dgpStubHandle: 96 } } +4ms
  unpi:parser --- parseNext [] +1ms
  serialport/stream _read reading +2ms
  serialport/binding-abstract read +2ms
  serialport/bindings/unixRead Starting read +2ms
  cc-znp:AREQ <-- GREENPOWER:secReq, { applicationID: 0, srcID: 24154724, gdpIeeeAddr: '0x0000000001709264', endpoint: 0, gpdfSecurityLevel: 2, gpdfSecurityFrameCounter: 1, dgpStubHandle: 96 } +2ms
  zigbee-shepherd:msgHdlr IND <-- GREENPOWER:secReq +3ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +6ms

```

mabels, 2019-09-05:
I testest yesterday a bit more. And the Z-Stack 3.0 CC2531 sadly stops working completly after around 4-5 Received messages from the Switch. 
Do you have any idea why, or how to debug?

Koenkk, 2019-09-05:
I've also experienced this with my Hue Tap. I think that the device needs to be commissioned into the network, but it's unclear for me how to do this (that's where I got stuck).

jk, 2019-11-04:
Any news? Issue should not be closed by the bot.

Koenkk, 2019-11-15:
I've did some initial implementation but I need your help guys to test (very experimental).

On a bare metal install:
```bash
# Go to zigbee2mqtt directory
cd node_modules
rm -rf zigbee-herdsman
git clone https://github.com/Koenkk/zigbee-herdsman.git -b feature/greenpower
cd zigbee-herdsman
npm install 
npm run build
```

After this start zigbee2mqtt and start sniffing the network (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html).

Also make sure that there is at least one Hue bulb in the network (this will act as the green power sink).

Now reset the device, on the Hue tap I needed to hold either buttton 1, 2 or 3 depending on the channel (1=15, 2=20, 3=25) for 10 seconds.

After this zigbee2mqtt should log output when pressing a button on the device. If it doesn't work please share the sniffer logs.

cmorlok, 2019-11-24:
Does this mean that only Hue bulbs can act as a sink? Or do the CC253x or
any other router have this capability, too?

Koenkk, 2019-11-25:
@cmorlok I haven't seen any other devices providing this yet.

cmorlok, 2019-11-25:
Could the CC253x coordinator/router act as a sink? Or is it a hardware
limitation?

Koenkk, 2019-11-25:
@cmorlok it's not a hardware limitation, hopefully in the future we might be able to adapt the firmware correctly.

Webunity, 2019-11-27:
@Koenkk great work so far, is my assumption correct that:

- Support for the Hue Smart Switch is experimental
- If you reset your Friends of Hue Smart Switch this way; it will no longer be connected to the Hue app?
- But you will gain immediate feedback when pressing a button without the need for a high polling frequency on the Hue bridge?

Koenkk, 2019-11-27:
I don't know anything about how it works with the hue bridge, but when connected with zigbee2mqtt, the hue hub is completely out of the picture.

Webunity, 2019-11-27:
Great, since i want my motion sensors and buttons presses to run through node-red anyway; i'll buy the components and see how it will work out 👍

Koenkk, 2019-11-28:
@Webunity for know only Hue bulbs provide this, but could possibly be added to a CC2531 router firmware later. (but I first want to know if this feature works at all).

oepoemoepoe, 2019-12-01:
@Koenkk I will test your instructions as soon as I get my CC2531 and set everything up. I will also contribute 1 or 2 new lights. I just found out about this possibility to connect more Zigbee devices than with my Hue and must say I love the fact that contributing to it is so easy!

swannes, 2019-12-18:
Can we sponsor hardware to Koen so he can work on this directly (without any promisses regarding suport for said device!!)

TheNetStriker, 2019-12-23:
@Koenkk I've just tried to test this but I've got the following compilation error:

```
> zigbee-herdsman@0.12.5 build /opt/zigbee2mqtt_master/node_modules/zigbee-herdsman
> tsc

src/controller/greenPower.ts:33:32 - error TS2339: Property 'znp' does not exist on type 'Adapter'.

33             await this.adapter.znp.request(5, 'extAddGroup', {endpoint: 242, groupid: 0x0b84, namelen: 0, groupname:[]});
                                  ~~~

src/zcl/buffaloZcl.ts:180:21 - error TS2339: Property 'message' does not exist on type 'Options'.

180         if (options.message.commandID === 224) {
                        ~~~~~~~

src/zcl/zclFrame.ts:231:25 - error TS2339: Property 'message' does not exist on type 'Options'.

231                 options.message = payload;
                            ~~~~~~~

Found 3 errors.
```

I've tried it with the release and master branch but always got those errors. What am I doing wrong?

Koenkk, 2019-12-26:
@TheNetStriker can you share the sniffer logs?

TheNetStriker, 2019-12-26:
@Koenkk No problem, I've uploaded a capture file from the SmartRF Packet Sniffer to my Dropbox: https://www.dropbox.com/s/i0o42ibaelqim3a/HueTap.psd?dl=0

There are three packets for every button press. I pressed every button on the Hue tap, so there are 12 packets captured.

Koenkk, 2020-01-01:
@TheNetStriker sorry but could you please sniff with wireshark? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

TheNetStriker, 2020-01-02:
@Koenkk I've just uploaded the Wireshark capture here: https://www.dropbox.com/s/pk1jt5h5yvtkb36/HueTap.pcapng?dl=0

I could not register for the ZBOSS software because I didn't get the activation e-mail, so I created a capture using the [zsmartsystems sniffer](https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer). I hope this also works for you.

Koenkk, 2020-01-07:
@TheNetStriker do you also have a hue bulb in range of the hue tap? Can you provide the sniffer logs when starting with:

> Now reset the device, on the Hue tap I needed to hold either buttton 1, 2 or 3 depending on the channel (1=15, 2=20, 3=25) for 10 seconds.

https://github.com/Koenkk/zigbee-herdsman/issues/22#issuecomment-554522559

TheNetStriker, 2020-01-07:
@Koenkk Yes I only have Hue bulbs in my network. (32 in total) there should be lots of lamps in range.

What exactly do you mean? Do you need sniffer logs when resetting the Hue Tap or do you need all packets from my network including the Hue Tap ones? If you need all packets there would be a lot of them because the high amount of lamps in my network.

Koenkk, 2020-01-08:
I need the logs when resetting the hue tap including other devices (because an other device, e.g bulb commissions it into the network), perhaps you can setup a separate environment for this.

TheNetStriker, 2020-01-09:
@Koenkk I have a second Zigbee usb stick. I could make a separate network for testing the Hue Smart Switch.

But does the Smart Switch really require such a commission into the network? As far as I know the Smart Switch just sends out a broadcast on the channel and every Zigbee controller can receive this broadcast, but the switch is not able to receive anything because of the limited amount of energy.  The controller just needs to remember to which broadcast to react to and which to ignore. Did I understand this correctly or is there indeed a two way communication?

Koenkk, 2020-01-11:
@TheNetStriker 

Due to energy limitations, the message size of a Hue tap (= Green power device) are much smaller. This protocol is called green power. Normal zigbee devices don't understand this protocol, therefore you need to have a green power proxy (e.g. a Hue bulb) which translates a green power message to a normal zigbee message so that other devices on the network can understand it. But before the green power proxy will start proxying the message, the green power device first has to be commissioned into the network. There is indeed no two-way communication.

oepoemoepoe, 2020-01-11:
@Koenkk  I only have one CC2531, so I cannot sniff at the same time. However, I can offer you to sniff my Hue system with the Hue Tap.

On the theoretical side: I do not have a single Hue-bulb, but a lot of other devices (OSRAM Smart Plug, Paul Neuhaus lights etc) and still I can use the Hue Tap in my hue network. So why do we need a hue bulb to commission it?

TheNetStriker, 2020-01-13:
@Koenkk Thanks for the info. Do you know if the CC2531 would be able to receive such a green power message? Because as I said I sniffed the packets on a separate frequency where no devices are in range and I was able to sniff packages from the Hue Tap.

Also how would this comissioning into the network work exactly? Do I have to put the coordinator into pairing mode or is this something the Hue bulbs do directly?

oepoemoepoe, 2020-01-14:
> @Koenkk Thanks for the info. Do you know if the CC2531 would be able to receive such a green power message? Because as I said I sniffed the packets on a separate frequency where no devices are in [...]

To my understanding, you just need to have a Hue bulb in the same network with which you want to pair the Hue Tap with. Then zigbee2mqtt should do something (at least print out) when you try to pair it.
This can in the future be done without a Hue bulb, wherefore a firmware upgrade of the CC2531 is necessary however. For now Koenkk wants us to test without.
And of course the coordinator must be in pairing mode to pair.

Koenkk, 2020-01-15:
Maybe in the future the CC can act as a proxy, however for now this is not possible with the TI firmware: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/t/856997

TheNetStriker, 2020-01-17:
@Koenkk I've just tried this and now it worked! I got the following output in the console while comissioning the Hue Tap:

,"maintrel":1,"revision":20191106}},"log_level":"debug","permit_join":false}'
{
  options: 0,
  srcID: 4223047,
  frameCounter: 186,
  commandID: 224,
  payloadSize: 27,
  commandFrame: {
    deviceID: 2,
    options: 129,
    extendedOptions: 242,
    securityKey: hidden,
    keyMic: 640474694,
    outgoingCounter: 954
  }
}

oepoemoepoe, 2020-01-19:
@Koenkk Since it seems that firmware modification is too complex, do you think it will ever be supported? 
Also, do you know whether the Hue Smart Plug can act as a proxy, too?

Koenkk, 2020-01-21:
@oepoemoepoe I really depends on Texas Instruments (I'm at least not going to develop this). 

@TheNetStriker good to see, I'm going to buy another green power device as I think the security key required for commissioning differs per device (and I'm not sure how to handle this yet).

TheNetStriker, 2020-01-21:
@Koenkk I have several Hue Tap switches and I just wanted to test this, but for some reason the console output doesn't work anymore. On startup I also get the MQTT publish: topic 'zigbee2mqtt/bridge/config' log with "permit_join":false even though I've enabled this value. Any idea what could cause this?

Koenkk, 2020-01-22:
Im not sure perhaps the debug logging can give some insight

TheNetStriker, 2020-01-23:
@Koenkk I get the following error when starting the greenpower version:

```
(node:5475) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - extAddGroup - {"endpoint":242,"groupid":2948,"namelen":0,"groupname":[]}' failed with status '184' (expected '0')
    at Znp.<anonymous> (/opt/zigbee2mqtt_master/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:245:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt_master/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
```

I also deleted the database and all groups, but the error is still there. Could this be the cause that this doesn't work anymore?

oepoemoepoe, 2020-02-07:
I have a Hue Plug now which acts as a proxy and I can confirm what @TheNetStriker is saying. I get this output when trying to pair:

`  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":64695,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":141,"securityuse":0,"timestamp":14191052,"transseqnumber":0,"len":22,"data":{"type":"Buffer","data":[17,0,4,32,11,183,252,71,0,200,0,0,0,34,0,198,48,219,107,200,183,59]}} +2s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true},"transactionSequenceNumber":0,"manufacturerCode":null,"commandIdentifier":4},"Payload":{"options":2848,"srcID":4717751,"frameCounter":200,"commandID":34,"payloadSize":0,"commandFrame":{}},"Cluster":{"ID":33,"attributes":{},"name":"greenPower","commands":{"notification":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1007}],"name":"notification"},"commisioningNotification":{"ID":4,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1007}],"name":"commisioningNotification"}},"commandsResponse":{"pairing":{"ID":1,"parameters":[{"name":"options","type":34},{"name":"srcID","type":35},{"name":"sinkGroupID","type":33},{"name":"deviceID","type":32},{"name":"frameCounter","type":35},{"name":"gpdKey","type":1001}],"name":"pairing"},"commisioningMode":{"ID":2,"parameters":[{"name":"options","type":32},{"name":"commisioningWindow","type":33}],"name":"commisioningMode"}}}},"networkAddress":64695,"endpoint":242,"linkquality":141,"groupID":0}' +2s
{
  options: 2848,
  srcID: 4717751,
  frameCounter: 200,
  commandID: 34,
  payloadSize: 0,
  commandFrame: {}
}`
and afterwards:
`  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":64695,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":141,"securityuse":0,"timestamp":14242644,"transseqnumber":0,"len":45,"data":{"type":"Buffer","data":[17,1,4,0,8,183,252,71,0,201,0,0,0,224,27,2,129,242,104,61,188,120,123,162,166,202,218,34,183,96,82,182,93,100,217,66,151,241,201,0,0,0,198,48,217]}} +4s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true},"transactionSequenceNumber":1,"manufacturerCode":null,"commandIdentifier":4},"Payload":{"options":2048,"srcID":4717751,"frameCounter":201,"commandID":224,"payloadSize":27,"commandFrame":{"deviceID":2,"options":129,"extendedOptions":242,"securityKey":{"type":"Buffer","data":[104,61,188,120,123,162,166,202,218,34,183,96,82,182,93,100]},"keyMic":4053222105,"outgoingCounter":201}},"Cluster":{"ID":33,"attributes":{},"name":"greenPower","commands":{"notification":{"ID":0,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1007}],"name":"notification"},"commisioningNotification":{"ID":4,"parameters":[{"name":"options","type":33},{"name":"srcID","type":35},{"name":"frameCounter","type":35},{"name":"commandID","type":32},{"name":"payloadSize","type":32},{"name":"commandFrame","type":1007}],"name":"commisioningNotification"}},"commandsResponse":{"pairing":{"ID":1,"parameters":[{"name":"options","type":34},{"name":"srcID","type":35},{"name":"sinkGroupID","type":33},{"name":"deviceID","type":32},{"name":"frameCounter","type":35},{"name":"gpdKey","type":1001}],"name":"pairing"},"commisioningMode":{"ID":2,"parameters":[{"name":"options","type":32},{"name":"commisioningWindow","type":33}],"name":"commisioningMode"}}}},"networkAddress":64695,"endpoint":242,"linkquality":141,"groupID":0}' +4s
{
  options: 2048,
  srcID: 4717751,
  frameCounter: 201,
  commandID: 224,
  payloadSize: 27,
  commandFrame: {
    deviceID: 2,
    options: 129,
    extendedOptions: 242,
    securityKey: <Buffer 68 3d bc 78 7b a2 a6 ca da 22 b7 60 52 b6 5d 64>,
    keyMic: 4053222105,
    outgoingCounter: 201
  }
}
`

svh1985, 2020-03-08:
Would love to add my Hue Tap's, how can I help? I have a lot of HUE devices and some IKEA devices in my Zigbee network.

Koenkk, 2020-03-08:
I will soon buy some green power devices to continue on this.

mdhom, 2020-03-11:
I'm also very interested in that topic, i have a Senic Friends of Hue Smart Switch and could run some tests in case someone tells me what to do ;)

Josov, 2020-03-12:
@Koenkk I could provide you with a sample green power switch of Enocean. Would this help?

Download link:
[https://www.enocean.com/en/products/enocean_modules_24ghz/ptm-216z/user-manual-pdf/](url)

Koenkk, 2020-03-12:
@Josov is this device compatible with the Hue bridge?

Josov, 2020-03-13:
@Koenkk I have several models a double rocker for HUE only and GENERIC Zigbee switches single and double rocker. The HUE switch is currently in The Netherlands. I propose to send you one of each.
(jos.oude.vrielink@gmail.com)

Koenkk, 2020-04-07:
Thanks to @Josov I've got an initial implementation.

Demo:
- Hue Tap https://www.youtube.com/watch?v=7DsGrAfxjG0
- Niko https://www.youtube.com/watch?v=Cwp2sjP4t4o

juamiso, 2020-04-08:
I really want to install battery-less Switches at home to use them via zigbee and iobroker. Do you recommend the friends of hue smart switch or which one do you plan to support? I've read that the Philips hue tap ist not very good in haptic. 
Great job!

TheNetStriker, 2020-04-09:
@Koenkk Looks great! Is it already possible to test this myself?

cmorlok, 2020-04-09:
@TheNetStriker The branch where the development happens was mentioned above: https://github.com/Koenkk/zigbee-herdsman/commits/feature/greenpower

Koenkk, 2020-04-09:
Soon it will be available.

mdhom, 2020-04-09:
Absolutely amazing! Will it instantly support *all* GreenPower devices, or only the two mentioned above? I've got the Gira Friends of Hue Switch which i'd love to integrate...

Koenkk, 2020-04-09:
Probably all of these friends of hue switch will be supported (only tested with a Niko but I expect others will also work).

Koenkk, 2020-04-09:
Implemented, discussion: https://github.com/Koenkk/zigbee2mqtt/issues/3322

closed 2020-04-09 by Koenkk

JHurk, 2020-05-01:
@mdhom I am interested in buying one or more Senic Gira Friends of Hue switches, can you confirm that they work with the current version of Zigbee2MQTT? I see you have them and asked @Koenkk if they work, I would really like to know before I buy one of them.

mdhom, 2020-05-01:
I haven't tested it yet because i a had other big troubles with my Installation, but i could test it tomorrow!

JHurk, 2020-05-01:
@mdhom thanks that would be nice. Look forward to hear from you, take your time, no rush.

mdhom, 2020-05-02:
@Koenkk is there already a docker image available containing the new Feature?

Koenkk, 2020-05-02:
@mdhom green power support is available in the latest release (zigbee2mqtt 1.13.0)

mdhom, 2020-05-02:
it somehow worked and somehow didn't 😅 I was prepared to keep the up right button pressed for some time for pairing (read that anywhere), but exactly when i pressed that button, the following error appeared (but only in console, not in log file):

```
(node:51436) UnhandledPromiseRejectionWarning: TypeError [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received undefined
    at Cipheriv.update (internal/crypto/cipher.js:155:11)
    at GreenPower.encryptSecurityKey (C:\s\Koenkk\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\greenPower.js:50:34)
    at GreenPower.<anonymous> (C:\s\Koenkk\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\greenPower.js:57:34)    at Generator.next (<anonymous>)
    at C:\s\Koenkk\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\greenPower.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (C:\s\Koenkk\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\greenPower.js:4:12)
    at GreenPower.onZclGreenPowerData (C:\s\Koenkk\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\greenPower.js:54:16)
    at Controller.<anonymous> (C:\s\Koenkk\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\controller.js:405:37)
    at Generator.next (<anonymous>)
(node:51436) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:51436) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
Shortly after that error message, the following log appeared, looking like actually pairing was successfull:
```
info  2020-05-02 20:07:23: Device '0x0000000001715f61' joined
info  2020-05-02 20:07:23: Successfully interviewed '0x0000000001715f61', device has successfully been paired
info  2020-05-02 20:07:23: Device '0x0000000001715f61' is supported, identified as: GreenPower On/off switch (GreenPower_On_Off_Switch)
info  2020-05-02 20:07:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0000000001715f61"}}'
info  2020-05-02 20:07:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x0000000001715f61","model":"GreenPower_On_Off_Switch","vendor":"GreenPower","description":"On/off switch","supported":true}}'
```
Now, when i press one of the four buttons, i can see these logs:
```
info  2020-05-02 20:07:48: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"action":"recall_scene_1","linkquality":78}'
info  2020-05-02 20:07:48: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":13,"action":"recall_scene_5"}'
info  2020-05-02 20:07:52: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":144,"action":"press_1_of_2"}'
info  2020-05-02 20:07:52: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":141,"action":"release_1_of_2"}'
info  2020-05-02 20:07:57: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":81,"action":"recall_scene_1"}'
info  2020-05-02 20:07:57: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":0,"action":"recall_scene_5"}'
info  2020-05-02 20:08:02: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":144,"action":"recall_scene_1"}'
info  2020-05-02 20:08:03: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":0,"action":"recall_scene_5"}'
info  2020-05-02 20:08:05: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":144,"action":"recall_scene_0"}'
info  2020-05-02 20:08:05: MQTT publish: topic 'zigbee2mqtt/0x0000000001715f61', payload '{"linkquality":5,"action":"recall_scene_4"}'
```
Interestingly, i can not reproduce the log with action "press_1_of_2" and "release_1_of_2", i only get the "recall_scene_XY" logs. Is there any documentation on what events are available?

Koenkk, 2020-05-04:
@mdhom what firmware are you running? Have you been able to start it after this?

hb0nes, 2021-02-03:
Hey Koenkk. I have an EnOcean switch here (Senic & Gira 4 button switch) but don't have any Hue devices yet. Am I correct in assuming this is why I see nothing in my zigbee2mqtt logs?

[7 more replies]