Koenkk/zigbee-herdsman #25: Failure to add Konke Device.

tb-killa, 2019-09-14:
Hi @Koenkk 

I see this if i want to add the Konke Motion Device because it doesnt seem to add correctly.

```

  zigbee-herdsman:unpi:parser <-- [254,23,68,129,0,0,32,0,12,100,1,1,0,2,0,190,246,101,0,0,3,25,16,0,12,100,29,202] +24s
  zigbee-herdsman:unpi:parser --- parseNext [254,23,68,129,0,0,32,0,12,100,1,1,0,2,0,190,246,101,0,0,3,25,16,0,12,100,29,202] +0ms
  zigbee-herdsman:unpi:parser --> parsed 23 - 2 - 4 - 129 - [
d��e
d] - 202 +1ms
  zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":32,"srcaddr":25612,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":2,"securityuse":0,"timestamp":6682302,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[25,16,0]}} +24s
  zigbee-herdsman:controller:log Received ZCL data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":16,"manufacturerCode":null,"commandIdentifier":0},"Payload":{},"ClusterID":32},"networkAddress":25612,"endpoint":1,"linkquality":2,"groupID":0}' +24s
  zigbee-herdsman:unpi:parser --- parseNext [] +5ms
  zigbee-herdsman:controller:log Skipping command 'checkin' because it is missing from the lookup +2ms
  zigbee-herdsman:unpi:parser <-- [254,28,68,129,0,0,1,2,12,100,1,1,0,0,0,58,64,102,0,0,8,24,17,10,0,0,41,138,7,12,100,29,116] +6s
  zigbee-herdsman:unpi:parser --- parseNext [254,28,68,129,0,0,1,2,12,100,1,1,0,0,0,58,64,102,0,0,8,24,17,10,0,0,41,138,7,12,100,29,116] +1ms
  zigbee-herdsman:unpi:parser --> parsed 28 - 2 - 4 - 129 - [
d:@f
  zigbee-herdsman:unpi:parser )�
d] - 116 +1ms
```

It seems to be that some Encoding Problem occures ?

Koenkk, 2019-09-14:
I dont see anything wrong here, the strange characters you see are because the bytes are decoded as utf8.

tb-killa, 2019-09-18:
Hi @Koenkk 

The Device doesnt seems to allow fully interview:

```
serialport/binding-abstract read +290ms
  serialport/bindings/unixRead Starting read +289ms
  serialport/bindings/unixRead Finished read 17 bytes +1ms
  serialport/stream binding.read finished +293ms
  zigbee-herdsman:unpi:parser <-- [254,12,69,202,176,138,231,123,233,254,255,215,107,8,0,0,121] +295ms
  zigbee-herdsman:unpi:parser --- parseNext [254,12,69,202,176,138,231,123,233,254,255,215,107,8,0,0,121] +1ms
  zigbee-herdsman:unpi:parser --> parsed 12 - 2 - 5 - 202 - [���{����] - 121 +1ms
  zigbee-herdsman:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":35504,"extaddr":"0x086bd7fffee97be7","parentaddr":0} +343ms
  zigbee-herdsman:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:controller:log Device joined '0x086bd7fffee97be7' +344ms
  zigbee-herdsman:controller:log Interview '0x086bd7fffee97be7' start +1ms
  zigbee-herdsman:controller:device Interview - start device '0x086bd7fffee97be7' +2m
  zigbee2mqtt:info 9/18/2019, 11:17:21 AM Starting interview of 'konke_motionsensor'
  zigbee2mqtt:info 9/18/2019, 11:17:21 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendlyName":"konke_motionsensor"}}'
  serialport/stream _read reading +16ms
  serialport/binding-abstract read +19ms
  serialport/bindings/unixRead Starting read +19ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +2ms
  serialport/bindings/poller Polling for "readable" +24ms
  serialport/bindings/poller received "readable" +1ms
  serialport/binding-abstract read +6ms
  serialport/bindings/unixRead Starting read +2ms
  serialport/bindings/unixRead Finished read 8 bytes +1ms
  serialport/stream binding.read finished +8ms
  zigbee-herdsman:unpi:parser <-- [254,3,68,128,0,1,4,194] +19ms
  zigbee-herdsman:unpi:parser --- parseNext [254,3,68,128,0,1,4,194] +0ms
  zigbee-herdsman:unpi:parser --> parsed 3 - 2 - 4 - 128 - [] - 194 +1ms
  zigbee-herdsman:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":4} +22ms
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":35504,"nwkaddrofinterest":35504} +359ms
  zigbee-herdsman:unpi:writer --> frame [254,4,37,2,176,138,176,138,35] +359ms
  serialport/stream _write 9 bytes of data +7ms
  serialport/binding-abstract write 9 bytes +9ms
  serialport/bindings/unixWrite Starting write 9 bytes offset 0 bytesToWrite 9 +345ms
  serialport/stream _read reading +4ms
  serialport/binding-abstract read +3ms
  serialport/bindings/unixRead Starting read +12ms
  serialport/bindings/unixWrite write returned null 9 +5ms
  serialport/bindings/unixWrite wrote 9 bytes +1ms
  serialport/bindings/unixWrite Finished writing 9 bytes +0ms
  serialport/stream binding.write write finished +4ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +4ms
  serialport/bindings/poller Polling for "readable" +19ms
  serialport/bindings/poller received "readable" +1ms
  serialport/binding-abstract read +7ms
  serialport/bindings/unixRead Starting read +3ms
  serialport/bindings/unixRead Finished read 6 bytes +1ms
  serialport/stream binding.read finished +6ms
  zigbee-herdsman:unpi:parser <-- [254,1,101,2,0,102] +19ms
  zigbee-herdsman:unpi:parser --- parseNext [254,1,101,2,0,102] +1ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 5 - 2 - [] - 102 +1ms
  zigbee-herdsman:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +344ms
  zigbee-herdsman:unpi:parser --- parseNext [] +3ms
  serialport/stream _read reading +8ms
  serialport/binding-abstract read +11ms
  serialport/bindings/unixRead Starting read +10ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +14ms
  serialport/bindings/poller received "readable" +586ms
  serialport/binding-abstract read +589ms
  serialport/bindings/unixRead Starting read +588ms
  serialport/bindings/unixRead Finished read 23 bytes +1ms
  serialport/stream binding.read finished +591ms
  zigbee-herdsman:unpi:parser <-- [254,18,69,130,176,138,0,176,138,2,64,128,2,16,82,82,0,0,44,82,0,0,123] +594ms
  zigbee-herdsman:unpi:parser --- parseNext [254,18,69,130,176,138,0,176,138,2,64,128,2,16,82,82,0,0,44,82,0,0,123] +1ms
[... 155 lines cut ...]
  zigbee-herdsman:unpi:parser --- parseNext [] +2ms
  zigbee2mqtt:debug 9/18/2019, 11:18:31 AM Received Zigbee message from 'konke_motionsensor' of type 'readResponse' with data '{"manufacturerName":"Konke","modelId":"3AFE14010402000D","powerSource":0}' from endpoint 1 with groupID 0
  zigbee2mqtt:error 9/18/2019, 11:18:31 AM Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:85:42)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:84:52)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:310:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:221:14)
    at process._tickCallback (internal/process/next_tick.js:68:7))
  serialport/stream _read reading +19ms
  serialport/binding-abstract read +21ms
  serialport/bindings/unixRead Starting read +21ms
  serialport/bindings/unixRead Finished read 8 bytes +1ms
  serialport/stream binding.read finished +3ms
[... cut ...]
  zigbee-herdsman:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240 +0ms
  zigbee-herdsman:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +21ms
  zigbee-herdsman:unpi:parser --- parseNext [] +2ms
  serialport/stream _read reading +5ms
  serialport/binding-abstract read +8ms
  serialport/bindings/unixRead Starting read +7ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +32ms
```

tb-killa, 2019-09-18:
Hi @Koenkk 

it seems that the data doesnt transport correctly :(

```
zigbee-herdsman:controller:log Permit joining +2m
  zigbee-herdsman:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +2m
  zigbee-herdsman:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228] +2m
  serialport/stream _write 10 bytes of data +2m
  serialport/binding-abstract write 10 bytes +2m
  serialport/bindings/unixWrite Starting write 10 bytes offset 0 bytesToWrite 10 +2m
  serialport/bindings/unixWrite write returned null 10 +1ms
  serialport/bindings/unixWrite wrote 10 bytes +0ms
  serialport/bindings/unixWrite Finished writing 10 bytes +1ms
  serialport/stream binding.write write finished +6ms
  serialport/bindings/poller received "readable" +2m
  serialport/binding-abstract read +14ms
  serialport/bindings/unixRead Starting read +2m
  serialport/bindings/unixRead Finished read 70 bytes +1ms
  serialport/stream binding.read finished +11ms
  zigbee-herdsman:unpi:parser <-- [254,59,68,129,0,0,0,0,96,52,1,1,0,36,0,164,11,212,0,0,39,24,2,1,4,0,0,66,5,75,111,110,107,101,5,0,0,66,16,51,65,70,69,49,52,48,49,48,52,48,50,48,48,48,68,7,0,0,48,0,96,52,29,224,254,1,101,54,0,82] +2m
  zigbee-herdsman:unpi:parser --- parseNext [254,59,68,129,0,0,0,0,96,52,1,1,0,36,0,164,11,212,0,0,39,24,2,1,4,0,0,66,5,75,111,110,107,101,5,0,0,66,16,51,65,70,69,49,52,48,49,48,52,48,50,48,48,48,68,7,0,0,48,0,96,52,29,224,254,1,101,54,0,82] +1ms
  zigbee-herdsman:unpi:parser --> parsed 59 - 2 - 4 - 129 - [`4$�
                                                                 �'BKonkeB3AFE14010402000D0`4] - 224 +1ms
  zigbee-herdsman:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":13408,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":13896612,"transseqnumber":0,"len":39,"data":{"type":"Buffer","data":[24,2,1,4,0,0,66,5,75,111,110,107,101,5,0,0,66,16,51,65,70,69,49,52,48,49,48,52,48,50,48,48,48,68,7,0,0,48,0]}} +2m
  zigbee-herdsman:controller:log Received ZCL data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":2,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":4,"status":0,"dataType":66,"attrData":"Konke"},{"attrId":5,"status":0,"dataType":66,"attrData":"3AFE14010402000D"},{"attrId":7,"status":0,"dataType":48,"attrData":0}],"ClusterID":0},"networkAddress":13408,"endpoint":1,"linkquality":36,"groupID":0}' +29ms
  zigbee-herdsman:unpi:parser --- parseNext [254,1,101,54,0,82] +6ms
  zigbee-herdsman:unpi:parser --> parsed 1 - 3 - 5 - 54 - [] - 82 +1ms
  zigbee-herdsman:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0} +2m
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  zigbee2mqtt:debug 9/18/2019, 11:48:59 AM Received Zigbee message from '0x086bd7fffee97be7' of type 'readResponse' with data '{"manufacturerName":"Konke","modelId":"3AFE14010402000D","powerSource":0}' from endpoint 1 with groupID 0
  zigbee2mqtt:error 9/18/2019, 11:48:59 AM Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:85:42)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:84:52)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:310:44)
    at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:221:14)
    at process._tickCallback (internal/process/next_tick.js:68:7))
  serialport/stream _read reading +22ms
  serialport/binding-abstract read +25ms
  serialport/bindings/unixRead Starting read +25ms
  serialport/bindings/unixRead Finished read 8 bytes +1ms
  serialport/stream binding.read finished +3ms
  zigbee-herdsman:unpi:parser <-- [254,3,69,182,0,0,0,240] +15ms
  zigbee-herdsman:unpi:parser --- parseNext [254,3,69,182,0,0,0,240] +1ms
  zigbee-herdsman:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240 +1ms
  zigbee-herdsman:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0} +25ms
  zigbee-herdsman:unpi:parser --- parseNext [] +1ms
  serialport/stream _read reading +5ms
  serialport/binding-abstract read +9ms
  serialport/bindings/unixRead Starting read +6ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +36ms
```

This are my database.db entrys, sadly new ones added even after restarting the z2m instance.

```
{"id":14,"type":"EndDevice","ieeeAddr":"0x086bd7fffee97be7","nwkAddr":13408,"manufId":4098,"manufName":"Konke","powerSource":"Unknown","modelId":"3AFE14010402000D","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[3],"clusters":{}}},"interviewCompleted":false,"meta":{},"_id":"vYoav5rL14yvaeYD"}
{"id":14,"type":"EndDevice","ieeeAddr":"0x086bd7fffee97be7","nwkAddr":13408,"manufId":4098,"manufName":"Konke","powerSource":"Unknown","modelId":"3AFE14010402000D","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{}}},"interviewCompleted":false,"meta":{},"_id":"vYoav5rL14yvaeYD"}
{"id":14,"type":"EndDevice","ieeeAddr":"0x086bd7fffee97be7","nwkAddr":13408,"manufId":4098,"manufName":"Konke","powerSource":"Unknown","modelId":"3AFE14010402000D","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[3],"clusters":{}}},"interviewCompleted":false,"meta":{},"_id":"vYoav5rL14yvaeYD"}
{"id":14,"type":"EndDevice","ieeeAddr":"0x086bd7fffee97be7","nwkAddr":13408,"manufId":4098,"manufName":"Konke","powerSource":"Unknown","modelId":"3AFE14010402000D","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280],"outClusterList":[3],"clusters":{}}},"interviewCompleted":false,"meta":{},"_id":"vYoav5rL14yvaeYD"}
```

closed 2019-09-19 by tb-killa