Koenkk/zigbee-herdsman #26: Problem installing. RPi3 latest raspibian lite

juantecnico, 2019-09-14:
Hi. have follow aguide to install zigbee2mqtt and i try to install zigbee-herdsman
I had cloned the git and try to install with npm install inside zigbee-herdsman directoy but this error always appears.

> 
pi@raspberrypi:/opt/zigbee-herdsman $ sudo npm install
npm WARN tarball tarball data for typescript@3.5.3 (sha512-ACzBtm/PhXBDId6a6sDJfroT2pOWt/oOnk4/dElG5G33ZL776N3Y6/6bKZJBFpd+b05F3Ct9qDjMeJmRWtE2/g==) seems to be corrupted. Trying one more time.
npm WARN tarball tarball data for typescript@3.5.3 (sha512-ACzBtm/PhXBDId6a6sDJfroT2pOWt/oOnk4/dElG5G33ZL776N3Y6/6bKZJBFpd+b05F3Ct9qDjMeJmRWtE2/g==) seems to be corrupted. Trying one more time.
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee-herdsman/node_modules/.staging/fsevents-e48a7acc/node_modules/strip-ansi/index.js'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee-herdsman/node_modules/.staging/fsevents-e48a7acc/node_modules/strip-ansi/license'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee-herdsman/node_modules/.staging/fsevents-e48a7acc/node_modules/strip-ansi/package.json'
npm WARN tar ENOENT: no such file or directory, open '/opt/zigbee-herdsman/node_modules/.staging/fsevents-e48a7acc/node_modules/strip-ansi/readme.md'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.9 (node_modules/fsevents):
npm WARN enoent SKIPPING OPTIONAL DEPENDENCY: ENOENT: no such file or directory, rename '/opt/zigbee-herdsman/node_modules/.staging/fsevents-e48a7acc/node_modules/abbrev' -> '/opt/zigbee-herdsman/node_modules/.staging/abbrev-00ac77d5'

npm ERR! code EINTEGRITY
npm ERR! Verification failed while extracting typescript@3.5.3:
npm ERR! Verification failed while extracting typescript@3.5.3:
npm ERR! sha512-ACzBtm/PhXBDId6a6sDJfroT2pOWt/oOnk4/dElG5G33ZL776N3Y6/6bKZJBFpd+b05F3Ct9qDjMeJmRWtE2/g== integrity checksum failed when using sha512: wanted sha512-ACzBtm/PhXBDId6a6sDJfroT2pOWt/oOnk4/dElG5G33ZL776N3Y6/6bKZJBFpd+b05F3Ct9qDjMeJmRWtE2/g== but got sha512-5w25EOryuAGqTJle2+AoxjNQqem3oGrGAk1iUSDyhMTt5cwLgQUoj8fDBfoRf2gMdnWHMXfOaM2z+W0Qf/8Hbg==. (2296540 bytes)

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-09-14T20_29_24_795Z-debug.log

> 

Tried width differents SD cards. nothing helps.
Tried to install typescript, installation went ok but not help.

Thanks.

john2exonets, 2019-09-28:
+1 .  Both typescript@3.5.3 and zigbee-herdsman@0.5.7 come up with corruption errors and won't install.

dartrax, 2019-10-31:
> EDIT2: well, that didn't last too long. When starting:
> Error: Cannot find module 'zigbee-herdsman/dist/depreciated/ziee'
> ...I have no 'depreciated' directory in 'dist'. 

I have the same problem.

> npm install zigbee-hersman@0.5.7 --save
> This gave me some warnings about typescript-eslint, bl, and skipping some optional dependencies, but the install completed.
> npm start
> Yeah! getting status messages and I can see 'bridge' MQTT packets going out!!

This does not solve it for me. Still the same :(

closed 2020-01-07 by stale[bot]