Koenkk/zigbee-herdsman #29: Error messages during startup

Kryzek, 2019-09-17:
```
  zigbee2mqtt:info 9/17/2019, 8:12:14 AM Starting zigbee2mqtt version 1.6.0 (commit #d24c41c)
  zigbee2mqtt:info 9/17/2019, 8:12:14 AM Starting zigbee-herdsman...
(node:4150) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'find' of undefined
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:154:51)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:3:12)
    at Function.find (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:144:16)
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:139:40)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
(node:4150) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 4)
(node:4150) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:4150) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'find' of undefined
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:154:51)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:3:12)
    at Function.find (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:144:16)
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:139:40)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
(node:4150) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 5)
(node:4150) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'find' of undefined
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:154:51)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:3:12)
    at Function.find (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:144:16)
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:139:40)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
(node:4150) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 6)
(node:4150) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'find' of undefined
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:154:51)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:3:12)
    at Function.find (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:144:16)
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:139:40)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
(node:4150) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 9)
(node:4150) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'find' of undefined
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:154:51)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:3:12)
    at Function.find (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:144:16)
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:139:40)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
(node:4150) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 10)
  zigbee2mqtt:info 9/17/2019, 8:12:15 AM zigbee-herdsman started

```
This is with latest zigbee2mqtt development version.

Koenkk, 2019-09-17:
Can you check if this has been fixed in the latest dev?

Kryzek, 2019-09-18:
```
  zigbee2mqtt:info 9/18/2019, 8:41:25 AM Starting zigbee2mqtt version 1.6.0 (commit #d7401c8)
  zigbee2mqtt:info 9/18/2019, 8:41:25 AM Starting zigbee-herdsman...
(node:11058) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'find' of undefined
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:154:61)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:3:12)
    at Function.find (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:144:16)
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:139:40)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
(node:11058) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 3)
(node:11058) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:11058) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'find' of undefined
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:154:61)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:3:12)
    at Function.find (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:144:16)
    at Function.<anonymous> (/opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:139:40)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt-herdsman/node_modules/zigbee-herdsman/dist/controller/model/device.js:7:71
    at new Promise (<anonymous>)
(node:11058) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 4)
  zigbee2mqtt:info 9/18/2019, 8:41:27 AM zigbee-herdsman started
  zigbee2mqtt:info 9/18/2019, 8:41:27 AM Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
```

Not yet it seems.

Koenkk, 2019-09-18:
What nodejs version are you using?

Kryzek, 2019-09-18:
```
node --version
v10.15.1
```

Koenkk, 2019-09-18:
Can you test with latest zigbee2mqtt-dev again? If it doesn't work please provide the log when running with `DEBUG=* npm start`.

Kryzek, 2019-09-19:
Error messages are now gone with latest zigbee2mqtt commit #e9a336f 👍

Koenkk, 2019-09-19:
Great!

commit 4c42698e (2019-09-17): Attempt to fix #29
closed 2019-09-19 by Koenkk