Koenkk/zigbee-herdsman #32: using cc2530 with UART and arduino

guitar9, 2019-09-20:
I dont know how much work this is and I dont know if it would work, but I wanted to use cc2530 over UART connected to an esp8266 or arduino. 
I dont know where to start. I flashed the z-stack firmware from Koenkk on the cc2530 and connected with UART to arduino. Somebody give me a starting point where to start? 
Should I start with the development Board from Texas Instruments?

SAGARP169, 2019-09-26:
@guitar9 Did you get any way out of it

guitar9, 2019-09-27:
@SAGARP169  no. But I am interested to start developing. I dont know really where to start. Do you have a starting point?

SAGARP169, 2019-09-28:
I have almost reached, but cannot communicate the model with micro controller, the only thing i can do is send and receive data through the third party software, i cannot read/send this data through any com port (i.e arduino ide serial monitor).

[image]

in the above image you can see how i send command from co-ordinator to router and vice versa.

What i have used is cc2530+cc2592 module from **EBYTE** Model-"**E18-MS1PA1-PCB**"
No firmware need to upload initially it comes with the ready firmware and software can be downloaded from there website for relative product
software which i used was "Zigbee Setting v1.2"

SAGARP169, 2019-09-28:
I am very bad in programming, so can't even understand program properly, if you get any related info please share 
Email:
Thank You!!

wassfila, 2019-10-19:
If you'd like to port zigbee2mqtt to arduino, that's a lots of code.
You should first go to the lower layers : zigbee2mqtt is using zigbee-shepherd, and zigbee-shepherd is using cc-znp which is the one using the serial port. Or you could start a new implementation by checking the TI documentation, the UART protocol for the znp is documented on "Z-Stack Monitor and Test API".

closed 2019-12-25 by stale[bot]