Koenkk/zigbee-herdsman #40: bind tracking

hobbyquaker, 2019-09-28:
Hey Koenkk, I think a new Issue for discussions around the bind tracking makes sense :-)

Right now I'm doing the tracking in the device.meta, looks like this:
```
"binds":[{"endpoint":1,"cluster":"genOnOff","targetType":"endpoint","targetDevice":"0x0017880102fe0180","targetEndpoint":11}]
```

Should we move that to the Endpoint Object instead? What do you think of the scheme itself? Using a Set instead of an Array? I think a Set will not offer an advantage as it will not prevent adding to Objects with same content. So right now I'm using a construct like this
```
if (!device.meta.binds.find(b => {
    return b.endpoint === bind.endpoint &&
            b.cluster === bind.cluster &&
            b.targetDevice === bind.targetDevice &&
            b.targetEndpoint === bind.targetEndpoint;
})) {
    device.meta.binds.push(bind);
    device.save();
}
```
to check if a bind already exists. Do you have a more elegant suggestion for this?

Koenkk, 2019-09-29:
This indeed should be moved to the endpoint, as you bind an endpoint not a device.

That code below looks good, however don't save this info in the `meta` but rather create a new property on `endpoint` called `binds`.

hobbyquaker, 2019-09-29:
A first try, could you pls review?
https://github.com/hobbyquaker/zigbee-herdsman/commit/1c4fee6295ee8df71b96f8d6ec746adb4946a628

Koenkk, 2019-09-29:
Looks good, can you make a PR? Then we will work further from there.

hobbyquaker, 2019-09-29:
will do asap.

closed 2019-10-02 by hobbyquaker