Koenkk/zigbee-herdsman #54: Hard vs. Soft-Reset

hobbyquaker, 2019-10-04:
Hi!

A question for my understand: what exactly does a hard reset do? A soft reset is (as far as I understood that right) so to say a restart of the Z-Stack Software, right?

I had sometimes the problem that device interviews on pairing failed - worked around that by stopping the herdsman, turning USB power off and on and then it always worked again. A soft-reset didn't help, would a hard reset help in that case?

Regards,
Sebastian

Koenkk, 2019-10-05:
- A soft reset is indeed a restart of the z-stack software
- AFAIK a hard reset will factory reset the device (remove all pairings); but I haven't used/dived into this tbh.

hobbyquaker, 2019-10-05:
thx for clarification

Koenkk, 2019-10-05:
However I think that the current implementation is not complete; it will reset z-stack but I think it is not usable anymore afterwards (probably requires restart of zigbee-herdsman). Also `database.db` should be cleared after this.

sjorge, 2019-12-04:
> However I think that the current implementation is not complete; it will reset z-stack but I think it is not usable anymore afterwards (probably requires restart of zigbee-herdsman). Also `database.db` should be cleared after this.

If a hard reset is the same as wiping the config section you can do with the TI flasher... then yes... the database needs to be remove and also the groups section in configuration... we can leave devices as is... also after that a new panID is needed as the old one is still taken by the devices that are now coordinater less

closed 2020-02-10 by stale[bot]