Koenkk/zigbee-herdsman #60: Support OTA upgrades

lorek123, 2019-10-09:
It was mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/1865#issuecomment-522321328 that this should be at some point implmented in herdsman

Koenkk, 2019-10-09:
Would be great if somebody could help with this!

lorek123, 2019-10-10:
I can help test it with Ikea Tradfri devices

tb-killa, 2019-10-13:
note to myself : use https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/linux/deCONZ/usr/bin/deCONZ-ikea-ota-download.sh to get ikea firmware files.

tb-killa, 2019-10-16:
@Koenkk Where on herdsman i could implement ``genOta`` in and out for EP1 ?
I will rework my stuff from old shepherd but need first some basic basement on herdsman :)

Koenkk, 2019-10-16:
I think you need to add the `genOta` cluster similar like https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/adapter/startZnp.ts#L48

Afterwards you can use https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L208 to get the coordinator and  first endpoint of it to start sending requests.

zigbee-herdsman has no replacement for zapp (which zigbee-shepherd had)

tb-killa, 2019-10-17:
@Koenkk I add a new Endpoint:
```
Object.assign(Object.assign({}, EndpointDefaults), { endpoint: 9, appprofid: 0x0104, appnuminclusters: 1, appinclusterlist: [Zcl.Utils.getCluster('genOta').ID], appnumoutclusters: 1, appoutclusterlist: [Zcl.Utils.getCluster('genOta').ID]}),
```

and got this new messages after some Times from my Bitron Test-Device:

```
zigbee-herdsman:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":19904,"srcendpoint":1,"dstendpoint":9,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3135755,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,1,1,0,113,16,235,0,225,0,0,0]}} +5s
  zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":25,"data":{"type":"Buffer","data":[1,1,1,0,113,16,235,0,225,0,0,0]},"networkAddress":19904,"endpoint":1,"linkquality":0,"groupID":0}' +5s
zigbee2mqtt:debug 2019-10-17T07:30:59: Received Zigbee message from 'wandthermostat_wohnzimmer', type 'raw', cluster 'genOta', data '{"type":"Buffer","data":[1,1,1,0,113,16,235,0,225,0,0,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2019-10-17T07:30:59: No converter available for 'AV2010/32' with cluster 'genOta' and type 'raw' and data '{"type":"Buffer","data":[1,1,1,0,113,16,235,0,225,0,0,0]}'
zigbee2mqtt:warn  2019-10-17T07:30:59: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee-herdsman:zStack:unpi:parser --- parseNext [] +12ms
  zigbee-herdsman:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,192,77,1,9,0,0,0,199,252,47,0,0,12,1,2,1,0,113,16,235,0,225,0,0,0,48,196,28,224] +3s
  zigbee-herdsman:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,192,77,1,9,0,0,0,199,252,47,0,0,12,1,2,1,0,113,16,235,0,225,0,0,0,48,196,28,224] +1ms
  zigbee-herdsman:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [�M  ��/
q��0�] - 224 +0ms
  zigbee-herdsman:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":19904,"srcendpoint":1,"dstendpoint":9,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3144903,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,2,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":25,"data":{"type":"Buffer","data":[1,2,1,0,113,16,235,0,225,0,0,0]},"networkAddress":19904,"endpoint":1,"linkquality":0,"groupID":0}' +3s
zigbee2mqtt:debug 2019-10-17T07:31:02: Received Zigbee message from 'wandthermostat_wohnzimmer', type 'raw', cluster 'genOta', data '{"type":"Buffer","data":[1,2,1,0,113,16,235,0,225,0,0,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2019-10-17T07:31:02: No converter available for 'AV2010/32' with cluster 'genOta' and type 'raw' and data '{"type":"Buffer","data":[1,2,1,0,113,16,235,0,225,0,0,0]}'
zigbee2mqtt:warn  2019-10-17T07:31:02: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee-herdsman:zStack:unpi:parser --- parseNext [] +11ms
  zigbee-herdsman:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,192,77,1,9,0,0,0,148,32,48,0,0,12,1,3,1,0,113,16,235,0,225,0,0,0,48,196,28,113] +3s
  zigbee-herdsman:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,192,77,1,9,0,0,0,148,32,48,0,0,12,1,3,1,0,113,16,235,0,225,0,0,0,48,196,28,113] +1ms
  zigbee-herdsman:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [�M  � 0
q��0�] - 113 +1ms
  zigbee-herdsman:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":19904,"srcendpoint":1,"dstendpoint":9,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3154068,"transseqnumber":0,"len":12,"data":{"type":"Buffer","data":[1,3,1,0,113,16,235,0,225,0,0,0]}} +3s
  zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":25,"data":{"type":"Buffer","data":[1,3,1,0,113,16,235,0,225,0,0,0]},"networkAddress":19904,"endpoint":1,"linkquality":0,"groupID":0}' +3s
zigbee2mqtt:debug 2019-10-17T07:31:05: Received Zigbee message from 'wandthermostat_wohnzimmer', type 'raw', cluster 'genOta', data '{"type":"Buffer","data":[1,3,1,0,113,16,235,0,225,0,0,0]}' from endpoint 1 with groupID 0
zigbee2mqtt:warn  2019-10-17T07:31:05: No converter available for 'AV2010/32' with cluster 'genOta' and type 'raw' and data '{"type":"Buffer","data":[1,3,1,0,113,16,235,0,225,0,0,0]}'
zigbee2mqtt:warn  2019-10-17T07:31:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee-herdsman:zStack:unpi:parser --- parseNext [] +11ms
  zigbee-herdsman:zStack:unpi:parser <-- [254,5,69,196,192,77,1,48,196,252] +3s
  zigbee-herdsman:zStack:unpi:parser --- parseNext [254,5,69,196,192,77,1,48,196,252] +0ms
  zigbee-herdsman:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [�M0�] - 252 +2ms
  zigbee-herdsman:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":19904,"relaycount":1,"relaylist":[50224]} +3s
  zigbee-herdsman:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:zStack:unpi:parser <-- [254,32,68,129,0,0,25,0,192,77,1,9,0,0,0,136,68,48,0,0,12,1,4,1,0,113,16,235,0,225,0,0,0,48,196,28,14] +11ms
  zigbee-herdsman:zStack:unpi:parser --- parseNext [254,32,68,129,0,0,25,0,192,77,1,9,0,0,0,136,68,48,0,0,12,1,4,1,0,113,16,235,0,225,0,0,0,48,196,28,14] +11ms
  zigbee-herdsman:zStack:unpi:parser --> parsed 32 - 2 - 4 - 129 - [�M  �D0
q��0�] - 14 +1ms
  zigbee-herdsman:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":25,"srcaddr":19904,"srcendpoint":1,"dstendpoint":9,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":3163272,"transseqnumber":0,"len":12,"data [... cut ...]
zigbee2mqtt:debug 2019-10-17T07:31:11: Received Zigbee message from 'wandthermostat_wohnzimmer', type 'raw', cluster 'genOta', data '{"type":"Buffer","data":[1,5,1,0,113,16,235,0,225,0,0,0]}' from endpoint 1 with groupID 0
PuTTYzigbee2mqtt:warn  2019-10-17T07:31:11: No converter available for 'AV2010/32' with cluster 'genOta' and type 'raw' and data '{"type":"Buffer","data":[1,5,1,0,113,16,235,0,225,0,0,0]}'
zigbee2mqtt:warn  2019-10-17T07:31:11: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee-herdsman:zStack:unpi:parser --- parseNext [] +10ms
```
Now we need to ignore all ``genOta`` on converters because we would handle them internally.

Koenkk, 2019-10-17:
Change https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/controller.ts#L424
to

```js
if (type && data && cluster !== 'genOta') {
```

tb-killa, 2019-10-17:
@Koenkk maybe we could work together ? 
I implement this stuff on branch 'ota' : https://github.com/tb-killa/zigbee-herdsman/tree/ota

sjorge, 2019-11-02:
Just of minor interest, ubisys allows us to request the OTA firmware once we have this working!

https://www.ubisys.de/wp-content/uploads/ubisys-s1-technical-reference.pdf

```
7.7.4.Over-the-Air Upgrade Cluster(Client)The image type for S1 is 0x7B02, for S1-R it is 0x7B05. You may request the latest firmware in ZigBee OTA image format to upgrade devices in the field using your own OTA server and back-end. If you operate the power switch S1together with the ZigBee/Ethernet Gateway ubisys G1, the latest firmware for S1 and S1-R will always be available automatically and you do not need to request it explicitly.
```

**You may request the latest firmware in ZigBee OTA image format to upgrade devices in the field using your own OTA server and back-end.**

BudBundi, 2019-11-28:
Ledvance/Osram Smart+ firmware can be found here: https://update.ledvance.com/firmware-overview?submit=all

Maybe this also helps https://phoscon.de/en/support#ota-update-osram-devices

jdtsmith, 2019-12-22:
What's the status of your OTA branch @tb-killa?  The Deconz folks have OTA working for a variety of devices.

andreasbrett, 2019-12-25:
If you need a tester, just let me know.

unityhd, 2019-12-25:
I also offer myself as a tester.
I have a CC2351 stick and some ZigBee devices from Ledvance/Osram, Xiaomi, Aqara etc.

atxbyea, 2019-12-31:
I too can test if needed, CC1352P-2 and Tradfri, hue, Xiaomi, gledopto here

dbruggner, 2020-01-03:
Ledvance/Osram has a REST API to download firmware info and firmware files: https://portal.update.ledvance.com/docs/services/firmware-rest-api/operations/5d8e482edb442c3151c64fae

twsI, 2020-01-13:
Is there any way I can help or contribute?

CodeFinder2, 2020-01-19:
Would also love to see this feature and glad to help testing (Bitron, OSRAM/Ledvance, Xiaomi/Aqara/Mija, Konke, Philips, Heimann, Ikea, ...) 🙈

twsI, 2020-01-22:
@Koenkk if you'd start with the general architecture, I could implement the code for certain vendors. Got Xiaomi, OSRAM, Ikea for testing as well

Koenkk, 2020-01-22:
@twsI good idea, will let you know here once its available.

svh1985, 2020-02-06:
OTA getting more important:
_Signify confirmed the existence of the vulnerability in their product, and issued a patched firmware version (Firmware 1935144040) which is now available on their site. We recommend users to make sure that their product received the automatic update of this firmware version._

Full article:
https://blog.checkpoint.com/2020/02/05/the-dark-side-of-smart-lighting-check-point-research-shows-how-business-and-home-networks-can-be-hacked-from-a-lightbulb/

lwfitzgerald, 2020-02-07:
I went looking for the OTA firmwares for Salus devices this evening and after sniffing the traffic from their iOS app, I found them :)

https://eu.salusconnect.io/demo/default/status/firmware?timestamp=0.

Koenkk, 2020-02-08:
First OTA update with zigbee-herdsman/zigbee2mqtt is a fact!

[image]

Before: `"dateCode":"20170315","swBuildId":"1.3.002"`
After: `"dateCode":"20180410","swBuildId":"1.3.002"`

CodeFinder2, 2020-02-08:
Is it a good idea to collect and upload (the most recent) firmware images somewhere? Maybe even collecting (links to) them on the appropriate wiki device pages, perhaps with a hint like 'tested'? 🙈

CodeFinder2, 2020-02-08:
Uhh, that's great! ☺️👌

In general, will it be difficult to port this to other vendors like OSRAM or Xiaomi?

PedroLamas, 2020-02-08:
I have a couple of Tradfri start plugs that will certainly make use of this feature!

timstanley1985, 2020-02-08:
Great work @Koenkk !  Can I suggest that a log messages at a lower level is added to say the end point has received an update.  E.g. Warning: {friendly_name} firmware update successfully completed.

kiall, 2020-02-08:
> There is no need for downloading stuff manually, zigbee2mqtt grabs the latest from the ikea servers.

What about for other brands who aren't so "open" with their firmware URLs? E.g. Philips firmwares are occasionally discovered and linked to, but there's no known equivalent to IKEAs version_info.json to base the discovery off.

Koenkk, 2020-02-08:
@kiall then we should make the version info ourselves

lorek123, 2020-02-08:
About the automatic updates: what will happen if we cut off the power during OTA update? If we risk bricking device then we should consider creating manual update procedure.

andreasbrett, 2020-02-08:
> About the automatic updates: what will happen if we cut off the power during OTA update? If we risk bricking device then we should consider creating manual update procedure.

I'd also say automatic updates should not be on by default. Users need to be able to control how and when their devices get updated. Also for automatic updates there should be some kind of time frame as devices are unavailable during the update which takes some minutes. It's the perfect situation where you might assume you're lights have crashed as they're irresponsive and then you power-toggle them to fix them. Bad decision!

sjorge, 2020-02-08:
> There is no need for downloading stuff manually, zigbee2mqtt grabs the latest from the ikea servers.

@Koenkk so for the manufacturer publishing something online we need to write a specific extension to take care of those right? At least from the code it looks like you added a special one for tradfri

So fallback that will try to parse a local dir with files that are manuId_devId.bin or something could work for those were we only sporadically get binaries for?

> In general, will it be difficult to port this to other vendors like OSRAM or Xiaomi?
Form the looks of it, not super hard if they have a nice json or other parseable index online like Tradfri does. Not sure either OSRAM or Xiaomi have those though. (At least signify (Hue) does not so that sucks)

> About the automatic updates: what will happen if we cut off the power during OTA update? If we risk bricking device then we should consider creating manual update procedure.

Same risk as when the ikea gateway will do the update I think. I have at least pulled the battery on their remotes when updates were stuck... so not sure how harmful it is.

Looking at the code, I looks still very drafty so I don't think the final result will have them randomly update. There will probably be a mqtt topic to check for updates and one to do the update. If not, I'm sure they could be added without too much hassle.

Kryzek, 2020-02-08:
Holy moly! No need to plug Trådfri gateway into network anymore to get latest firmwares. Huge kudos @Koenkk!

Koenkk, 2020-02-08:
Implemented, let's continue here: https://github.com/Koenkk/zigbee2mqtt/issues/2921

closed 2020-02-08 by Koenkk