Koenkk/zigbee-herdsman #61: please support Osram Device with modelID 'Plug Z3'

Matten-Matten, 2019-10-09:
https://github.com/hobbyquaker/node-red-contrib-zigbee/issues/42

Matten-Matten, 2019-10-09:
Moin, ich hab ein paar fragen bezüglich der Konfiguration und Bedienung.
Bei mir taucht folgender Fehler auf.
`configure failed 0xf0d1b80000125523 (LEDVANCE Plug 1) Cannot read property 'bind' of undefined`

was ist dieses 'bind' und was konfiguriere ich dort genau? Ich stehe da irgendwie aufm schlauch. 
[image]

Kannst du mir das erläutern? Danke im voraus.

Gruß
Matten Matten

Koenkk, 2019-10-09:
Please make a pr for https://github.com/Koenkk/zigbee-shepherd-converters

closed 2019-10-09 by Koenkk