Koenkk/zigbee-herdsman #63: Error after updating to Hassio v1.0

ciB89, 2019-10-11:
Hey, I updated Hassio to latest version 1.0 and now zigbee2mqtt exits with an error:
Running zigbee2mqtt-edge on latest commit.

```
zigbee2mqtt:error 2019-10-11T14:18:52: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-10-11T14:18:52: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
zigbee2mqtt:error 2019-10-11T14:18:52: Failed to start zigbee
zigbee2mqtt:error 2019-10-11T14:18:52: Exiting...
zigbee2mqtt:error 2019-10-11T14:18:52: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
```

ciB89, 2019-10-11:
Just realized its an issue with lates dev version, not hassio itself. Downgrading to an older dev-version fixed it for me.

Koenkk, 2019-10-12:
I cannot reproduce this, are you sure it isn't a connection issue and that the right path was specified?

lolorc, 2019-10-12:
also has this issue, and reverted by to previous commit
```
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error Error while parsing to ZpiObject 'TypeError: Cannot read property 'find' of undefined                                                                           │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at Function.fromUnpiFrame (/home/pi/zigbee2mqtt.dev/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/zpiObject.js:42:59)                               │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at Znp.onUnpiParsed (/home/pi/zigbee2mqtt.dev/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:70:48)                                           │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at Parser.emit (events.js:198:13)                                                                                                                           │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at Parser.parseNext (/home/pi/zigbee2mqtt.dev/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:49:26)                                       │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at Parser._transform (/home/pi/zigbee2mqtt.dev/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:35:14)                                      │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at Parser.Transform._read (_stream_transform.js:190:10)                                                                                                     │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at Parser.Transform._write (_stream_transform.js:178:12)                                                                                                    │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at doWrite (_stream_writable.js:415:12)                                                                                                                     │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at writeOrBuffer (_stream_writable.js:399:5)                                                                                                                │
Oct 12 16:50:09   zigbee-herdsman:zStack:znp:error     at Parser.Writable.write (_stream_writable.js:299:11)' +0ms                                                                                                 │
Oct 12 16:50:09   zigbee-herdsman:zStack:unpi:parser --- parseNext [] +2ms                                                                                                                                         │
Oct 12 16:50:09   serialport/stream _read reading { start: 481, toRead: 65055 } +4ms                                                                                                                               │
Oct 12 16:50:09   serialport/binding-abstract read +5ms                                                                                                                                                            │
Oct 12 16:50:09   serialport/bindings/unixRead Starting read +5ms                                                                                                                                                  │
Oct 12 16:50:09   serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms                                                                                                                   │
Oct 12 16:50:09   serialport/bindings/poller Polling for "readable" +6ms                                                                                                                                           │
Oct 12 16:50:15 zigbee2mqtt:error 2019-10-12T14:50:15: Error while starting zigbee-herdsman                                                                                                                        │
Oct 12 16:50:15 winston:file written true false                                                                                                                                                                    │
Oct 12 16:50:15 zigbee2mqtt:error 2019-10-12T14:50:15: Error: SRSP - SYS - nvLength after 6000ms                                                                                                                   │
Oct 12 16:50:15     at Timeout.object.timer.setTimeout [as _onTimeout] (/home/pi/zigbee2mqtt.dev/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)                                                        │
Oct 12 16:50:15     at ontimeout (timers.js:436:11)                                                                                                                                                                │
Oct 12 16:50:15     at tryOnTimeout (timers.js:300:5)                                                                                                                                                              │
Oct 12 16:50:15     at listOnTimeout (timers.js:263:5)                                                                                                                                                             │
Oct 12 16:50:15     at Timer.processTimers (timers.js:223:10)                                                                                                                                                      │
[... cut ...]
Oct 12 16:50:15     at listOnTimeout (timers.js:263:5)                                                                                                                                                             │
Oct 12 16:50:15     at Timer.processTimers (timers.js:223:10)                                                                                                                                                      │
Oct 12 16:50:15 winston:file written true false                                                           
```

Koenkk, 2019-10-12:
@lolorc is this on CC2652R? in that case see https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-541321343

fnxpt, 2019-10-12:
Im having the same error, just updated to the latest dev and latest firmware for CC2652R

Koenkk, 2019-10-12:
@fnxpt can you provide the logging when running with `DEBUG=* npm start` (provide it on pastebin.com and link it here).

fnxpt, 2019-10-12:
Cleaned up the docker and the volume directory and now Im getting Error: AREQ - ZDO - stateChangeInd after 60000ms

Koenkk, 2019-10-12:
Probably https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-541340010

fnxpt, 2019-10-12:
@Koenkk it worked.... strange since it was working before

ciB89, 2019-10-14:
After reinstalling zigbee2mqtt it works again

closed 2019-10-14 by ciB89