Koenkk/zigbee-herdsman #65: Implement Channel Change Command

tb-killa, 2019-10-13:
Hi @Koenkk 
i found that deconz use some Sort of Channel Change Procedure for End-Devices.
Found some Stuff on TI-Forum: https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/515585/1871970

It would be great if we would also implement this because this would allow the change of zigbee channel (automaticly) by command and hopefuly the End-Devices also move to new channel.

tb-killa, 2019-10-15:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/change_channel.cpp

sjorge, 2019-12-17:
Would this also allow for pairing device that need to be on  specific channel and then move them to the configure/prefered one?

sjorge, 2020-02-15:
Unstale, I think this would definitely be useful.

closed 2020-04-25 by stale[bot]