Koenkk/zigbee-herdsman #69: Device.lqi always returns max 3 neighbours

hobbyquaker, 2019-10-14:
Is this normal behavior or a bug?

Koenkk, 2019-10-15:
I'm not sure, can you check via sniffing what wireshark shows?

hobbyquaker, 2019-10-15:
I see the `Link Quality Response, Status: Success` for each device multiple times, some appear twice, others even 5 or 6 times. They seem to be duplicates, they all include the same neighbourtable of only 3 devices. Are these duplicates because of Zigbee Routing?

[image]

[image]
In the Responses I can see that the device really only reports 3 neighbours, so this seems not to be a bug but some (strange?) behavior of my devices?

Koenkk, 2019-10-16:
I'm not sure what is happening and ifthis is correct behaviour, but from the point of zigbee-herdsman this seems to be OK.

groenmarsmannetje, 2019-10-21:
I have a similar observation. Before I started to use the dev version with source routing firmware and less then approx 40 devices the network map would show an almost perfect mesh network with every router connected to each other after a couple of days.

Now I am on dev version with latest source routing firmware and more then 40 devices. And what I see now is that besides the controller there are 2 other nearby routers that all connect to almost all of the other devices. A lot of the other routers have max.  3 connections. This doesn’t change much when you leave the network untouched for a couple of hours or days.

I had the feeling that this started when I moved to dev version, but I cannot verify anymore because I changed too much since then (new firmware and new devices).

timota, 2019-10-31:
I made some tests to check:
- if i add only 2 devices (end-devices) to coordinator - i can see `linkquality` for them and so z2mAssistant can build proper map. 
- if i add 3-d device:
    - in logs i can see that device reports `linkquality`
`zigbee2mqtt:info  2019-10-31T19:37:38: MQTT publish: topic 'zigbee2mqtt/0x00158d000274fe84', payload '{"battery":80,"voltage":2965,"contact":false,"linkquality":51,"last_seen":1572550658817,"device":{"friendlyName":"0x00158d000274fe84","ieeeAddr":"0x00158d000274fe84","networkAddress":17964,"type":"EndDevice","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","applicationVersion":3,"stackVersion":2,"zclVersion":1,"hardwareVersion":2,"dateCode":"20161128","softwareBuildID":"3000-0001"}}'`
 
but in z2mAssistant reports all attributes except `Link quality is unknown/255.`, so it cant build map

- if i add to the same setup additional router, and after it add 3 additional end-devices. z2mAssistant reports `linkquolity` for first 2 devices (including router itself) but not for 3-d device.
- same picture if i add another router and next 3 end-devices.

for me it looks like somewhere limit for 3 device (including router/coordinator) has been set. 

i definitely didnt see this issue with Stable 1.6 and CC2531 as coordinator.

BTW: current coordinator is cc1352P2

clockbrain, 2019-10-31:
I think what's going on relates to nuances of the neighbor lqi table. Its includes a start index and count. In master the network map scan code ignores these and returns all values in the table that have a relationship type <=3 
https://github.com/Koenkk/zigbee2mqtt/blob/e26ad2aca2eeb0aa366982481eb921ee603fb6d1/lib/zigbee.js#L364

In dev this scan has been moved to zigbee-herdsman and probably correctly pays attention to the  count (and maybe should also pay attention to start index) .

https://github.com/Koenkk/zigbee-herdsman/blob/e55f57311efc167c23394edd7263fcfc89c05830/src/adapter/z-stack/znp/buffaloZnp.ts#L50

So, my assumption is that when end devices are sleeping they effectively expire out of the neighbor table by a change in the startindex and/or count, I don't know if this is generic behaviour or related to specific device firmware. I really have no idea.

timota, 2019-10-31:
> So, my assumption is that when end devices are sleeping they effectively expire out of the neighbor table by a change in the startindex and/or count,

to be honest no idea as well, but additional tests shows that even if devices are sleeping, i mean first 2, it still shows only them connected, even if i constantly pressing wakeup button on 3d one and start rescan

Koenkk, 2019-11-10:
I've found the issue. As can be seen here: https://github.com/Koenkk/zigbee-herdsman/issues/69#issuecomment-542383090. The first requests return the first 3 values, while the table holds 8 values. Zigbee-herdsman didn't request the remaining 5. Fixed, thanks all!

closed 2019-11-10 by Koenkk