Koenkk/zigbee-herdsman #72: ConBee implementation

ChrisHae, 2019-10-15:
Hello, I have noticed you started on implementation of the conbee adapter. I am from dresden elektronik and we'd like to help with the implementation. What do you think is the best way to synchronize the workflow? 

We could contribute the code for the following tasks, to start with:

- CRC calculation
- SLIP frame builder
- Query and set Zigbee parameters

... in the files driver.ts and conbeeAdapter.ts. 

Furthermore if you need any documentation or hardware to proceed just drop me a line. E.g. a ConBee II stick or the deCONZ serial line protocol documentation.

Note that this adapter not only work with the ConBee USB stick but also with the RaspBee Zigbee module for Raspberry Pi, because they share the same serial protocol. A more generic adapter name, like phoscon or deconz adapter, might be more fitting.

Koenkk, 2019-10-15:
Hi!

Could you perhaps contact me on telegram (@koenkk)?

Koenkk, 2019-10-16:
I did some updates
- Renamed conbee to deconz
- zigbee-herdsman will automatically activate the deconz adapter when it detects one
  - auto detection is also support, for zigbee2mqtt just set
```yaml
serial:
  port: null
```

felixstorm, 2019-10-30:
@Koenkk Is Conbee 2 support considered to be working as of now?

I am currently pretty displeased with the CC2531 in my environment using HA 1.2 firmware (unable to pair more than 20 *routing* devices (ubisys), loosing devices from time to time and ended up splitting my network, also tried source routing variant, but that seemed to make it even worse). And since the Zigbee side used to work fine and appeared to be very stable with deCONZ and a Conbee (1) for about 1 year before switching to zigbee2mqtt (besides incomplete support for the ubisys window covering devices :wink:), being able to use a Conbee (2) might be another option for me to try...

Thanks,
Felix

Koenkk, 2019-10-30:
@felixstorm no it's not supported yet, you might want to give the CC2652R a try though https://github.com/Koenkk/zigbee2mqtt/issues/1429 (which is fully supported in latest dev).

felixstorm, 2019-12-03:
> 
> 
> @felixstorm no it's not supported yet, you might want to give the CC2652R a try though [Koenkk/zigbee2mqtt#1429](https://github.com/Koenkk/zigbee2mqtt/issues/1429) (which is fully supported in latest dev).

Just fyi for anybody following - I switched to the CC2652R and it has been working flawlessly for a few weeks now.

BuhtigithuB, 2020-01-05:
Should this one be closed if Conbee/deconz is supported now as suggested by your answer... Sorry I have been and out of domotic and try to make progress on free time so I have been poking you on this topic here and there, sorry about that... Difficult to keep up will all the fork redirect I fall on and understand where the train is going exactly... :D

osos, 2020-01-21:
Any news, expected timeline ?

Looking forward to test this out

asdftemp, 2020-02-22:
[This](https://github.com/Koenkk/zigbee-herdsman/issues/72) popt up on my feed

Hedda, 2020-03-03:
New reddit post says that Dresden Elektronik estimated that support will be ready mid-March 2020

https://www.reddit.com/r/homeassistant/comments/f8v81p/conbee_2_support_for_zigbee2mqtt_confirmed/

https://imgur.com/a/Tda0JJO

I wonder if this will only work with the newer ConBee II (ConBee 2) adapter or if it will also work with the older first-generation ConBee and RaspBee adapters as well?

I know that both the new and the old ConBee adapters support the same serial UART protocol: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/158

manup, 2020-03-03:
It should work with ConBee I / II and RaspBee I / II since the protocol for all is the same. The difference is the device path:

* ConBee I   /dev/ttyUSBx
* ConBee II   /dev/ttyACMx
* RaspBee I / II /dev/ttyAMA0 or /dev/ttyS0

cheak1974, 2020-03-03:
Is this also working on windows? I have a conbee II plugged into my windows PC which I use for testing. It is connected as COM3 in device manager. How can I use it with zigbee2mqtt?

arjankanis, 2020-03-08:
Is it possible to test the Conbee build using the deconz branch?

Koenkk, 2020-03-14:
This has now been integrated and is available in the latest zigbee2mqtt dev branch. Many thanks @ChrisHae !

Keep in mind that this is not ready for production yet.

It seems to start fine but unfortunately I couldn't pair any devices, nothing appeared in the log after resetting a device (tried with some Xiaomi plugs, sensors and an Innr bulb), log: https://pastebin.com/gBLANJHn

ChrisHae, 2020-03-16:
I will update this over time and try to eliminate bugs, improve performance and add features.
Please note that interPAN functions likte touchlink are not implemented yet.
Right now I investigate why no devices are joining any more. This already worked but now I no longer receive any deviceAnnounce events when a device is powered on or was reset.

ChrisHae, 2020-03-16:
@Koenkk It seems devices not joining is a fw problem. Please try FW version deCONZ_ConBeeII_0x264a0700.bin
http://deconz.dresden-elektronik.de/deconz-firmware/

manup, 2020-03-16:
Another thing to improve USB startup was found today by disabling Linux MTP probes and Network Manager messing with serial interface. This might be useful for the CC2531 too.

Source: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2419#issuecomment-599576512

MTP probes and Network Manager access for ConBee II can be disabled by following udev rules:

    cat /etc/udev/rules.d/69-conbee.rules

```

# ConBee II disbale modem manager and mtp probing
ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{MTP_NO_PROBE}="1"
```

After creating this file the udev rules needs to be reloaded:

    sudo udevadm control --reload-rules

The difference should be visible in journald when plugging the USB dongle:

    sudo journald -xb -f

bwolleb, 2020-03-17:
@ChrisHae Thank you for your work on the ConBeeII support. I have a ConBeeII stick an I plan to use it with zigbee2mqtt. As mentioned, it starts correctly now on the `dev` branch but no device can join. Flashing the 0x264a0700 firmware didn't help for me.

ChrisHae, 2020-03-17:
I uploaded a fix to this problem to the deconz branch. In addition to the zdp permit join request there is now also a write parameter request that sets the permit join parameter. I created a pull request for the master branch. Then you can test if it works.

bwolleb, 2020-03-17:
I just tried zigbee2mqtt using the `deconz` branch of zigbee-herdsman and my device joined successfully!

Alex18081, 2020-03-17:
Thanks for everyone involved! For me with iobroker, Conbee II has started and devices are also connected
[image]

kirovilya, 2020-03-17:
@ChrisHae We successfully start the stick Conbee II based on zigbee-heardsman 0.12.64 in iobroker.zigbee. Thank you!

salopette, 2020-03-18:
Hi, do you have to flash the Conbee II or is it simply recognized automatically in the Zigbee2mqtt?

Alex18081, 2020-03-18:
> Hi, do you have to flash the Conbee II or is it simply recognized automatically in the Zigbee2mqtt?

Just plug it in. But I have the last firmware on it

salopette, 2020-03-18:
[image]

Alex18081, 2020-03-18:
> 
> 
> [image]

Yes

Nailik, 2020-03-19:
Not working for me yet.
I have HomeAssistant setup on Docker and use the addon (https://github.com/danielwelch/hassio-zigbee2mqtt/blob/master/CHANGELOG.md)  

Stacktrace:

`zigbee2mqtt:info  2020-03-19 12:25:19: Logging to console and directory: '/share/zigbee2mqtt/log/2020-03-19.12-25-19' filename: log.txt
zigbee2mqtt:debug 2020-03-19 12:25:19: Removing old log directory '/share/zigbee2mqtt/log/2020-03-19.12-23-47'
zigbee2mqtt:debug 2020-03-19 12:25:19: Can't load state from file /share/zigbee2mqtt/state.json (doesn't exist)
zigbee2mqtt:info  2020-03-19 12:25:19: Starting zigbee2mqtt version 1.12.0 (commit #unknown)
zigbee2mqtt:info  2020-03-19 12:25:19: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-03-19 12:25:19: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2121195-if00"}}'
zigbee2mqtt:error 2020-03-19 12:25:26: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-19 12:25:26: Failed to start zigbee
zigbee2mqtt:error 2020-03-19 12:25:26: Exiting...
zigbee2mqtt:error 2020-03-19 12:25:26: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.12.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.12.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.12.0 start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.12.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-03-19T11_25_26_613Z-debug.log
2020-03-19T12:25:27: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-03-19T12:25:27: PM2 log: App [npm:0] starting in -fork mode-
2020-03-19T12:25:27: PM2 log: App [npm:0] online
> zigbee2mqtt@1.12.0 start /zigbee2mqtt-1.12.0
> node index.js`

Alex18081, 2020-03-19:
> "serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2121195-if00"}

maybe try port /dev/ttyACM0

arjankanis, 2020-03-19:
The same problem (/dev/ttyACM0) in combination with zigbee2mqtt and the conbee 2.

Nailik, 2020-03-19:
> > "serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2121195-if00"}
> 
> maybe try port /dev/ttyACM0

I already tried it, same issue. The deconz addon https://github.com/home-assistant/hassio-addons/tree/master/deconz works with /dev/ttyACM0

kelvin5290, 2020-03-19:
> I already tried it, same issue. The deconz addon https://github.com/home-assistant/hassio-addons/tree/master/deconz works with /dev/ttyACM0

i face the same issue and look like the debug log is not full print out. ref to the adapter.js, it will determine the device is zStack or deconz. Those log is missing. and it go to start zStack by default.

Koenkk, 2020-03-19:
The detection is not working indeed, is the hassio adon running as privileged? https://www.zigbee2mqtt.io/information/docker.html#parameters-explanation

kelvin5290, 2020-03-19:
> The detection is not working indeed, is the hassio adon running as privileged? https://www.zigbee2mqtt.io/information/docker.html#parameters-explanation

yes , the addon should run as privileged.

here with the log
[image]

tafilz, 2020-03-19:
>Note that this adapter not only work with the ConBee USB stick but also with the RaspBee Zigbee module for Raspberry Pi, because they share the same serial protocol. A more generic adapter name, like phoscon or deconz adapter, might be more fitting.

The  deconz adapter should also work with the RaspBee module. The auto detection works only with USB devices.

Can we have maybe a setting like this, so it's possible to set the adapter manually to use serial only devices like the RaspBee?
```
serial:
  adapter: <z-stack|deconz>
  port: /dev/ttyS0
```

Koenkk, 2020-03-19:
@tafilz implemented, you can now specify the adapter like (https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/information/configuration.md#configuration)

```yaml
serial:
  adapter: deconz
```

Will be available in the latest hassio edge in +- 4 hours from now.

tafilz, 2020-03-19:
It seems to start now. But it doesnt seem to work right now

```
mosquitto_1    | 1584651976: mosquitto version 1.6.9 starting
mosquitto_1    | 1584651976: Config loaded from /mosquitto/config/mosquitto.conf.
mosquitto_1    | 1584651976: Opening ipv4 listen socket on port 1883.
mosquitto_1    | 1584651976: Opening ipv6 listen socket on port 1883.
zigbee2mqtt_1  | Using '/app/data' as data directory
zigbee2mqtt_1  |
zigbee2mqtt_1  | > zigbee2mqtt@1.12.0-dev start /app
zigbee2mqtt_1  | > node index.js
zigbee2mqtt_1  |
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:17: Logging to console and directory: '/app/data/log/2020-03-19.22-06-17' filename: log.txt
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:17: Starting zigbee2mqtt version 1.12.0-dev (commit #e3fafcd)      zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:17: Starting zigbee-herdsman...
zigbee2mqtt_1  | CREATED DECONZ ADAPTER
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:18: zigbee-herdsman started
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:18: Coordinator firmware version: '{"type":"RaspBee","meta":{"transportrev":0,"product":0,"majorrel":38,"minorrel":53,"maintrel":0,"revision":"0x26350500"}}'
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:19: Currently 0 devices are joined:
zigbee2mqtt_1  | zigbee2mqtt:warn  2020-03-19 22:06:19: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt_1  | zigbee2mqtt:warn  2020-03-19 22:06:19: Allowing new devices to join.
zigbee2mqtt_1  | zigbee2mqtt:warn  2020-03-19 22:06:19: Set `permit_join` to `false` once you joined all devices.      zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:19: Zigbee: allowing new devices to join.
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:20: Connecting to MQTT server at mqtt://mosquitto
mosquitto_1    | 1584651980: New connection from 172.19.0.2 on port 1883.
mosquitto_1    | 1584651980: New client connected from 172.19.0.2 as mqttjs_55b5a1d6 (p2, c1, k60).
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:20: Connected to MQTT server
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:20: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt_1  | zigbee2mqtt:info  2020-03-19 22:06:20: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.12.0-dev","commit":"e3fafcd","coordinator":{"type":"RaspBee","meta":{"transportrev":0,"product":0,"majorrel":38,"minorrel":53,"maintrel":0,"revision":"0x26350500"}},"log_level":"info","permit_join":true}'
zigbee2mqtt_1  | (node:17) UnhandledPromiseRejectionWarning: RangeError: Offset is outside the bounds of the DataView
zigbee2mqtt_1  |     at DataView.getUint8 (<anonymous>)
zigbee2mqtt_1  |     at parseReadReceivedDataResponse (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:210:27)
zigbee2mqtt_1  |     at parseFrame (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:334:32)zigbee2mqtt_1  |     at Object.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:289:63)
zigbee2mqtt_1  |     at Generator.next (<anonymous>)
zigbee2mqtt_1  |     at /app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:8:71
zigbee2mqtt_1  |     at new Promise (<anonymous>)
zigbee2mqtt_1  |     at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:4:12)   zigbee2mqtt_1  |     at Object.processFrame [as default] (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:288:12)
zigbee2mqtt_1  |     at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:37:69)
zigbee2mqtt_1  | (node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
zigbee2mqtt_1  | (node:17) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
zigbee2mqtt_1  | (node:17) UnhandledPromiseRejectionWarning: RangeError: Offset is outside the bounds of the DataView
zigbee2mqtt_1  |     at DataView.getUint8 (<anonymous>)
zigbee2mqtt_1  |     at parseReadReceivedDataResponse (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:210:27)
zigbee2mqtt_1  |     at parseFrame (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:334:32)zigbee2mqtt_1  |     at Object.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:289:63)
zigbee2mqtt_1  |     at Generator.next (<anonymous>)
zigbee2mqtt_1  |     at /app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:8:71
zigbee2mqtt_1  |     at new Promise (<anonymous>)
zigbee2mqtt_1  |     at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:4:12)   zigbee2mqtt_1  |     at Object.processFrame [as default] (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/driver/frameParser.js:288:12)
zigbee2mqtt_1  |     at Driver.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:37:69)
zigbee2mqtt_1  | (node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
```

Koenkk, 2020-03-19:
A debug log would help indeed but this probably has to be fixed by @ChrisHae (as its inside the deconz adapter)

Nailik, 2020-03-19:
How could you run it? For me the addon always removes the adapter type.

tafilz, 2020-03-19:
I'm using the docker-compose approach @Nailik

ChrisHae, 2020-03-20:
@tafilz this really looks like a bug in deCONZ adapter. I will have a look. I didn't test it with RaspBee yet

muellererwin, 2020-03-22:
I have a ConBeeII.
I'll get this error, why is it not using adapter deconz? (see configuration.yaml futher down)

zigbee2mqtt:info  2020-03-22 11:21:32: Logging to console and directory: '/app/data/log/2020-03-22.11-21-32' filename: log.txt
zigbee2mqtt:info  2020-03-22 11:21:32: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
zigbee2mqtt:info  2020-03-22 11:21:32: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-03-22 11:21:40: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-22 11:21:40: Failed to start zigbee
zigbee2mqtt:error 2020-03-22 11:21:40: Exiting...
zigbee2mqtt:error 2020-03-22 11:21:40: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)

configuration.yaml
...
# Serial settings
serial:
  adapter: deconz
  port: /dev/ttyACM0

msnelling, 2020-03-22:
> I have a ConBeeII.
> I'll get this error, why is it not using adapter deconz? (see configuration.yaml futher down)
> 
> zigbee2mqtt:info 2020-03-22 11:21:32: Logging to console and directory: '/app/data/log/2020-03-22.11-21-32' filename: log.txt
> zigbee2mqtt:info 2020-03-22 11:21:32: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
> zigbee2mqtt:info 2020-03-22 11:21:32: Starting zigbee-herdsman...
> zigbee2mqtt:error 2020-03-22 11:21:40: Error while starting zigbee-herdsman
> zigbee2mqtt:error 2020-03-22 11:21:40: Failed to start zigbee
> zigbee2mqtt:error 2020-03-22 11:21:40: Exiting...
> zigbee2mqtt:error 2020-03-22 11:21:40: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
> at ZStackAdapter. (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
> at Generator.throw ()
> at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
> 

@muellererwin  I was getting this error too until I added the following to my `docker-compose.yaml`
```
version: "3.5"
services:
  z2m:
    image: koenkk/zigbee2mqtt:latest
    devices:
      - /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2120741-if00:/dev/ttyACM0
    volumes:
      - /run/udev:/run/udev:ro
    environment:
      - TZ=Europe/London
    privileged: true
```
Node the `privileged: true` and the `/run/udev:/run/udev:ro` volume mapping.

tafilz, 2020-03-22:
> 
> 
> I have a ConBeeII.
> I'll get this error, why is it not using adapter deconz? (see configuration.yaml futher down)
> 
> zigbee2mqtt:info 2020-03-22 11:21:32: Logging to console and directory: '/app/data/log/2020-03-22.11-21-32' filename: log.txt
> zigbee2mqtt:info 2020-03-22 11:21:32: Starting zigbee2mqtt version 1.12.0 (commit #840b9d9)
> zigbee2mqtt:info 2020-03-22 11:21:32: Starting zigbee-herdsman...
> zigbee2mqtt:error 2020-03-22 11:21:40: Error while starting zigbee-herdsman
> zigbee2mqtt:error 2020-03-22 11:21:40: Failed to start zigbee
> zigbee2mqtt:error 2020-03-22 11:21:40: Exiting...
> zigbee2mqtt:error 2020-03-22 11:21:40: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
> at ZStackAdapter. (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
> at Generator.throw ()
> at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
> 
> configuration.yaml
> ...
> # Serial settings
> 
> serial:
> adapter: deconz
> port: /dev/ttyACM0

I think the npm version doesn't have the setting yet. 
Your version is running on commit 840b9d9 and the setting was intruduced with commit 6b9baf0.
Are you using the dev-branch of zigbee2mqtt? There is should work.

Also the ConBee stick should be able to get auto detected.
 
>  Path auto detection is also support, for zigbee2mqtt just set
> 
> 
> ```yaml
> serial:
>   port: null
> ```

muellererwin, 2020-03-22:
@msnelling: Thanks for your tips. Up and running now!

ChrisHae, 2020-03-23:
I uploaded a fix for RaspBee. It now uses 16bit NWK addresses that will solve some problems. I think it will be available soon.

muellererwin, 2020-03-24:
My setup:
I am running zigbee2mqtt in a docker container on a raspberry pi.
WIth your help the interaction with a directly connected ConBeeII-USB-device worked.

Now I tried to have the ConBeeII on a remote raspberry pi published by ser2net and from a zigbee2mqtt docker to connect it via serial: adapter: deconz port: 'tcp://ip:port'.
This failed. Is this in general not possible with the current status of ConBeeII integration?
Are there any chances to integrate it? Would be great!

kelvin5290, 2020-03-24:
Hi all, my conbee2 can successfully control the device, but after a day look like having some problems and I can’t control the device anymore.
The log: 

``` herdsman:deconz:driver Timeout for aps request CMD: 0x12 seq: 173
2020-03-24T15:42:45.299Z zigbee-herdsman:deconz:adapter PERMIT_JOIN FAILED - APS TIMEOUT
(node:236) UnhandledPromiseRejectionWarning: undefined
(node:236) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 894)
2020-03-24T15:42:50.502Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0x7 seqNr: 32
2020-03-24T15:42:50.671Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 33
2020-03-24T15:43:00.718Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0x7 seqNr: 33
2020-03-24T15:43:00.884Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 34
2020-03-24T15:43:10.936Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0x7 seqNr: 34
2020-03-24T15:43:11.092Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 35
2020-03-24T15:43:21.162Z zigbee-herdsman:deconz:driver Timeout for request - CMD: 0x7 seqNr: 35
2020-03-24T15:43:21.304Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 36
 ```

arjankanis, 2020-03-24:
Same for me.  Crashes after a few hours of running. I'm using the latest Hass with zigbee2mqtt addon and Conbee II.

zigbee2mqtt:info  2020-03-24 20:47:44: MQTT publish: topic 'zigbee2mqtt/0x00158d00024ee1c8', payload '{"battery":100,"voltage":3015,"illuminance":8,"linkquality":50,"occupancy":false,"illuminance_lux":8}'
(node:234) UnhandledPromiseRejectionWarning: undefined
(node:234) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 26)

OXOTH1K, 2020-03-24:
Not work for me with this options:

```
serial:
  adapter: deconz
  port: /dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_DM0103RY-if00-port0
```

Log file:

```
zigbee2mqtt:error 2020-03-24 23:22:35: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-03-24 23:22:35: Failed to start zigbee
zigbee2mqtt:error 2020-03-24 23:22:35: Exiting...
zigbee2mqtt:error 2020-03-24 23:22:35: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
```

arjankanis, 2020-03-24:
> I'm use conbee 1.
> Not work for me with this options:
> 
> ```
> serial:
>   adapter: deconz
>   port: /dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_DM0103RY-if00-port0
> ```
> 
> Log file:
> 
> ```
> zigbee2mqtt:error 2020-03-24 23:22:35: Error while starting zigbee-herdsman
> zigbee2mqtt:error 2020-03-24 23:22:35: Failed to start zigbee
> zigbee2mqtt:error 2020-03-24 23:22:35: Exiting...
> zigbee2mqtt:error 2020-03-24 23:22:35: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
>     at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:71:23)
>     at Generator.throw (<anonymous>)
>     at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
> ```

The addon from danielwelch isn't ready for the serial addition. The config.json must be modified to accept the serial addition. For now can try my fork of zigbee2mqtt-edge; https://github.com/arjankanis/hassio-zigbee2mqtt

OXOTH1K, 2020-03-24:
> The addon from danielwelch isn't ready for the serial addition. The config.json must be modified to accept the serial addition. For now can try my fork of zigbee2mqtt-edge; [...]

I was already use this fork. Same trouble on it.

OXOTH1K, 2020-03-24:
> The addon from danielwelch isn't ready for the serial addition. The config.json must be modified to accept the serial addition. For now can try my fork of zigbee2mqtt-edge; [...]

Your addon was started, but not configured devices. For example Konke temperature sensor:

```
zigbee2mqtt:info  2020-03-25 02:40:38: Configuring '0x086bd7fffee976be'
zigbee2mqtt:error 2020-03-25 02:41:40: Failed to configure '0x086bd7fffee976be', attempt 1 (Error: Bind 0x086bd7fffee976be/1 msTemperatureMeasurement from '0x00212effff03b4b2/1' failed (undefined)
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

Nailik, 2020-03-25:
For me pairing worked.

Sometimes i have seen this error
```
(node:234) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'stack' of undefined
    at Controller.callExtensionMethod (/app/lib/controller.js:349:103)
(node:234) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
(node:234) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
(node:234) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'stack' of undefined
    at Controller.callExtensionMethod (/app/lib/controller.js:349:103)
(node:234) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 4)
```

And light dimming is not working at all. On and Off is working (On -> light goes 100% on) but dimming has no effekt. E.g when the Light is off and i set it to 5% i see mqtt says light on and 5% but it's not turning on. When i set the light to 0% it's going off.

ChrisHae, 2020-03-26:
ConBee II loses connection after some time is a known problem. I investigate it right now. Until now you have to restart zigbee2mqtt to make ConBeeII work again. Perhaps there is a Firmware update in the near future that will improve stability. 

Then there is also this tweak that @manup posted some days ago that should also improve stability:

> Another thing to improve USB startup was found today by disabling Linux MTP probes and Network Manager messing with serial interface. This might be useful for the CC2531 too.
> 
> Source: [dresden-elektronik/deconz-rest-plugin#2419 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2419#issuecomment-599576512)
> 
> MTP probes and Network Manager access for ConBee II can be disabled by following udev rules:
> 
> ```
> cat /etc/udev/rules.d/69-conbee.rules
> ```
> 
> ```
> 
> # ConBee II disbale modem manager and mtp probing
> ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{ID_MM_DEVICE_IGNORE}="1"
> ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{MTP_NO_PROBE}="1"
> ```
> 
> After creating this file the udev rules needs to be reloaded:
> 
> ```
> sudo udevadm control --reload-rules
> ```
> 
> The difference should be visible in journald when plugging the USB dongle:
> 
> ```
> sudo journald -xb -f
> ```

ChrisHae, 2020-03-26:
> For me pairing worked.
> 
> Sometimes i have seen this error
> 
> ```
> (node:234) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'stack' of undefined
>     at Controller.callExtensionMethod (/app/lib/controller.js:349:103)
> (node:234) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
> (node:234) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
> (node:234) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'stack' of undefined
>     at Controller.callExtensionMethod (/app/lib/controller.js:349:103)
> (node:234) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 4)
> ```
> 
> And light dimming is not working at all. On and Off is working (On -> light goes 100% on) but dimming has no effekt. E.g when the Light is off and i set it to 5% i see mqtt says light on and 5% but it's not turning on. When i set the light to 0% it's going off.
> 
> (Innr light bulbs and some dimmer i don't know the name. When i use deconz without mqtt the dimming works)

I will test this next. What Client/GUI are you using to control the lights ?

Nailik, 2020-03-26:
> I will test this next. What Client/GUI are you using to control the lights ?

I am using HomeAssistant with the fork of the addon mentioned above (https://github.com/arjankanis/hassio-zigbee2mqtt)

In the MQtt logs the brightness looks good so i assume this is not a gui problem (or i hope)

Koenkk, 2020-03-27:
@ChrisHae I see that the waitFor method has not been implemented yet (https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/deconz/adapter/deconzAdapter.ts#L480). This is required for e.g. OTA updates. Could this be implemented? (original issue: https://github.com/Koenkk/zigbee2mqtt/issues/3210)

kelvin5290, 2020-03-30:
HI all, may i know the channel of conbee ii is following the configuration.yaml or it is a constant

i am trying to sniff the packet but cannot sniff any packet on certain channel.

ChrisHae, 2020-03-31:
I uploaded a fix to for dimming and change the color of lights.
@Koenkk ok I will implement the waitFor function next.

ChrisHae, 2020-03-31:
@kelvin5290 yes it seems the configuration from configuration.yaml will not be written to the ConBee2 device. I will have a look and add this feature in the next few days.
If you are interested what the current channel is look for a zigbee2mqtt debug message right after you start zigbee2mqtt. It looks like this:
`zigbee2mqtt:debug 2020-03-31 17:02:04: Zigbee network parameters: "panID":49515,"extendedPanID":"0x00212effff03d49a","channel":15}`

kelvin5290, 2020-04-01:
@ChrisHae Thank you for reply. 
also, i face another issue. When i going to pairing a in wall switch. The configuration of reporting packet look like contain some problem. The below image shown the information. Thank you.
[image]

[458 more replies]

closed 2022-01-09 by github-actions[bot]