Koenkk/zigbee-herdsman #74: Unknown ZDO command (200)

kirovilya, 2019-10-16:
Sometimes when I enable pairing, I get a strange command (with code 200). Parser falls:
```
2019-10-16T18:37:05.609Z zigbee-herdsman:controller:log Permit joining
2019-10-16T18:37:05.610Z zigbee-herdsman:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2019-10-16T18:37:05.611Z zigbee-herdsman:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2019-10-16T18:37:09.570Z zigbee-herdsman:zStack:unpi:parser <-- [254,1,101,54,0,82]
2019-10-16T18:37:09.570Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
2019-10-16T18:37:09.570Z zigbee-herdsman:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [] - 82
2019-10-16T18:37:09.570Z zigbee-herdsman:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2019-10-16T18:37:09.570Z zigbee-herdsman:zStack:unpi:parser --- parseNext []
2019-10-16T18:37:09.579Z zigbee-herdsman:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
2019-10-16T18:37:09.579Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2019-10-16T18:37:09.579Z zigbee-herdsman:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240
2019-10-16T18:37:09.579Z zigbee-herdsman:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2019-10-16T18:37:09.579Z zigbee-herdsman:zStack:unpi:parser --- parseNext []
2019-10-16T18:37:09.742Z zigbee-herdsman:zStack:unpi:parser <-- [254,11,69,200,0,0,30,116,232,24,0,75,18,0,2,71]
2019-10-16T18:37:09.743Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,11,69,200,0,0,30,116,232,24,0,75,18,0,2,71]
2019-10-16T18:37:09.743Z zigbee-herdsman:zStack:unpi:parser --> parsed 11 - 2 - 5 - 200 - [t�K] - 71
2019-10-16T18:37:09.744Z zigbee-herdsman:zStack:znp:error Error while parsing to ZpiObject 'Error: CommandID '200' from subsystem '5' not found
    at Function.fromUnpiFrame (c:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\adapter\z-stack\znp\zpiObject.js:44:19)
    at Znp.onUnpiParsed (c:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\adapter\z-stack\znp\znp.js:74:48)
    at emitOne (events.js:116:13)
    at Parser.emit (events.js:211:7)
    at Parser.parseNext (c:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\adapter\z-stack\unpi\parser.js:49:26)
    at Parser._transform (c:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\adapter\z-stack\unpi\parser.js:35:14)
    at Parser.Transform._read (_stream_transform.js:186:10)
    at Parser.Transform._write (_stream_transform.js:174:12)
    at doWrite (_stream_writable.js:396:12)
    at writeOrBuffer (_stream_writable.js:382:5)'
2019-10-16T18:37:09.744Z zigbee-herdsman:zStack:unpi:parser --- parseNext []
2019-10-16T18:38:18.516Z zigbee-herdsman:controller:log Disable joining
2019-10-16T18:38:18.518Z zigbee-herdsman:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":0,"tcsignificance":0}
2019-10-16T18:38:18.520Z zigbee-herdsman:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,0,0,26]
2019-10-16T18:38:18.542Z zigbee-herdsman:zStack:unpi:parser <-- [254,1,101,54,0,82]
2019-10-16T18:38:18.542Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
2019-10-16T18:38:18.543Z zigbee-herdsman:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [] - 82
2019-10-16T18:38:18.543Z zigbee-herdsman:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2019-10-16T18:38:18.543Z zigbee-herdsman:zStack:unpi:parser --- parseNext []
2019-10-16T18:38:18.546Z zigbee-herdsman:zStack:unpi:parser <-- [254,1,69,203,0,143,254,3,69,182,0,0,0,240]
2019-10-16T18:38:18.547Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,1,69,203,0,143,254,3,69,182,0,0,0,240]
2019-10-16T18:38:18.547Z zigbee-herdsman:zStack:unpi:parser --> parsed 1 - 2 - 5 - 203 - [] - 143
2019-10-16T18:38:18.547Z zigbee-herdsman:zStack:znp:AREQ <-- ZDO - permitJoinInd - {"duration":0}
2019-10-16T18:38:18.548Z zigbee-herdsman:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2019-10-16T18:38:18.548Z zigbee-herdsman:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240
2019-10-16T18:38:18.548Z zigbee-herdsman:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2019-10-16T18:38:18.548Z zigbee-herdsman:zStack:unpi:parser --- parseNext []
```

Koenkk, 2019-10-16:
which command is this?

kirovilya, 2019-10-16:
@Koenkk I do not know :) I can not understand by the log.
perhaps this is the coordinator himself reporting.

kirovilya, 2019-10-16:
I had a situation when the devices did not want to pair. the error raised.
changed the channel (from 15 to 20) - the error disappeared, the devices began to pair.
Then, no more this error.

but the situation is strange. this error came out constantly

Koenkk, 2019-10-16:
Command is not document in z-stack documentation, found definition in the source code but I'm not sure what it means/does.

kirovilya, 2019-10-17:
Thank you. I think these are some zigbee 3.0 devices trying to start pairing and send a signal ZDO_CONCENTRATOR_IND_CBID
I'll see if the error repeats after completion

closed 2019-10-17 by kirovilya