Koenkk/zigbee-herdsman #75: port doesn't accept symlink.

lolorc, 2019-10-16:


lolorc, 2019-10-16:
I used to run two instances of z2m with 2 coordinators on the same computer, autodetection won't help in that case

Koenkk, 2019-10-17:
Can you add on https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts#L46

```ts
serialPortOptions.path = fs.realpathSync(serialPortOptions.path);
```

and add to https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts#L6

```ts
import fs from 'fs';
```

lolorc, 2019-10-17:
@Koenkk yep that's ok.
I've been to lazy, I've edited the js inside node_modules :)
```
Oct 17 10:25:42 winston:file logged 514 debug 2019-10-17T08:25:42: Using zigbee-herdsman with settings: '{"network":{"panID":"HIDDEN","extenedPanID":"HIDDEN","channelList":[],"networkKey":"HIDDEN"},"databasePath":"/home/pi/zigbee2mqtt.data/database.db","backupPath":"/home/pi/zigbee2mqtt.data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB.CC2652"}}'
Oct 17 10:25:42 
Oct 17 10:25:43   zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms

Koenkk, 2019-10-17:
fixed, thanks!

closed 2019-10-17 by Koenkk