Koenkk/zigbee-herdsman #77: Docker device mapping issue

plasticrake, 2019-10-18:
I switched my `zigbee2mqtt` docker container from `latest` to `latest-dev`. On startup I was getting:

```
> zigbee2mqtt@1.6.0 start /app
> node index.js

zigbee2mqtt:error 2019-10-18T22:06:41: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-10-18T22:06:41: Failed to start zigbee
zigbee2mqtt:error 2019-10-18T22:06:41: Exiting...
zigbee2mqtt:error 2019-10-18T22:06:41: Error: Path '/dev/ttyACM0' does not exist
    at Object.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/serialPortUtils.js:34:19)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/serialPortUtils.js:5:58)
    at process._tickCallback (internal/process/next_tick.js:68:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
```

I added this in `serialPortUtils.js` near the issue to get some debug info:

```js
        let devices = await SerialPort.list()
        console.warn(devices);
        const child_process = require('child_process');
        console.warn(child_process.execSync('ls /dev/tty*').toString('utf8'));
```

which output:

```js
[
  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyUSB0'
  },
  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyUSB1'
  },
  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyUSB2'
  },
  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyS0'
  },
  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyS1'
  },
  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyS2'
  },
  { manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined,
    path: '/dev/ttyS3'
  }
]

/dev/tty
/dev/ttyACM0
```

Note that all the devices returned by `serialport.list()` don't appear to match what is in `/dev/`. I tried changing` /dev/ttyACM0` to `/dev/ttyUSB0` in both my docker config and z2m config and that fixed it! However this still seems like an issue that would affect other users not using one of these particular device names.

Probably related to serialport/node-serialport#1944

For reference my original docker-compose was:
```yaml
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:latest-dev
    depends_on:
    - mosquitto
    volumes:
    - ./zigbee2mqtt:/app/data
    devices:
    - /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0:/dev/ttyACM0
    restart: unless-stopped
    network_mode: host
```

Koenkk, 2019-10-18:
Does it work when adding `privilleged: true` ?

plasticrake, 2019-10-18:
With `ttyACM0` and `privileged: true` I get a slightly different error:

```
zigbee2mqtt:error 2019-10-18T23:49:46: Error: ENOENT: no such file or directory, lstat '/dev/ttyACM0'
    at Object.realpathSync (fs.js:1461:7)
    at Object.realpathSync [as RealpathSync] (/app/node_modules/zigbee-herdsman/dist/utils/realpathSync.js:10:25)
    at Function.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/adapter.js:56:50)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/adapter.js:5:58)
```
the device list still looks identical to before.

With `ttyUSB0` and `privileged: true` I get past that, but a different error:

```
zigbee2mqtt:error 2019-10-18T23:44:39: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-10-18T23:44:39: Failed to start zigbee
zigbee2mqtt:error 2019-10-18T23:44:39: Exiting...
zigbee2mqtt:error 2019-10-18T23:44:39: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm ERR! Exit status 1
```

Koenkk, 2019-10-19:
As you have the same issue as https://github.com/Koenkk/zigbee2mqtt/issues/2157, let's continue there.

closed 2019-10-19 by Koenkk