Koenkk/zigbee-herdsman #78: How configure serialport path on Windows?

kirovilya, 2019-10-24:
Previously, under Windows, you could specify a com port by name, for example, "COM57". Now it doesn't work.
```
Error: ENOENT: no such file or directory, lstat 'C:\Projects\ioBroker\node_modules\iobroker\COM57'
    at Object.realpathSync (fs.js:1657:15)
    at Object.realpathSync [as RealpathSync] (C:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\utils\realpathSync.js:10:25)
    at Function.<anonymous> (C:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\adapter\adapter.js:56:50)
    at Generator.next (<anonymous>)
    at fulfilled (C:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\adapter\adapter.js:5:58)
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:189:7)
```

Trying to change to '\\.\COM57\' like describe here https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/information/windows.md
```
ENOTSUP: operation not supported on socket, lstat '\\.\COM57\'
    at Object.realpathSync (fs.js:1617:13)
    at Object.realpathSync [as RealpathSync] (C:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\utils\realpathSync.js:10:25)
    at Function.<anonymous> (C:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\adapter\adapter.js:56:50)
    at Generator.next (<anonymous>)
    at fulfilled (C:\Projects\ioBroker\node_modules\iobroker.zigbee\node_modules\zigbee-herdsman\dist\adapter\adapter.js:5:58)
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:189:7)
```

Koenkk, 2019-10-24:
Is this with herdsman 0.11.1?

kirovilya, 2019-10-25:
0.11.0
I think we need to check platform before apply "realpath"
```serialPortOptions.path = RealpathSync(serialPortOptions.path);```

Koenkk, 2019-10-25:
Can you try with 0.11.1, that should fix it

kirovilya, 2019-10-25:
I looked at the commit - yes, it will work, but it is very strange, it is easier to check the platform :)

Koenkk, 2019-10-25:
that may be nicer, but still many things can go wrong, even on linux in docker the `serialport.list` won't always work. This just acts as a fallback mechanism.

closed 2019-10-27 by kirovilya