Koenkk/zigbee-herdsman #81: Allow permit join to router

kirovilya, 2019-10-29:
In the previous version of the "shepherd" it was possible to start pairing on a specific router.
I took this opportunity.
https://github.com/kirovilya/zigbee-shepherd/blob/master/lib/components/controller.js#L395

By transmitting the network address of the router, you could start pairing on a specific device. Now this opportunity has disappeared.

I propose to return this opportunity :)

tb-killa, 2019-11-02:
For this we need to rewrite permit function on https://github.com/Koenkk/zigbee-herdsman/blob/5098984f088edfb76972e22c2b1d075079c048bf/src/adapter/z-stack/adapter/zStackAdapter.ts#L142
For first test you could add data manually and report if test passed :)

alex04103, 2019-12-20:
Just wanted to report that I had a range issue and the bulb with this issue was not automatically using the repeater, which has better signal strength, but stayed connected to the coordinator.

I modified the `zStackAdapter.ts` as follows and pairing to the repeater then was successful without any issue:

```
    permitJoin(seconds) {
        return __awaiter(this, void 0, void 0, function* () {
            yield this.queue.execute(() => __awaiter(this, void 0, void 0, function* () {
//                const payload = { addrmode: 0x0F, dstaddr: 0xFFFC, duration: seconds, tcsignificance: 0 };
                const payload = { addrmode: 0x02, dstaddr: 0x52b5, duration: seconds, tcsignificance: 0 };
                yield this.znp.request(Subsystem.ZDO, 'mgmtPermitJoinReq', payload);
            }));
        });
    }
```

kirovilya, 2020-02-16:
Done https://github.com/Koenkk/zigbee-herdsman/commit/0015e9cbb1056e90937e75701c7d5d1bdbc7fc93

Thank you

closed 2020-02-16 by kirovilya

qm3ster, 2020-02-16:
Isn't this https://github.com/Koenkk/zigbee-herdsman/commit/0015e9cbb1056e90937e75701c7d5d1bdbc7fc93