Koenkk/zigbee-herdsman #82: Bad interview of TERNCY-PP01

kirovilya, 2019-11-02:
Need help analyzing a failed interview with the new TERNCY PP01 device.
[bad_interview.log](https://github.com/Koenkk/zigbee-herdsman/files/3800872/bad_interview.log)

First of all, I added a new endpoint 110 for the coordinator (in startZnp.js), as this is where some commands from the device come.
Second, I tried to add a 'checkin' command in the events.js file, but not handle. Yes, I seen not implemented issue https://github.com/Koenkk/zigbee-herdsman/issues/73

But I don’t understand why the interview is interrupted ... maybe the device did not wait for an answer to its checkin request ...
Even after such an unsuccessful interview, messages from the device come correctly. But the fact that the interview was not completed. Any ideas?

tb-killa, 2019-11-02:
Hi maybe your don't do a default Response and need this tweak too?

https://github.com/Koenkk/zigbee-herdsman/commit/2de4d56521d4f5d88026c999886b367ef895d8bc

kirovilya, 2019-11-03:
@tb-killa execution does not reach 'ssIasZone enrolling'. falls on receiving active endpoints https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L311
```Interview failed for '0x000d6f00106a64e8 with error 'Error: AREQ - ZDO - activeEpRsp after 10000ms'```

kirovilya, 2019-11-03:
very strange ... yesterday I mated several times and there has always been a failure ...
Today I performed pairing through the debugger and it turned out to be successful ... maybe some pauses (stops in the debugger) did the trick, because in the normal pairing mode I again failed.
[good_interview.log](https://github.com/Koenkk/zigbee-herdsman/files/3801262/good_interview.log)

kirovilya, 2019-11-03:
Sometimes it fall with ```Interview failed for '0x000d6f00106a64e8 with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms'``` on line https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L319
:((

Koenkk, 2019-11-03:
I guess that while pairing the device falls asleep. Can you sniff the traffic while pairing it? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

kirovilya, 2019-11-03:
@Koenkk yes, i can. but pairing will be only from 2531, there is no native gateway

Koenkk, 2019-11-03:
Yes thats OK

kirovilya, 2019-11-03:
two pairing processes
[pp01.zip](https://github.com/Koenkk/zigbee-herdsman/files/3801366/pp01.zip)
pp01-1 ended with Error: AREQ - ZDO - simpleDescRsp after 10000ms
pp01-2 ended with Error: AREQ - ZDO - activeEpRsp after 10000ms

Koenkk, 2019-11-03:
It seems that the device also reports it's model ID just after pairing (pp01-1.pcapng no. 36). Add this here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L280

```js
if (this.modelID && this.modelID.startsWith('TERNCY-PP01')) {
    debug('Interview procedure failed but got modelID starting with tenrncy, assuming Terncy end device');
    this._type = 'EndDevice';
    this._manufacturerID = 4648;
    this._manufacturerName = 'TERNCY';
    this._powerSource = 'Battery';
    this._interviewing = false;
    this._interviewCompleted = true;
    this.save();
}
```

kirovilya, 2019-11-03:
With this code, the interview goes well :) but aren't we missing something?

Here is another example of an unsuccessful interview - now this device is ORVIBO, but it was not possible to get the model.
[orvibo-1.zip](https://github.com/Koenkk/zigbee-herdsman/files/3802050/orvibo-1.zip)

This trend will lead us to an early description of devices along with their properties, clusters and endpoints - fingerprints. Here is an example implementation in Smarttings:
https://github.com/monninn82/SmartThingsPublic-Backup/blob/master/devicetypes/smartthings/zigbee-scene-keypad.src/zigbee-scene-keypad.groovy#L31

Koenkk, 2019-11-03:
Something else is going on here. 

- In the active endpoint request it responds with endpoint 0 and 1
- However according to the ZCL endpoint 0 is an invalid ID and shouldn't be used
- zigbee-herdsman than request the simple descriptor of endpoint 0, which of course fails because this is an invalid endpoint. 
- As endpoint 0 should never be used, we can ignore this safely, change

https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/device.ts#L312

to 

```
this._endpoints = activeEndpoints.endpoints.map((e): Endpoint => {
            return Endpoint.create(e, undefined, undefined, [], [], this.networkAddress, this.ieeeAddr);
        }).filter((e) => e.ID !== 0);
```

kirovilya, 2019-11-03:
Now I got it model '3c4e4fc81ed442efaf69353effcdfc5f' :)) Thank you.
But it does not coincide with that of smarttings :))))

Now I can make converters and PR.

closed 2019-11-30 by kirovilya