Koenkk/zigbee-herdsman #83: Cant get Xiaomi Motion to reconnect

ksuclipse, 2019-11-03:
Motion sensors disconnected tried to repair but for some reason keep getting zigbee-herdsman:controller:log Not interviewing '0x00158d00035ffa60', completed 'true', in progress 'false'
I removed all occurances of the device in database.db and configuration

here is the relevant logs:

`2019-11-03T05:12:18.705Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,69,203,254,113,254,3,69,182,0,0,0,240]
2019-11-03T05:12:18.705Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,69,203,254,113,254,3,69,182,0,0,0,240]
2019-11-03T05:12:18.705Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 2 - 5 - 203 - [�] - 113
2019-11-03T05:12:18.705Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - permitJoinInd - {"duration":254}
2019-11-03T05:12:18.705Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2019-11-03T05:12:18.706Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [] - 240
2019-11-03T05:12:18.706Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2019-11-03T05:12:18.706Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-03T05:12:55.077Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,110,129,0,109]
2019-11-03T05:12:55.077Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,110,129,0,109]
2019-11-03T05:12:55.077Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [n�] - 109
2019-11-03T05:12:55.077Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":33134,"relaycount":0,"relaylist":[]}
2019-11-03T05:12:55.078Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-03T05:12:55.122Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,110,129,0,109]
2019-11-03T05:12:55.122Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,110,129,0,109]
2019-11-03T05:12:55.122Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [n�] - 109
2019-11-03T05:12:55.123Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":33134,"relaycount":0,"relaylist":[]}
2019-11-03T05:12:55.123Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-03T05:12:55.199Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,202,92,96,250,95,3,0,141,21,0,110,129,164]
2019-11-03T05:12:55.199Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,202,92,96,250,95,3,0,141,21,0,110,129,164]
2019-11-03T05:12:55.200Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [�\`�_�n�] - 164
2019-11-03T05:12:55.200Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":23754,"extaddr":"0x00158d00035ffa60","parentaddr":33134}
2019-11-03T05:12:55.202Z zigbee-herdsman:controller:log Device '0x00158d00035ffa60' joined
2019-11-03T05:12:55.202Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-03T05:12:55.203Z zigbee-herdsman:controller:log Device '0x00158d00035ffa60' accepted by handler
2019-11-03T05:12:55.203Z zigbee-herdsman:controller:log Not interviewing '0x00158d00035ffa60', completed 'true', in progress 'false'
2019-11-03T05:15:38.656Z zigbee-herdsman:controller:log Permit joining`

Koenkk, 2019-11-03:
The device did actually reconnect: Device '0x00158d00035ffa60' joined. But zigbee-herdsman doesnt interview it again because it already did.

ksuclipse, 2019-11-03:
That’s what I thought but I’m not getting any mqtt updates and when I hit the button on the motion sensor it does the 3 flashes instead of two which supposedly means it can’t connect to the controller.

Koenkk, 2019-11-03:
- Did it work before?
- Could you sniff the traffic when motion is detected so we can check if the sensor is broken or it is a zigbee2mqtt issue (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html).

closed 2020-01-09 by stale[bot]