Koenkk/zigbee-herdsman #84: Unknown cluster 23

kirovilya, 2019-11-03:
Got unknown cluster 23 with device ORVIBO CR11S8UZ:
```
2019-11-03T06:48:38.964Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,23,0,123,200,1,1,0,102,0,52,130,107,0,0,6,25,0,8,7,0,0,123,200,29,126]
2019-11-03T06:48:38.965Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,23,0,123,200,1,1,0,102,0,52,130,107,0,0,6,25,0,8,7,0,0,123,200,29,126]
2019-11-03T06:48:38.966Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [{�f4�k{�] - 126
2019-11-03T06:48:38.969Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":23,"srcaddr":51323,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":7045684,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[25,0,8,7,0,0]}}
2019-11-03T06:48:38.972Z zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":23,"data":{"type":"Buffer","data":[25,0,8,7,0,0]},"networkAddress":51323,"endpoint":1,"linkquality":102,"groupID":0}'
2019-11-03T06:48:38.976Z zigbee-herdsman:controller:error Error while retrieving cluster for raw 'Error: Cluster with key '23' does not exist'
2019-11-03T06:48:38.985Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```
Can't find description of this cluster...

tb-killa, 2019-11-03:
Hi @kirovilya is this your mentioned device?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1666

kirovilya, 2019-11-03:
@tb-killa yes

kirovilya, 2019-11-03:
some handlers for this cluster and device https://github.com/monninn82/SmartThingsPublic-Backup/blob/master/devicetypes/smartthings/zigbee-scene-keypad.src/zigbee-scene-keypad.groovy#L61

kirovilya, 2019-11-03:
I think we can use dummy-cluster for this case:
```
    cluster23: {
        ID: 23,
        attributes: {},
        commands: {},
        commandsResponse: {},
    },
```

Koenkk, 2019-11-03:
Yes, but also commands and attributes have to be added otherwise it has no use.

kirovilya, 2019-11-04:
This is only necessary to eliminate parsing errors. We already have similar clusters without attributes and commands https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl/definition/cluster.ts#L3457

Processing data can be handled like a buffer.

Koenkk, 2019-11-04:
So the idea is to parse the raw buffer in zigbee-herdsman-converters?

kirovilya, 2019-11-04:
@Koenkk yes
```
orvibo_raw: {
        cluster: 'cluster23',
        type: 'raw',
        convert: (model, msg, publish, options) => {
            // 25,0,8,3,0,0 - click btn 1
            // 25,0,8,3,0,2 - hold btn 1
            // 25,0,8,3,0,3 - release btn 1
            // 25,0,8,11,0,0 - click btn 2
            // 25,0,8,11,0,2 - hold btn 2
            // 25,0,8,11,0,3 - release btn 2
            // 25,0,8,7,0,0 - click btn 3
            // 25,0,8,7,0,2 - hold btn 3
            // 25,0,8,7,0,3 - release btn 3
            // 25,0,8,15,0,0 - click btn 4
            // 25,0,8,15,0,2 - hold btn 4
            // 25,0,8,15,0,3 - release btn 4
            // TODO: do not know how to get to use 5,6,7,8 buttons
            const btn_lookup = {
                     3: 'btn_1',
                    11: 'btn_2',
                     7: 'btn_3',
                    15: 'btn_4',  
                  },
                  act_lookup = {
                     0: 'click',
                     2: 'hold',
                     3: 'release',
                  },
                  button = btn_lookup[msg.data[3]],
                  action = act_lookup[msg.data[5]];
            if (button && action) {
                return {[button]: action};
            }
        }, 
    },
```

Koenkk, 2019-11-05:
Instead of having to define an empty cluster, zigbee-herdsman will now expose the cluster ID instead (as a number). So adding it to `clusters.ts` is not needed anymore. 

Your converter should then use a number instead of string:

```js
orvibo_raw: {
        cluster: 23,
        type: 'raw',
        convert: (model, msg, publish, options) => {
            // 25,0,8,3,0,0 - click btn 1
            // 25,0,8,3,0,2 - hold btn 1
            // 25,0,8,3,0,3 - release btn 1
            // 25,0,8,11,0,0 - click btn 2
            // 25,0,8,11,0,2 - hold btn 2
            // 25,0,8,11,0,3 - release btn 2
            // 25,0,8,7,0,0 - click btn 3
            // 25,0,8,7,0,2 - hold btn 3
            // 25,0,8,7,0,3 - release btn 3
            // 25,0,8,15,0,0 - click btn 4
            // 25,0,8,15,0,2 - hold btn 4
            // 25,0,8,15,0,3 - release btn 4
            // TODO: do not know how to get to use 5,6,7,8 buttons
            const btn_lookup = {
                     3: 'btn_1',
                    11: 'btn_2',
                     7: 'btn_3',
                    15: 'btn_4',  
                  },
                  act_lookup = {
                     0: 'click',
                     2: 'hold',
                     3: 'release',
                  },
                  button = btn_lookup[msg.data[3]],
                  action = act_lookup[msg.data[5]];
            if (button && action) {
                return {[button]: action};
            }
        }, 
    },
```

closed 2019-11-05 by Koenkk

kirovilya, 2019-11-05:
Ok. thank you