Koenkk/zigbee-herdsman #86: Occasionally get UnhandledPromiseRejectionWarning from waitress.js

sjorge, 2019-11-05:
# Bug Report
## What happened
```
(node:88401) UnhandledPromiseRejectionWarning: Error: Timeout - 36534 - 11 - 21 - 1 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
(node:88401) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:88401) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

I get this error every now and then after having paired 2 new E1743 remotes. Both got unbound from group with ID 99 and bound to 2 other groups, each group has 1 bulb in it, one has an IKEA bulb the other a Hue bulb.

This may be related to the 'emulated' reporting, but I also got it at least once after removing a group.

## What did you expect to happen
n/a

## How to reproduce it (minimal and precise)
Not sure, I thought it might be the emulated reporting stuff for the hue bulb but I drop and raised the brightness for it a few times without triggering it.

## Debug Info
zigbee2mqtt version: 1.6.0-dev @ 2fb4e20
CC253X firmware version: 20190425

sjorge, 2019-11-05:
I'll update and keep an eye out on the logs for more of this

sjorge, 2019-11-05:
I can get you more context from the log from yesterday...

```
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:09: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_desk_lamp', payload '{"state":"ON","brightness":254,"color_temp":37,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:09: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":10,"brightness":254,"color":{"x":0.7006,"y":0.2993},"color_temp":37}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:25: MQTT publish: topic 'zigbee2mqtt/bedroom/remote_desk_lamp', payload '{"linkquality":13,"click":"off"}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:26: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_desk_lamp', payload '{"state":"OFF","brightness":254,"color_temp":37,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:26: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"OFF","linkquality":13,"brightness":254,"color":{"x":0.7006,"y":0.2993},"color_temp":37}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:27: MQTT publish: topic 'zigbee2mqtt/bedroom/remote_desk_lamp', payload '{"linkquality":13,"click":"on"}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:27: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_desk_lamp', payload '{"state":"ON","brightness":254,"color_temp":37,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:27: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":10,"brightness":254,"color":{"x":0.7006,"y":0.2993},"color_temp":37}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:29: MQTT publish: topic 'zigbee2mqtt/bedroom/remote_desk_lamp', payload '{"linkquality":0,"click":"brightness_down"}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:31: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_desk_lamp', payload '{"state":"ON","brightness":160,"color_temp":37,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:31: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":23,"brightness":160,"color":{"x":0.7006,"y":0.2993},"color_temp":37}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:53:33: MQTT publish: topic 'zigbee2mqtt/bedroom/remote_desk_lamp', payload '{"linkquality":0,"click":"brightness_stop"}'
(node:88401) UnhandledPromiseRejectionWarning: Error: Timeout - 36534 - 11 - 21 - 1 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
(node:88401) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:88401) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:54:02: MQTT publish: topic 'zigbee2mqtt/bedroom/motion', payload '{"battery":100,"voltage":3025,"illuminance":1,"linkquality":13,"occupancy":false}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:55:03: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_night_light', payload '{"color":{"x":0.7006,"y":0.2993},"state":"ON","brightness":254,"color_temp":153}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:55:03: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light', payload '{"state":"ON","linkquality":10,"brightness":254,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:56:30: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_night_light', payload '{"color":{"x":0.7006,"y":0.2993},"state":"ON","brightness":254,"color_temp":153}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:56:30: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light', payload '{"state":"ON","linkquality":7,"brightness":254,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:57:34: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_night_light', payload '{"color":{"x":0.7006,"y":0.2993},"state":"ON","brightness":254,"color_temp":153}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:57:34: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light', payload '{"state":"ON","linkquality":10,"brightness":254,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:57:57: MQTT publish: topic 'zigbee2mqtt/bedroom/remote_desk_lamp', payload '{"linkquality":0,"click":"off"}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:57:58: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_desk_lamp', payload '{"state":"OFF","brightness":160,"color_temp":37,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:57:58: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"OFF","linkquality":13,"brightness":160,"color":{"x":0.7006,"y":0.2993},"color_temp":37}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:58:11: MQTT publish: topic 'zigbee2mqtt/bedroom/remote_desk_lamp', payload '{"linkquality":0,"click":"on"}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:58:12: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_desk_lamp', payload '{"state":"ON","brightness":160,"color_temp":37,"color":{"x":0.7006,"y":0.2993}}'
ESC[32mzigbee2mqtt:info ESC[39m 2019-11-04T16:58:12: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":15,"brightness":160,"color":{"x":0.7006,"y":0.2993},"color_temp":37}'
```

sjorge, 2019-11-05:
I thought I had 3 of those... I do but only one is in zigbee2mqtt, the other two are in node-red from around the same timestamp... I wonder if it may be something were ntpd correct the clock offset and it messed with the timers or something? It's all around the same time ~16:53-16:54

sjorge, 2019-11-05:
Also I looked up which device the network address belonged too..

```
{"id":5,"type":"Router","ieeeAddr":"0x0017880104259333","nwkAddr":36534,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"LCT012","epList":[11,242],"endpoints":{"11":{"profId":49246,"epId":11,"devId":528,"inClusterList":[0,3,4,5,6,8,768,4096,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCT012","manufacturerName":"Philips","powerSource":1,"zclVersion":1,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170908","swBuildId":"1.29.0_r21169"}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"currentX":45317,"currentY":20204,"colorTemperature":153}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20170908","swBuildId":"1.29.0_r21169","zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1}}
```

Koenkk, 2019-11-06:
Please provide the output when running with the latest docker container, otherwise logging is incomplete.

sjorge, 2019-11-08:
Not had the error since the update.

sjorge, 2019-11-08:
Got another one today:
```
(node:24522) UnhandledPromiseRejectionWarning: Error: Timeout - 36534 - 11 - 42 - 6 - 1 after 10000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:425:11)
    at tryOnTimeout (timers.js:289:5)
    at listOnTimeout (timers.js:252:5)
    at Timer.processTimers (timers.js:212:10)
(node:24522) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 3)
```

Koenkk, 2019-11-10:
Can you provide the log of this error when running zigbee2mqtt with `DEBUG=zigbee-herdsman*`

sjorge, 2019-11-10:
That would be the same as running with logging set to debug?

Koenkk, 2019-11-11:
No, you will also get the zigbee-herdsman debug logging on the stdout with this.

sjorge, 2019-11-11:
```
zigbee2mqtt:debug 2019-11-11T15:41:16: Received Zigbee message from 'bedroom/night_light', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":254}' from endpoint 1 with groupID 0
2019-11-11T15:41:16.973Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":48855,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":19,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,34,11,10,0]}}
2019-11-11T15:41:16.973Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,215,190,1,1,8,0,19,0,30,5,24,34,11,10,0,120]
2019-11-11T15:41:16.974Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-11T15:41:16.984Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2019-11-11T15:41:16.984Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2019-11-11T15:41:16.985Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [] - 100
2019-11-11T15:41:16.985Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2019-11-11T15:41:16.985Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-11T15:41:16.992Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,19,213]
2019-11-11T15:41:16.992Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,19,213]
2019-11-11T15:41:16.992Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [] - 213
2019-11-11T15:41:16.992Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":19}
2019-11-11T15:41:16.992Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt:info  2019-11-11T15:41:17: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_night_light', payload '{"color":{"x":0.7006,"y":0.2993},"state":"ON","brightness":254,"color_temp":153}'
zigbee2mqtt:info  2019-11-11T15:41:17: MQTT publish: topic 'zigbee2mqtt/bedroom/night_light', payload '{"state":"ON","linkquality":65,"brightness":254,"color":{"x":0.7006,"y":0.2993}}'
2019-11-11T15:41:41.606Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":38370,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":20,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,12,0,0,0]}}
2019-11-11T15:41:41.606Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,226,149,1,1,0,0,20,0,30,5,16,12,0,0,0,78]
2019-11-11T15:41:41.616Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2019-11-11T15:41:41.616Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2019-11-11T15:41:41.616Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [] - 100
2019-11-11T15:41:41.617Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2019-11-11T15:41:41.617Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-11T15:41:41.623Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,20,210]
2019-11-11T15:41:41.624Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,20,210]
2019-11-11T15:41:41.624Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [] - 210
2019-11-11T15:41:41.624Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":20}
2019-11-11T15:41:41.624Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-11T15:41:41.634Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,226,149,0,245]
2019-11-11T15:41:41.634Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,226,149,0,245]
2019-11-11T15:41:41.635Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [�] - 245
2019-11-11T15:41:41.635Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":38370,"relaycount":0,"relaylist":[]}
2019-11-11T15:41:41.635Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2019-11-11T15:41:41.690Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,0,0,226,149,1,1,0,65,0,139,129,137,0,0,8,24,12,1,0,0,0,32,3,226,149,29,56]
2019-11-11T15:41:41.691Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,0,0,226,149,1,1,0,65,0,139,129,137,0,0,8,24,12,1,0,0,0,32,3,226,149,29,56]
2019-11-11T15:41:41.691Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [�A���
                                                                                                        �] - 56
2019-11-11T15:41:41.691Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":38370,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":9011595,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,12,1,0,0,0,32,3]}}
2019-11-11T15:41:41.692Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true},"transactionSequenceNumber":12,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":0,"status":0,"dataType":32,"attrData":3}],"Cluster":{"ID":0,"attributes":{"zclVersion":{"ID":0,"type":32,"name":"zclVersion"},"appVersion":{"ID":1,"type":32,"name":"appVersion"},"stackVersion":{"ID":2,"type":32,"name":"stackVersion"},"hwVersion":{"ID":3,"type":32,"name":"hwVersion"},"manufacturerName":{"ID":4,"type":66,"name":"manufacturerNa [... cut ...]
zigbee2mqtt:info  2019-11-11T15:43:28: MQTT publish: topic 'zigbee2mqtt/bedroom/group_remote_desk_lamp', payload '{"state":"ON","brightness":254,"color_temp":366,"color":{"x":0.4573,"y":0.41}}'
zigbee2mqtt:info  2019-11-11T15:43:28: MQTT publish: topic 'zigbee2mqtt/bedroom/desk_lamp', payload '{"state":"ON","linkquality":65,"brightness":254,"color":{"x":0.4573,"y":0.41},"color_temp":366}'
```

I had it running all day without and error, until I was messing with my desk layout and unplugged some stuff including 1 hue bulb.

I think in this case it makes sense as the bulb was offline I think when it tried to do something with it.