Koenkk/zigbee-herdsman #87: [Question] Valid PANId

Sineos, 2019-11-10:
I'm using @hobbyquaker node-red zigbee implementation (herdsman v0.12.3) and switched from CC2531 to Z-Stack 3.0 with CC1352P-2.

In this process I changed my Network Key and the PANId. The PANId currently set in the node settings is 1A63.
It worked quite flawlessly and I could easily pair all my devices (even without bringing them near the coordinator) but one issue remains: I can no longer join my CC2530 + CC2591 router.

I have already re-flashed the router but to no avail. Before starting to play again with PANId and Network Key, I have some questions:

1. Valid PANIds are 16 bit UInt as Hex, correct?
2. Valid Network Keys are 32 bit UInt as Hex?
3. Are the Hex values case sensitive?

Koenkk, 2019-11-10:
1. Yes
2. The encryption key is 128 bits, (16 * 8 bit unit)
3. No

closed 2019-11-10 by Koenkk

Sineos, 2019-11-10:
Uups, of course 128 bit (mentally got stuck at the 32 chars). Thnx for the correction