Koenkk/zigbee-herdsman #88: Network gets slow when using Ikea dimmer

h4nc, 2019-11-11:
I updated to 1.7.0. and it worked until I tried to use my ikea dimmer. My system immediately got stuck. 
I tried several things (restarting, reinstalling) but only reflashing my stick made it work again. 

There is no error but when I'm using my ikea dimmer now the system gets extremely slow. Other devices won't work for some seconds. It feels like the dimmer sends to many messages and the system gets stuck because it has to worked itself though the messages. Inputs from other devices are handled after that. 

If I don't use the dimmer other devices work immediately. The device worked fine in former versions.

If someone can reproduce this users with ikea dimmers should be warned, because they might need to reflash, if the ikea dimmer does something nasty here.

Koenkk, 2019-11-11:
I don't think messages of the dimmer are the issue, but that the automation made based on the dimmer are an issue. Due to https://github.com/Koenkk/zigbee-herdsman-converters/commit/ad29b2b6bcdb8ae561d21f2c8baf5c82fd681e51 it publishes more frequent -> if your automation responds to all of them using e.g. a dim command on a zigbee bulb, there will be a lot of commands.

h4nc, 2019-11-11:
I see, thanks. Wasn't thinking of that. Was thinking that z2m must be the issue because this is what I changed.

So the issue that I had to reflash my device way most probably something different, I had an error there but did not save it (sry). Can't find it in the logs any more. 

This is my automation. Would it be better to use the sensor entity of the dimmer? I think this would probably change to often too? Or maybe adding some delay to the action at the end would help. But there is probably a more elegant way to do this. 

```
- alias:  ikea rotate - plus
  initial_state: 'on'
  trigger:
    platform: mqtt
    topic: 'zigbee2mqtt/Zigbee Dimmer WZ'
  condition:
    - condition: template
      value_template: "{{ trigger.payload_json.action in ['rotate_right', 'rotate_right_quick'] }}"
    - condition: state
      entity_id: light.wz
      state: 'on'             
  action:         
    - service: light.turn_on
      data_template:
        entity_id: light.wz
        brightness: "{{  states.light.wz.attributes.brightness | int + 25 }}"

```

Koenkk, 2019-11-11:
From the zigbee2mqtt side you can use the debounce: https://www.zigbee2mqtt.io/information/configuration.html#device-specific-configuration

h4nc, 2019-11-11:
Perfect solution, thanks!

closed 2019-11-11 by h4nc

lolorc, 2019-11-11:
another workaround would be to add some kind of repeat number like LIRC does. It sends the remote code with a sequence number by. It would be easier to pick the first message  or every message with sequence modulo something.
debounce will work on every message as far as I understand

``` 
Nov 10 22:37:29 lirc/status/tv/v- 0
Nov 10 22:37:29 lirc/status/tv/v- 1
Nov 10 22:37:29 lirc/status/tv/v- 2
Nov 10 22:37:29 lirc/status/tv/v- 3
Nov 10 22:37:29 lirc/status/tv/v- 4
Nov 10 22:37:29 lirc/status/tv/v- 5
```

h4nc, 2019-11-12:
Seems like debounce works fine, lag is gone. Don't know if lire would make more sense. @Koenkk ?

lolorc, 2019-11-12:
i guess the elapsed attribute could also be used in HA automation: don't treat msg with a
 small elapsed.

h4nc, 2019-11-12:
But would this mean the broker has to deal with all those messages. Isn’t it better to avoid this on the z2m side with debounce?

Koenkk, 2019-11-12:
I feel debounce is a cleaner solution, it doesn't add any extra stuff to the payload keeping it clean. @h4nc would you mind making a PR to update the docs? https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/ICTC-G-1.md

lolorc, 2019-11-12:
my point of view here is that I don't want z2m to mask/merge/do anything-supposedly-smart with my zigbee messages and attributes.
I don't like the cache setting either, it's just here to circumvent a limitation of HA.

h4nc, 2019-11-12:
@lolorc I understand your point of view, but for me debounce is good enough as it solves my issue

> would you mind making a PR to update the docs?

@Koenkk done