Koenkk/zigbee-herdsman #89: 45853GE missing energy reporting

djchen, 2019-11-14:
I re-paired one of my 45853GE plugs which has energy reporting and it no longer reports power usage.

1. Removed plug from network
2. Reset plug and it re-joined network
3. Able to turn it on and off but does not report power usage
I've repeated this 3 times, same results.

Logs from pairing:
```
info  2019-11-14T05:04:41: Device '<snip>' joined
info  2019-11-14T05:04:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"<snip>"}}'
info  2019-11-14T05:04:41: Starting interview of '<snip>'
info  2019-11-14T05:04:52: MQTT publish: topic 'homeassistant/switch/<snip>/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/<snip>/set","state_topic":"zigbee2mqtt/<snip>","name":"<snip>_switch","unique_id":"<snip>_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_<snip>"],"name":"<snip>","sw_version":"Zigbee2mqtt 1.7.1","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-14T05:04:52: MQTT publish: topic 'homeassistant/sensor/<snip>/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/<snip>","json_attributes_topic":"zigbee2mqtt/<snip>","name":"<snip>_power","unique_id":"<snip>_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_<snip>"],"name":"<snip>","sw_version":"Zigbee2mqtt 1.7.1","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-14T05:04:52: MQTT publish: topic 'homeassistant/sensor/<snip>/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/<snip>","json_attributes_topic":"zigbee2mqtt/<snip>","name":"<snip>_linkquality","unique_id":"<snip>_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_<snip>"],"name":"<snip>","sw_version":"Zigbee2mqtt 1.7.1","model":"Plug-in smart switch (45853GE)","manufacturer":"GE"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-14T05:04:52: Successfully interviewed '<snip>', device has successfully been paired
info  2019-11-14T05:04:52: Device '<snip>' is supported, identified as: GE Plug-in smart switch (45853GE)
info  2019-11-14T05:04:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"<snip>","model":"45853GE","vendor":"GE","description":"Plug-in smart switch","supported":true}}'
info  2019-11-14T05:04:52: Configuring '<snip>'
info  2019-11-14T05:04:53: MQTT publish: topic 'zigbee2mqtt/Bedroom Left Window Fan', payload '{"state":"ON","linkquality":131,"power":54.9,"last_seen":1573707893031}'
info  2019-11-14T05:04:53: Succesfully configured '<snip>'
```

Problem device in database.db
```
{"id":22,"type":"Router","ieeeAddr":"<snip>","nwkAddr":54555,"manufId":4388,"manufName":"Jasco Products","powerSource":"Mains (single phase)","modelId":"45853","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,2821,1794],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"45853","manufacturerName":"Jasco Products","powerSource":1,"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1}},"seMetering":{"attributes":{"multiplier":1,"divisor":10000}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"<snip>","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":259,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{},"binds":[]}},"appVersion":18,"stackVersion":80,"hwVersion":1,"zclVersion":1,"interviewCompleted":true,"meta":{"configured":2}}
```

Fully working device in database.db (another of the same plug)
```
{"id":16,"type":"Router","ieeeAddr":"<diff_device>","nwkAddr":6601,"manufId":4388,"manufName":"Jasco Products","powerSource":"Mains (single phase)","modelId":"45853","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,1794,2821],"outClusterList":[3,10,25],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1,"manufacturerName":"Jasco Products","modelId":"45853","powerSource":1}},"genIdentify":{"dir":{"value":3},"attributes":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attributes":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attributes":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attributes":{"onOff":0}},"genTime":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":2},"attributes":{}},"seMetering":{"dir":{"value":1},"attributes":{"currentSummDelivered":{"0":0,"1":0},"status":0,"unitOfMeasure":0,"multiplier":1,"divisor":10000,"summaFormatting":251,"demandFormatting":11,"historicalConsumpFormatting":120,"meteringDeviceType":0,"instantaneousDemand":0}},"haDiagnostic":{"dir":{"value":1},"attributes":{"averageMacRetryPerApsMessageSent":1,"lastMessageLqi":255,"lastMessageRssi":-17}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"<snip>","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":259,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{"zclVersion":1,"appVersion":18,"stackVersion":80,"hwVersion":1,"manufacturerName":"Jasco Products","modelId":"45853","powerSource":1}},"genIdentify":{"dir":{"value":3},"attributes":{"identifyTime":0}},"genOnOff":{"dir":{"value":2},"attributes":{}},"haDiagnostic":{"dir":{"value":1},"attributes":{"averageMacRetryPerApsMessageSent":1,"lastMessageLqi":255,"lastMessageRssi":-17}}},"binds":[]}},"hwVersion":1,"meta":{"configured":2}}
```

The entry for the problem device has data in the DB.

Koenkk, 2019-11-14:
Before re-pairing, did you do https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove (this is crucial otherwise the device won't be configured again to provide power measurements)

djchen, 2019-11-14:
> Before re-pairing, did you do https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove (this is crucial otherwise the device won't be configured again to provide power measurements)

Yes I did. I also confirmed the device was removed from the configuration.yaml and database.db after issuing the remove.

Koenkk, 2019-11-14:
Can you try changing the device definition in `devices.js` to

```js
{
    zigbeeModel: ['45853'],
    model: '45853GE',
    vendor: 'GE',
    description: 'Plug-in smart switch',
    supports: 'on/off',
    fromZigbee: [fz.on_off, fz.generic_power],
    toZigbee: [tz.on_off, tz.ignore_transition],
    meta: {configureKey: 3},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint);
        const payload = [{
            attribute: 'instantaneousDemand',
            minimumReportInterval: 10,
            maximumReportInterval: 60,
            reportableChange: 1,
        }];
        await endpoint.configureReporting('seMetering', payload);
        await endpoint.read('seMetering', ['multiplier', 'divisor']);
    },
},
```

Instructions on how to modify this file locally can be found here: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

If this doesn't work, please share `database.db`.

djchen, 2019-11-14:
Steps I just took
1. Removed device from network
2. Stopped zigbee2mqtt and made sure it was removed from state.json, configuration.yaml, and database.db
3. Started zigbee2mqtt with the modified devices.js

Unfortunately it still does not show power usage. Is there a way I can privately send you my `database.db`?

Koenkk, 2019-11-14:
Yes on telegram (@koenkk)

Koenkk, 2019-11-15:
I've checked your `database.db` and it seems that it didn't configure with the new configuration. (`"configured":2"`) while it should be `3` (as `configureKey = 3` in the new config).

Did you see a fail to configure when starting zigbee2mqtt?

Also can you provide the logs when running with debug logging and attaching a load to the switch?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

If you provide anything via telegram again, please don't provide it in a secret chat (I'm not able to read it via telegram web then)

djchen, 2019-11-16:
That's because I reverted the devices.js and re-paired. I'm going to spin up a new zigbee2mqtt instance with a different cc2531 stick and the modified devices.js and try it again this weekend.

Thanks! I'll report back soon.

djchen, 2019-11-20:
@Koenkk Made the changes to devices.js and tried to re-pair with debug logging. Saw this:

```
info  2019-11-20T07:12:06: Removing 'Humidifier'
warn  2019-11-20T07:12:07: Device '0x0022<snip>' left the network
info  2019-11-20T07:12:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0022<snip>"}}'
error 2019-11-20T07:12:07: Failed to remove Humidifier (Error: DatabaseEntry with ID '3' does not exist)
warn  2019-11-20T07:12:07: Device '0x0022<snip>' left the network
info  2019-11-20T07:12:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0022<snip>"}}'
warn  2019-11-20T07:12:07: Device '0x0022<snip>' left the network
info  2019-11-20T07:12:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0022<snip>"}}'
info  2019-11-20T07:12:14: Device 'Humidifier' joined
info  2019-11-20T07:12:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"Humidifier"}}'
info  2019-11-20T07:12:14: Starting interview of 'Humidifier'
info  2019-11-20T07:12:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Humidifier"}}'
debug 2019-11-20T07:12:15: Device 'Humidifier' announced itself
debug 2019-11-20T07:12:16: Received Zigbee message from 'Humidifier', type 'readResponse', cluster 'genBasic', data '{"modelId":"45853","manufacturerName":"Jasco Products","powerSource":1}' from endpoint 1 with groupID 0
debug 2019-11-20T07:12:16: No converter available for '45853GE' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"45853","manufacturerName":"Jasco Products","powerSource":1}'
debug 2019-11-20T07:12:16: Received Zigbee message from 'Humidifier', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":18,"stackVersion":80}' from endpoint 1 with groupID 0
debug 2019-11-20T07:12:16: No converter available for '45853GE' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1,"appVersion":18,"stackVersion":80}'
debug 2019-11-20T07:12:17: Received Zigbee message from 'Humidifier', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2019-11-20T07:12:17: No converter available for '45853GE' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
info  2019-11-20T07:12:17: Successfully interviewed 'Humidifier', device has successfully been paired
info  2019-11-20T07:12:17: Device 'Humidifier' is supported, identified as: GE Plug-in smart switch (45853GE)
info  2019-11-20T07:12:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"Humidifier","model":"45853GE","vendor":"GE","description":"Plug-in smart switch","supported":true}}'
info  2019-11-20T07:12:17: Configuring 'Humidifier'
debug 2019-11-20T07:12:17: Received Zigbee message from 'Humidifier', type 'readResponse', cluster 'seMetering', data '{"multiplier":1,"divisor":10000}' from endpoint 1 with groupID 0
info  2019-11-20T07:12:17: Succesfully configured 'Humidifier'
```

djchen, 2019-11-20:
Side note, I upgraded from CC2531 to CC26X2R1 and had to re-pair everything. I started with fresh config and all of the 45853GE lost power metering.

Koenkk, 2019-11-20:
I don't see a configure for the `45853GE` in your logs.

- Do you still have the changes of https://github.com/Koenkk/zigbee-herdsman/issues/89#issuecomment-554045466 ?
- If yes, can you share your `database.db`?

djchen, 2019-11-24:
@Koenkk Sorry for the delay, I finally got the extra plug and setup a separate instance of Z2M. (This will make it easier for me to test things without breaking my network)

Logs:
https://gist.github.com/djchen/3e3590c1b4c391a197ece9d66be3fe11
DB:
https://gist.github.com/djchen/de237220b6acfdc0b8a7d1b14bcbad18

This is with the modified devices.js.

Koenkk, 2019-11-25:
I think this is a bug in the zigbee2mqtt 1.7.1 release. Can you try to switch to the latest dev branch, remove the device from `database.db` and pair it again?

djchen, 2019-11-26:
I pulled down the lastest-dev docker image just now, reverted the devices.js change, and started with a fresh install. I'm still not seeing power usage.
> zigbee2mqtt:info  2019-11-25 21:08:02: Starting zigbee2mqtt version 1.7.1+dev (commit #7d1635a)
The commit shorthash matches the latest dev branch commit.

Logs:
https://gist.github.com/djchen/c8251f19d271f1ae309a55d4883609c2
DB:
https://gist.github.com/djchen/96c93a87b23e03c059a7546ba62b5868

djchen, 2019-11-26:
Heres the same as above but with the devices.js modification.

Logs:
https://gist.github.com/djchen/bc7a423068a8751b86f78bd2aabb64ae
DB:
https://gist.github.com/djchen/6f3f572eb8c74b1ed77177dbaa11646e

djchen, 2019-11-27:
@Koenkk I am able to get a power value if I change
`await endpoint.read('seMetering', ['multiplier', 'divisor']);`
to
`await endpoint.read('seMetering', ['multiplier', 'divisor', 'instantaneousDemand']);`

```
zigbee2mqtt:debug 2019-11-27 01:20:59: Received Zigbee message from '0x0022a300001af3b6', type 'readResponse', cluster 'seMetering', data '{"multiplier":1,"divisor":10000,"instantaneousDemand":76}' from endpoint 1 with groupID 0

zigbee2mqtt:info  2019-11-27 01:20:59: MQTT publish: topic 'zigbee2mqtt/0x0022a300001af3b6', payload '{"power":7.6,"linkquality":94,"last_seen":1574846459784}'
```

It seems like the configureReporting of instantaneousDemand isn't working properly?

Koenkk, 2019-11-27:
The device somehow doesn't seem to care about the configured reporting, can you try with:

```js
{
    zigbeeModel: ['45853'],
    model: '45853GE',
    vendor: 'GE',
    description: 'Plug-in smart switch',
    supports: 'on/off',
    fromZigbee: [fz.on_off, fz.generic_power],
    toZigbee: [tz.on_off, tz.ignore_transition],
    meta: {configureKey: 4},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await configureReporting.onOff(endpoint);
        const payload = [{
            attribute: 'instantaneousDemand',
            minimumReportInterval: 0,
            maximumReportInterval: 5,
            reportableChange: 0,
        }];
        await endpoint.configureReporting('seMetering', payload);
        await endpoint.read('seMetering', ['multiplier', 'divisor']);
    },
},
```

djchen, 2019-12-01:
@Koenkk That doesn't seem to work. I checked the database.db and see `..."meta":{"configured":4}...` but not seeing energy reporting with a 5-10W load.

Koenkk, 2019-12-03:
@djchen any change you could sniff the network when attaching a load? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

yayiblue, 2019-12-21:
Hello!
I am using zigbee2mqtt too well.
Thank you for always.

I have no power value for PM-C140-ZB.
I repeated this result about 10 times.

1. The power status does not rise.
2. Sometimes the item itself is missing (same as djchen)
3. This works fine with zigbee2mqtt 1.5.1.
4.zigbee2mqtt 1.5.1-> migration-> zigbee2mqtt 1.7.1 (or 1.8.0)
     Only migrated equipment will reflect power status,
    New or repaired in 1.7.1 (or 1.8.0)
    The equipment cannot reflect the condition.
   (on / off, linkquality is a state reflection.)
5. I modified and tried https://github.com/Koenkk/zigbee2mqtt/issues/2332#issuecomment-554043446 but it is the same.

[log.txt](https://github.com/Koenkk/zigbee-herdsman/files/3991507/log.txt)

Phanzar, 2020-01-05:
I'am also having the same issue with the edp re:dy plugs.

Zigbee2mqtt 1.6 is able to correctly activate the reporting of the instantaneous power from the plugs, but on the newer versions (1.7.1 and 1.8) this doesn't work, new plugs added don't report the instantaneous power, only the ones added on version 1.6 are working.

Koenkk, 2020-01-07:
I found the root cause of this issue, when updating to the latest dev you should see the zigbee2mqtt configures the device (Succesfully configured [FRIENDLY_NAME]), after this it should work.