Koenkk/zigbee-herdsman #98: Aqara Opple Switch support

uncle-fed, 2019-11-27:
https://homekitnews.com/2019/10/23/aqara-and-opple-release-new-smart-switches/

I got my 4-button switch today (also 2 and 6 button switches are available) and eager to make it work with zigbee2mqtt. The device is currently not supported and I started following the instructions here (https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html) and immediately got an issue. The instruction states that pairing can be done in any case with the message  that goes like `Successfully interviewed '0x00.....', device has successfully been paired`.

What I get is this:

```Nov 27 20:00:47 rpi npm[3807]: zigbee2mqtt:info  2019-11-27T20:00:47: Device '0x04cf8cdf3c75c05f' joined
Nov 27 20:00:47 rpi npm[3807]: zigbee2mqtt:info  2019-11-27T20:00:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x04cf8cdf3c75c05f"}}'
Nov 27 20:00:47 rpi npm[3807]: zigbee2mqtt:info  2019-11-27T20:00:47: Starting interview of '0x04cf8cdf3c75c05f'
Nov 27 20:00:47 rpi npm[3807]: zigbee2mqtt:info  2019-11-27T20:00:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c75c05f"}}'
Nov 27 20:00:58 rpi npm[3807]: zigbee2mqtt:error 2019-11-27T20:00:58: Failed to interview '0x04cf8cdf3c75c05f', device has not successfully been paired
Nov 27 20:00:58 rpi npm[3807]: zigbee2mqtt:info  2019-11-27T20:00:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x04cf8cdf3c75c05f"}}'

uncle-fed, 2019-11-27:
Debug log below

```  serialport/bindings/poller received "readable" +17s
  serialport/binding-abstract read +17s
  serialport/bindings/unixRead Starting read +17s
  serialport/bindings/unixRead Finished read { bytesRead: 18,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 90 15 34 01 02 00 00 00 00 da fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +1ms
  serialport/stream binding.read finished { bytesRead: 18 } +17s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,201,67,103,95,192,117,60,223,140,207,4,0,0,0,235] +17s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,201,67,103,95,192,117,60,223,140,207,4,0,0,0,235] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 201 - [Cg_�u<ߌ�] - 235 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - leaveInd - {"srcaddr":26435,"extaddr":"0x04cf8cdf3c75c05f","request":0,"removechildren":0,"rejoin":0} +17s
  zigbee-herdsman:controller:log Device leave '0x04cf8cdf3c75c05f' +17s
  zigbee-herdsman:controller:log Removing device from database '0x04cf8cdf3c75c05f' +0ms
zigbee2mqtt:warn  2019-11-27T20:17:44: Device '0x04cf8cdf3c75c05f' left the network
winston:file written true false
zigbee2mqtt:info  2019-11-27T20:17:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x04cf8cdf3c75c05f"}}'
winston:file written true false
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +21ms
winston:file logged 3512 warn  2019-11-27T20:17:44: Device '0x04cf8cdf3c75c05f' left the network

winston:file logged 3692 info  2019-11-27T20:17:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x04cf8cdf3c75c05f"}}'

  serialport/stream _read reading { start: 226, toRead: 65310 } +27ms
  serialport/binding-abstract read +30ms
  serialport/bindings/unixRead Starting read +30ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +1ms
  serialport/bindings/poller Polling for "readable" +33ms
  serialport/bindings/poller received "readable" +1s
  serialport/binding-abstract read +1s
  serialport/bindings/unixRead Starting read +1s
  serialport/bindings/unixRead Finished read { bytesRead: 17,
  buffer:
   <Buffer fe 0e 61 02 02 00 02 06 03 90 15 34 01 02 00 00 00 00 da fe 03 61 08 00 01 55 3e fe 03 61 08 00 01 55 3e fe 06 61 08 00 04 00 08 00 00 63 fe 03 61 08 ... > } bytes +1ms
  serialport/stream binding.read finished { bytesRead: 17 } +1s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,143,210,95,192,117,60,223,140,207,4,0,0,144] +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,143,210,95,192,117,60,223,140,207,4,0,0,144] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [��_�u<ߌ�] - 144 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":53903,"extaddr":"0x04cf8cdf3c75c05f","parentaddr":0} +1s
  zigbee-herdsman:controller:log Device '0x04cf8cdf3c75c05f' joined +1s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +6ms
  zigbee-herdsman:controller:log Device '0x04cf8cdf3c75c05f' accepted by handler +3ms
  zigbee-herdsman:controller:log New device '0x04cf8cdf3c75c05f' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x04cf8cdf3c75c05f' +0ms
zigbee2mqtt:info  2019-11-27T20:17:45: Device '0x04cf8cdf3c75c05f' joined
winston:file written true false
zigbee2mqtt:info  2019-11-27T20:17:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x04cf8cdf3c75c05f"}}'
winston:file written true false
  zigbee-herdsman:controller:log Interview '0x04cf8cdf3c75c05f' start +15ms
zigbee2mqtt:info  2019-11-27T20:17:45: Starting interview of '0x04cf8cdf3c75c05f'
winston:file written true false
zigbee2mqtt:info  2019-11-27T20:17:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c75c05f"}}'
winston:file written true false
  zigbee-herdsman:controller:device Interview - start device '0x04cf8cdf3c75c05f' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":53903,"nwkaddrofinterest":53903} +18s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,143,210,143,210,35] +18s
  serialport/stream _write 9 bytes of data +40ms
  serialport/binding-abstract write 9 bytes +42ms
  serialport/bindings/unixWrite Starting write 9 bytes offset 0 bytesToWrite 9 +18s
  serialport/stream _read reading { start: 243, toRead: 65293 } +2ms
  serialport/binding-abstract read +3ms
winston:file logged 3754 info  2019-11-27T20:17:45: Device '0x04cf8cdf3c75c05f' joined

winston:file logged 3914 info  2019-11-27T20:17:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x04cf8cdf3c75c05f"}}'

winston:file logged 3984 info  2019-11-27T20:17:45: Starting interview of '0x04cf8cdf3c75c05f'

winston:file logged 4162 info  2019-11-27T20:17:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x04cf8cdf3c75c05f"}}'

[... 106 lines cut ...]
  serialport/stream binding.read finished { bytesRead: 6 } +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,4,0,96] +23ms
[... cut ...]
zigbee2mqtt:info  2019-11-27T20:17:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x04cf8cdf3c75c05f"}}'
winston:file written true false
winston:file logged 4343 error 2019-11-27T20:17:56: Failed to interview '0x04cf8cdf3c75c05f', device has not successfully been paired

winston:file logged 4520 info  2019-11-27T20:17:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x04cf8cdf3c75c05f"}}'

uncle-fed, 2019-11-27:
OK, I clicked on the switch main buttons multiple times and then and held 2 buttons out of 4 together press in various combination.. and suddenly the switch got paired. I really don't know what made it work, but the "official reset button" on the back of the switch that you need to hold for 5 seconds to start pairing did not work by itself.

Koenkk, 2019-11-27:
There is already a pr open, https://github.com/Koenkk/zigbee-herdsman-converters/pull/774

closed 2019-11-27 by Koenkk