Koenkk/zigbee-herdsman #101: IP socket as alternativ to serial port

KoljaWindeler, 2019-11-29:
Hi, 
want do you think about adding a IP sockets connection in addition to the serial port?

I'm using socat to create a virtual serial socket which sort of work, but every one in a while the connection crashes. At that point my script has to shut down socat and zigbee2mqtt and start all over. 

It would be nice if zigbee-herdsman could connect directly via IP and also handle those rare reconnects. 
Kolja

Koenkk, 2019-11-29:
This is already possible in the latest-dev, docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/how_tos/how_to_connect_to_a_remote_adapter.md

closed 2019-11-29 by Koenkk

KoljaWindeler, 2019-11-29:
outstanding! thanks a lot! (sorry, didn't find it in the docs .. )

KoljaWindeler, 2019-11-29:
hi @Koenkk any chance that you'll add a reconnect mechanism? It works now without socat and thats great, but will still require to restart once it disconnects .. not a big issue but might be a nice improvement

Koenkk, 2019-12-03:
@hdo could you take a look at this?

hdo, 2019-12-03:
> @hdo could you take a look at this?

I'll take a look into this.

@KoljaWindeler Why does the disconnect happen? How can i reproduce this?
My setup is running for almost 3 weeks and i didn't encounter any disconnects.

KoljaWindeler, 2019-12-05:
Hi, I guess this isn't due to zigbee2mqtt .. but sometimes wifi signals get yammed or the client device restarts for other reasons etc .. it's just likely that the connection won't stay stable forever, right?
Thanks for your work!

hdo, 2019-12-05:
> Hi, I guess this isn't due to zigbee2mqtt .. but sometimes wifi signals get yammed or the client device restarts for other reasons etc .. it's just likely that the connection won't stay stable forever, right?
> Thanks for your work!

This could be a reason. Are you using ESP32 as serial tcp bridge?