Koenkk/zigbee-herdsman #102: Device paired successfully but no communication with zigbee2mqtt

Exhortae, 2019-11-29:
Hi,

I have a strange problem that I can't seem to solve

I bought a gledopto downlight 9W (Model ID GL-D-004ZS) and paired successfully with  zigbee2mqtt

`// Device is pairing successfully

zigbee2mqtt:info  2019-11-29T05:33:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x00124b001ef16597"}}'
zigbee2mqtt:info  2019-11-29T05:33:24: Device '0x00124b001ef16597' joined
zigbee2mqtt:info  2019-11-29T05:33:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001ef16597"}}'
zigbee2mqtt:info  2019-11-29T05:33:24: Starting interview of '0x00124b001ef16597'
zigbee2mqtt:info  2019-11-29T05:33:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001ef16597"}}'
zigbee2mqtt:info  2019-11-29T05:33:35: Successfully interviewed '0x00124b001ef16597', device has successfully been paired
zigbee2mqtt:info  2019-11-29T05:33:35: Device '0x00124b001ef16597' is supported, identified as: Gledopto LED RGB + CCT downlight 9W  (GL-D-004ZS)
zigbee2mqtt:info  2019-11-29T05:33:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001ef16597","model":"GL-D-004ZS","vendor":"Gledopto","description":"LED RGB + CCT downlight 9W ","supported":true}}'
`

I had to add it's configuration in /zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman-converters/devices.js to make it supported 

`/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman-converters/devices.js

    {
        zigbeeModel: ['GL-D-004ZS'],
        model: 'GL-D-004ZS',
        vendor: 'Gledopto',
        description: 'LED RGB + CCT downlight 9W ',
        extend: gledopto.light,
        supports: 'on/off, brightness, color temperature, color',
    },
`
The problem is that apart from the message that the gledopto downlight joined the network I get absolutely no communication between it and the zigbee gateway (CC2531)

I have the 6W version of this downlight (GL-D-003Z) and it communicates with no problem with zigbee2mqtt (actually I didn't have to add an entry for it because it was already there in the default devices.js configuration)

This is the log of the  (GL-D-003Z) :

`zigbee2mqtt:info  2019-11-29T05:20:28: 0x00124b001ea82c9a (0x00124b001ea82c9a): GL-D-003Z - Gledopto LED RGB + CCT downlight  (Router)      
               
zigbee2mqtt:info  2019-11-29T05:36:43: MQTT publish: topic 'homeassistant/sensor/0x00124b001ea82c9a/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00124b001ea82c9a","json_attributes_topic":"zigbee2mqtt/0x00124b001ea82c9a","name":"0x00124b001ea82c9a_linkquality","unique_id":"0x00124b001ea82c9a_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001ea82c9a"],"name":"0x00124b001ea82c9a","sw_version":"Zigbee2mqtt 1.7.1","model":"LED RGB + CCT downlight  (GL-D-003Z)","manufacturer":"Gledopto"},"availability_topic":"zigbee2mqtt/bridge/state"}',
zigbee2mqtt:info  2019-11-29T05:36:43: MQTT publish: topic 'homeassistant/light/0x00124b001ea82c9a/light/config', payload '{"brightness":true,"color_temp":true,"xy":true,"schema":"json","command_topic":"zigbee2mqtt/0x00124b001ea82c9a/set","state_topic":"zigbee2mqtt/0x00124b001ea82c9a","name":"0x00124b001ea82c9a_light","unique_id":"0x00124b001ea82c9a_light_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001ea82c9a"],"name":"0x00124b001ea82c9a","sw_version":"Zigbee2mqtt 1.7.1","model":"LED RGB + CCT downlight  (GL-D-003Z)","manufacturer":"Gledopto"},"availability_topic":"zigbee2mqtt/bridge/state"}'
               `

I checked the database.db file and I have the 2 entries, first one is the 6W downlight GL-D-003Z which works well, second one is the 9W downlight GL-D-004ZS that sends no message to zigbee2mqtt 

Thanks

Koenkk, 2019-11-30:
Can you provide the debug log when sending to `zigbee2mqtt/0x00124b001ef16597/set` payload `ON` or `OFF`?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Exhortae, 2019-11-30:
Hi,

I tried with a new downlight (same model though), this is the log since the pairing to the issue of a payload off

```
info  2019-11-30T20:46:38: Device '0x00124b001ef1670c' joined
info  2019-11-30T20:46:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b001ef1670c"}}'
info  2019-11-30T20:46:38: Starting interview of '0x00124b001ef1670c'
info  2019-11-30T20:46:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b001ef1670c"}}'
debug 2019-11-30T20:46:39: Device '0x00124b001ef1670c' announced itself
debug 2019-11-30T20:46:39: Device '0x00124b001ef1670c' announced itself
debug 2019-11-30T20:46:48: Received Zigbee message from '0x00124b001ef1670c', type 'readResponse', cluster 'genBasic', data '{"modelId":"GL-D-004ZS","manufacturerName":"GLEDOPTO","powerSource":1}' from endpoint 11 with groupID 0
debug 2019-11-30T20:46:48: Received Zigbee message from '0x00124b001ef1670c', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1,"appVersion":1,"stackVersion":2}' from endpoint 11 with groupID 0
debug 2019-11-30T20:46:48: Received Zigbee message from '0x00124b001ef1670c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20131206","swBuildId":"1.0.6"}' from endpoint 11 with groupID 0
info  2019-11-30T20:46:48: Successfully interviewed '0x00124b001ef1670c', device has successfully been paired
info  2019-11-30T20:46:48: Device '0x00124b001ef1670c' is supported, identified as: Gledopto LED RGB + CCT downlight 9W PLUS  (GL-D-004ZS)
info  2019-11-30T20:46:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001ef1670c","model":"GL-D-004ZS","vendor":"Gledopto","description":"LED RGB + CCT downlight 9W PLUS ","supported":true}}'
debug 2019-11-30T20:47:49: Saving state to file /share/zigbee2mqtt/state.json
debug 2019-11-30T20:52:49: Saving state to file /share/zigbee2mqtt/state.json
debug 2019-11-30T20:53:31: Received MQTT message on 'zigbee2mqtt/0x00124b001ef1670c/set' with data '{"state":"OFF"}'
debug 2019-11-30T20:53:31: Publishing 'set' 'state' to '0x00124b001ef1670c'
info  2019-11-30T20:53:31: MQTT publish: topic 'zigbee2mqtt/0x00124b001ef1670c', payload '{"state":"OFF"}'

```
The downlight do turn off, I think it's actually a problem of integration of the device with hassio. I tried to add the device manually in hassio and it worked (I can change color, brightness..), 

It's strange that It doesn't get recognized automatically in hassio.

Koenkk, 2019-12-03:
For the device to be recognized in home assistant it needs to be added to https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js

I've added the device now and also to homeassistant.js.

It will be supported in the hassio edge version in a few hours, no manual changes from your side are needed anymore.

closed 2019-12-03 by Koenkk