Koenkk/zigbee-herdsman #104: Deferred operations (configure, bind/unbind, remove) with battery devices

kirovilya, 2019-12-06:
I noticed that battery devices often do not respond to commands. For example: configuration, bind/unbind, device deletion.

If it was somehow managed to be solved the 'configure' operation, then with two other cases it does not always work.
I think that devices need to use a similar approach - try to send a command several times when the device gets in touch.

Also, it is desirable to make this functionality inside the herdsman so as not to produce their own implementations of such queues.

kirovilya, 2020-02-06:
This task is important for all sleeping devices

Koenkk, 2020-02-06:
I think before sending these commands you manually have to wake them up. In zigbee2mqtt it is now possible to trigger a configure manually: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

kirovilya, 2020-02-06:
Yes. but we can also receive the nearest message from the device and send them a command in the hope that it has not had time to fall asleep yet. also as we do with the 'configure'.

I think this should be done for all the commands that we send to devices.

Koenkk, 2020-02-06:
Wouldn't it be unhandy when a user executes a bind but only e.g. after an hour the bind is executed?

kirovilya, 2020-02-06:
I do not know. for example, we do this configure and it also performs a bind inside...

CodeFinder2, 2020-03-02:
> Wouldn't it be unhandy when a user executes a bind but only e.g. after an hour the bind is executed?

I also think that this can cause weird behavior (and new Github issues 😅). Nonetheless I like the idea of a "command queue"...maybe we can add an (optional) flag to such commands that may be queued (meaning that the command should be queued if it cannot be executed right away)? And the default behavior would be the current one. But I guess we then also need some way to inspect (/clear) the current queue...

renne, 2020-10-08:
> Wouldn't it be unhandy when a user executes a bind but only e.g. after an hour the bind is executed?

If you have to climb two stairs between the PC and and a battery powered Zigbee device you do not have a chance to configure the device unless you un-mount it.

I suggest a queue table in the web-frontend with a trash-can button to delete single commands and a check-box per command to delete multiple commands. Homematic also uses queued commands with a table in the web-frontend.

closed 2020-11-17 by github-actions[bot]