Koenkk/zigbee-herdsman #108: Failed to pair btcino power meter

n1zzo, 2019-12-15:
Hi @Koenkk,
I'm trying to pair an unsupported device to zigbee2mqtt:
the btcino F20T60A power consumption meter,
however the device pairing fails during the interview phase.
[Here](https://pastebin.com/4nSxSAnR)' s the output log from zigbee2mqtt with debug logs enabled.

At these links you can find capture files for a [successful pairing](https://www.icloud.com/iclouddrive/0Oq9K_rYTs5JWaCpkshkG4mLg#successful%5Fpairing) with the official
gateway and a [Failed pairing](https://www.icloud.com/iclouddrive/0EXJgCG8dsW_Ys4cIJP32Lg1g#failed%5Fpairing) with zigbee2mqtt.

I noticed a difference in the official gateway sending a double Transport Key message, and I was trying to reproduce that, however that would require a change in the CC2531 firmware right?

Do you have any hints on this?

Koenkk, 2019-12-15:
The node descriptor request fails, can you try with zigbee2mqtt 1.8.0 as this will try to get it for 6 times instead of 2.

n1zzo, 2019-12-15:
Sure, still fails, here's the output:

```
  zigbee-herdsman:controller:device Interview - start device '0x0004740000899315' +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":27665,"nwkaddrofinterest":27665} +2m
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,17,108,17,108,35] +2m
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +56ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +2m
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
zigbee2mqtt:error 2019-12-15T18:49:57: Not connected to MQTT server!
  zigbee-herdsman:controller:device Interview - first node descriptor request failed for '0x0004740000899315', retrying... +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":27665,"nwkaddrofinterest":27665} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,17,108,17,108,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,0,204,21,147,137,0,0,116,4,0,0,0,48] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,0,204,21,147,137,0,0,116,4,0,0,0,48] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [���t] - 48 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":52224,"extaddr":"0x0004740000899315","parentaddr":0} +12s
  zigbee-herdsman:controller:log Device '0x0004740000899315' joined +12s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x0004740000899315' accepted by handler +1ms
  zigbee-herdsman:controller:log Device '0x0004740000899315' is already in database with different networkAddress, updating networkAddress +1ms
  zigbee-herdsman:controller:log Not interviewing '0x0004740000899315', completed 'false', in progress 'true' +11ms
zigbee2mqtt:error 2019-12-15T18:50:07: Not connected to MQTT server!
  zigbee-herdsman:controller:device Interview - failed for device '0x0004740000899315' with error 'Error: AREQ - ZDO - nodeDescRsp after 10000ms
  zigbee-herdsman:controller:device     at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
  zigbee-herdsman:controller:device     at ontimeout (timers.js:436:11)
  zigbee-herdsman:controller:device     at tryOnTimeout (timers.js:300:5)
  zigbee-herdsman:controller:device     at listOnTimeout (timers.js:263:5)
  zigbee-herdsman:controller:device     at Timer.processTimers (timers.js:223:10)' +11s
  zigbee-herdsman:controller:error Interview failed for '0x0004740000899315 with error 'Error: AREQ - ZDO - nodeDescRsp after 10000ms' +0ms
zigbee2mqtt:error 2019-12-15T18:50:10: Failed to interview 'btcino power meter', device has not successfully been paired
zigbee2mqtt:error 2019-12-15T18:50:10: Not connected to MQTT server!
zigbee2mqtt:error 2019-12-15T18:50:10: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"btcino power meter"}}
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,0,204,21,147,137,0,0,116,4,0,0,0,48] +11s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,0,204,21,147,137,0,0,116,4,0,0,0,48] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [���t] - 48 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":52224,"extaddr":"0x0004740000899315","parentaddr":0} +11s
  zigbee-herdsman:controller:log Device '0x0004740000899315' joined +11s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x0004740000899315' accepted by handler +2ms
  zigbee-herdsman:controller:log Interview '0x0004740000899315' start +0ms
```

Koenkk, 2019-12-16:
It seems you are not on the latest herdsman, after updating to 1.8.0 did you do a `rm -rf node_modules && npm install` in the zigbee2mqtt directory?

n1zzo, 2019-12-16:
You are right, sorry for the delay,
here's a fresh docker installation @ 1.8.0.

```
zigbee2mqtt:info  2019-12-16 20:20:30: Starting interview of 'btcino power meter'
zigbee2mqtt:error 2019-12-16 20:20:30: Not connected to MQTT server!
zigbee2mqtt:error 2019-12-16 20:20:30: Cannot send message: topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"btcino power meter"}}
  zigbee-herdsman:controller:device Interview - start device '0x0004740000899315' +5s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":33837,"nwkaddrofinterest":33837} +15s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,45,132,45,132,35] +15s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +23ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +15s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
zigbee2mqtt:error 2019-12-16 20:20:33: Not connected to MQTT server!
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x0004740000899315', attempt 1 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":33837,"nwkaddrofinterest":33837} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,45,132,45,132,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,45,132,21,147,137,0,0,116,4,0,0,0,85] +745ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,45,132,21,147,137,0,0,116,4,0,0,0,85] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [45,132,21,147,137,0,0,116,4,0,0,0] - 85 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":33837,"extaddr":"0x0004740000899315","parentaddr":0} +11s
  zigbee-herdsman:controller:log Device '0x0004740000899315' joined +11s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x0004740000899315' accepted by handler +1ms
  zigbee-herdsman:controller:log Not interviewing '0x0004740000899315', completed 'false', in progress 'true' +1ms
zigbee2mqtt:error 2019-12-16 20:20:43: Not connected to MQTT server!
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x0004740000899315', attempt 2 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":33837,"nwkaddrofinterest":33837} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,45,132,45,132,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +9s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
zigbee2mqtt:error 2019-12-16 20:20:53: Not connected to MQTT server!
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,45,132,21,147,137,0,0,116,4,0,0,0,85] +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,45,132,21,147,137,0,0,116,4,0,0,0,85] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [45,132,21,147,137,0,0,116,4,0,0,0] - 85 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":33837,"extaddr":"0x0004740000899315","parentaddr":0} +16s
  zigbee-herdsman:controller:log Device '0x0004740000899315' joined +16s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x0004740000899315' accepted by handler +1ms
  zigbee-herdsman:controller:log Not interviewing '0x0004740000899315', completed 'false', in progress 'true' +1ms
  zigbee-herdsman:controller:device Interview - node descriptor request failed for '0x0004740000899315', attempt 3 +10s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":33837,"nwkaddrofinterest":33837} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,45,132,45,132,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
zigbee2mqtt:error 2019-12-16 20:21:03: Not connected to MQTT server!
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,45,132,21,147,137,0,0,116,4,0,0,0,85] +7s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,45,132,21,147,137,0,0,116,4,0,0,0,85] +1ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +2ms
  zigbee-herdsman:controller:log Device '0x0004740000899315' accepted by handler +1ms
  zigbee-herdsman:controller:log Interview '0x0004740000899315' start +0ms
```

If you looked at the capture traces, the only difference seems that the Transport Key is sent just once
instead of twice, in your opinion, could that be the culprit?
I'll try to modify the ZStack firmware to achieve the double Transport Key pairing.

Thank you for the help!

closed 2020-02-21 by stale[bot]