Koenkk/zigbee-herdsman #110: Error message when old Philips Hue bulbs are requested to report status updates

bodiroga, 2019-12-19:
Hi @Koenkk!

As requested in https://github.com/Koenkk/zigbee2mqtt/issues/1064#issuecomment-566697914, I open this issue to track the problem.

The problem is that, after enabling 'reporting' in zigbee2mqtt's configuration file, some old Philips Hue bulbs report that they don't support the feature and zigbee-herdsman ignores the message and just waits until the timeout timer expires. Instead, I propose to catch that message, cancel the reporting timer and print something like "warn - Device 'x' doesn't allow reporting functionality."

Attached to this message you can find the packages captured by a CC2531 sniffer (my main controller is a new CC2652R) and zigbee2mqtt's log file, with debug enabled:

[Wireshark - zigbee2mqtt.zip](https://github.com/Koenkk/zigbee-herdsman/files/3985350/Wireshark.-.zigbee2mqtt.zip)
[zigbee2mqtt.txt](https://github.com/Koenkk/zigbee-herdsman/files/3985356/zigbee2mqtt.txt)

Don't hesitate to ask anything you want, please :wink: 

Thank you for your awesome support and best regards,

Aitor

closed 2020-02-28 by stale[bot]

ndfred, 2020-09-17:
I am seeing the same issue on my old Philips Hue bulbs. Following `0x0017880100db65c0`, the bulb is recognised:

    zigbee2mqtt:info  2019-12-19 20:20:31: philips-white-1 (0x0017880100db65c0): 433714 - Philips Hue Lux A19 bulb E27 (Router)

Reporting is set up:

    zigbee2mqtt:debug 2019-12-19 20:20:32: Setup reporting for '0x0017880100db65c0' - 11 - genOnOff

zigbee-herdsman executes the bind request, this takes 1s:

    2019-12-19T19:20:32.107Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":49932,"srcaddr":"0x0017880100db65c0","srcendpoint":11,"clusterid":6,"dstaddrmode":3,"dstaddress":"0x00124b001caabf2b","dstendpoint":1}
    2019-12-19T19:20:32.107Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,12,195,192,101,219,0,1,136,23,0,11,6,0,3,43,191,170,28,0,75,18,0,1,72]
    2019-12-19T19:20:32.160Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69,254,5]
    2019-12-19T19:20:32.160Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69,254,5]
    2019-12-19T19:20:32.160Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69
    2019-12-19T19:20:32.160Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0}
    2019-12-19T19:20:32.161Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5]
    2019-12-19T19:20:32.161Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [69,196,64,69,1,33,181,20]
    2019-12-19T19:20:32.161Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,64,69,1,33,181,20]
    2019-12-19T19:20:32.161Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [64,69,1,33,181] - 20
    2019-12-19T19:20:32.161Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":17728,"relaycount":1,"relaylist":[46369]}
    2019-12-19T19:20:32.161Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
    2019-12-19T19:20:32.172Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,161,12,195,0,40]
    2019-12-19T19:20:32.172Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,161,12,195,0,40]
    2019-12-19T19:20:32.172Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 161 - [12,195,0] - 40
    2019-12-19T19:20:32.172Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - bindRsp - {"srcaddr":49932,"status":0}

Zigbee2MQTT reports a failure after 10s:

    zigbee2mqtt:error 2019-12-19 20:20:42: Failed to setup reporting for '0x0017880100db65c0' - Error: Timeout - 49932 - 11 - 2 - 6 - 7 after 10000ms
        at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
        at ontimeout (timers.js:436:11)
        at tryOnTimeout (timers.js:300:5)
        at listOnTimeout (timers.js:263:5)
        at Timer.processTimers (timers.js:223:10)

I wonder if `bindRsp` setting `"status":0` is enough to determine that binding has failed?

ndfred, 2020-09-17:
Looks like we are skipping binding / reporting when it is already set up. For the record here is what the whole sequence looks like with a recent Zigbee2MQTT install.

Following `0x0017880102218e1d`, the bulb is recognised:

    Sep 17 21:19:36 bagend npm[7840]: Zigbee2MQTT:info  2020-09-17 21:19:36: living_room_bookshelf_light (0x0017880102218e1d): 8718696548738 - Philips Hue white ambiance E26/E27 (Router)

Reporting is set up, zigbee-herdsman executes the bind followed by the ConfigureReporting request, this takes 1s (with a random message coming back 6s later):

    Sep 17 21:21:06 bagend npm[7840]: Zigbee2MQTT:debug 2020-09-17 21:21:06: Setup reporting for '0x0017880102218e1d' - 11 - genOnOff
    Sep 17 21:21:06 bagend npm[7840]: 2020-09-17T20:21:06.981Z zigbee-herdsman:controller:endpoint Bind 0x0017880102218e1d/11 genOnOff from '0x00124b0018df5b10/1'
    Sep 17 21:21:06 bagend npm[7840]: 2020-09-17T20:21:06.982Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - bindReq - {"dstaddr":29499,"srcaddr":"0x0017880102218e1d","srcendpoint":11,"clusterid":6,"dstaddrmode":3,"dstaddress":"0x00124b0018df5b10","dstendpoint":1}
    Sep 17 21:21:06 bagend npm[7840]: 2020-09-17T20:21:06.982Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,37,33,59,115,29,142,33,2,1,136,23,0,11,6,0,3,16,91,223,24,0,75,18,0,1,175]
    Sep 17 21:21:06 bagend npm[7840]: 2020-09-17T20:21:06.994Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,33,0,69]
    Sep 17 21:21:06 bagend npm[7840]: 2020-09-17T20:21:06.995Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,33,0,69]
    Sep 17 21:21:06 bagend npm[7840]: 2020-09-17T20:21:06.995Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 33 - [0] - 69
    Sep 17 21:21:06 bagend npm[7840]: 2020-09-17T20:21:06.995Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - bindReq - {"status":0}
    Sep 17 21:21:06 bagend npm[7840]: 2020-09-17T20:21:06.996Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.019Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,59,115,0,202]
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.019Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,59,115,0,202]
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.019Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [59,115,0] - 202
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.019Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":29499,"relaycount":0,"relaylist":[]}
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.020Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.105Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,59,115,0,202]
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.105Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,59,115,0,202]
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.105Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [59,115,0] - 202
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.105Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":29499,"relaycount":0,"relaylist":[]}
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.105Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.189Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,161,59,115,0,175]
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.189Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,161,59,115,0,175]
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.190Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 161 - [59,115,0] - 175
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.190Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - bindRsp - {"srcaddr":29499,"status":0}
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.190Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.192Z zigbee-herdsman:controller:endpoint ConfigureReporting 0x0017880102218e1d/11 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.192Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0017880102218e1d:29499/11 (0,0)
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.193Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":29499,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":12,"options":0,"radius":30,"len":11,"data":{"type":"Buffer","data":[16,7,6,0,0,0,16,0,0,44,1]}}
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.194Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,21,36,1,59,115,11,1,6,0,12,0,30,11,16,7,6,0,0,0,16,0,0,44,1,65]
    Sep 17 21:21:07 bagend npm[7840]: 2020-09-17T20:21:07.206Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
[... cut ...] DefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 29499 - 11 - 7 - 6 - 7 after 10000ms)
    Sep 17 21:21:30 bagend npm[7840]:     at Timeout._onTimeout (/srv/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    Sep 17 21:21:30 bagend npm[7840]:     at listOnTimeout (internal/timers.js:549:17)
    Sep 17 21:21:30 bagend npm[7840]:     at processTimers (internal/timers.js:492:7)

ndfred, 2020-09-17:
I managed to get around the issue by hardcoding the Philips Hue bulb models that were failing to report that way, works well but obviously too broad a fix to be reasonably upstreamed (see [devices.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js) for reference):

    diff --git a/lib/extension/report.js b/lib/extension/report.js
    index 2cfb04c..05d7b24 100644
    --- a/lib/extension/report.js
    +++ b/lib/extension/report.js
    @@ -16,6 +16,9 @@ const devicesNotSupportingReporting = [
         zigbeeHerdsmanConverters.devices.find((d) => d.model === 'BASICZBR3'),
         zigbeeHerdsmanConverters.devices.find((d) => d.model === 'ZM-CSW032-D'),
         zigbeeHerdsmanConverters.devices.find((d) => d.model === 'TS0001'),
    +    zigbeeHerdsmanConverters.devices.find((d) => d.model === '8718696548738'),
    +    zigbeeHerdsmanConverters.devices.find((d) => d.model === '7299355PH'),
    +    zigbeeHerdsmanConverters.devices.find((d) => d.model === '9290012573A'),
     ];
 
     const reportKey = 1;

sjorge, 2020-09-18:
@Koenkk if I am not mistaken we have a try/catch for reporting somewhere for a different device already.

I would say we do the same for the hue bulbs? Try to configure it, this we have reporting on older bulbs with old firmwares and newer devices that do support some reporting. But don't throw exceptions.

ndfred, 2020-09-18:
@Koenkk: I just meant to share my hacky local fix if this behaviour bothers other people, I didn’t mean to suggest that upstream. Catching that the device doesn’t support reporting and logging that would work, but I would actually either blacklist these devices after the first time we see that behaviour or allow to configure bulbs not to use reporting in the configuration file.

Another alternative is what you are suggesting: blacklist a series of firmware versions, though that could get very long really quickly.

Koenkk, 2020-09-19:
@ndfred can you share the `data/database.db` of this bulb that does not support reporting?

ndfred, 2020-09-21:
Sure, I just looked through them and there are actually quite a few, I'll share one database entry for each model assuming they are all the same. I upgraded them all to the latest firmware they support. The color bulbs and light strip are really ancient, the white ambiance bulbs are more recent purchases.

8718696548738 - Philips Hue white ambiance E26/E27 (Router):

    {
      "id": 17,
      "type": "Router",
      "ieeeAddr": "0x0017880102218e1d",
      "nwkAddr": 29499,
      "manufId": 4107,
      "manufName": "Philips",
      "powerSource": "Mains (single phase)",
      "modelId": "LTW001",
      "epList": [
        11,
        242
      ],
      "endpoints": {
        "11": {
          "profId": 49246,
          "epId": 11,
          "devId": 544,
          "inClusterList": [
            0,
            3,
            4,
            5,
            6,
            8,
            4096,
            768
          ],
          "outClusterList": [
            25
          ],
          "clusters": {
            "genBasic": {
              "attributes": {
                "modelId": "LTW001",
                "manufacturerName": "Philips",
                "powerSource": 1,
                "zclVersion": 1,
                "appVersion": 2,
                "stackVersion": 1,
                "hwVersion": 20,
                "dateCode": "20191216",
                "swBuildId": "5.130.1.30000"
              }
            }
          },
          "binds": [
            {
              "cluster": 6,
              "type": "endpoint",
              "deviceIeeeAddress": "0x00124b0018df5b10",
              "endpointID": 1
            }
          ]
        },
        "242": {
          "profId": 41440,
          "epId": 242,
          "devId": 97,
          "inClusterList": [
            33
          ],
          "outClusterList": [
            33
          ],
          "clusters": {
[... 171 lines cut ...]
          "devId": 97,
          "inClusterList": [
            33
          ],
          "outClusterList": [
            33
          ],
          "clusters": {
          },
          "binds": [

          ]
        }
      },
      "appVersion": 2,
      "stackVersion": 1,
      "hwVersion": 5,
      "dateCode": "20181129",
      "swBuildId": "5.127.1.26581",
      "zclVersion": 1,
      "interviewCompleted": true,
      "meta": {
      },
      "lastSeen": 1600629356487
    }

Koenkk, 2020-09-21:
And non of them support reporting, right? 

@sjorge I want to block setting up reporting based on the `swBuildId` meaning we would block `5.127.1.26581` and `5.130.1.30000`, does this match with your case?

sjorge, 2020-09-21:
Not near the bulb, but only the very early firmware revision has reporting for genOnOff for me, so blocking it on anything from 2019 onwards seems fine I think.

~ sjorge

ndfred, 2020-09-21:
Hue bulbs have auto update for Zigbee2MQTT right? If that is the case, I guess we can assume people who care enough to run reporting would have the latest firmware.

As a side note, it would be nice to report a failure to set reporting immediately so people can see that in the logs right away, it took a while for me to have all the bulbs come back with a timeout failure. I don't know if that clogs some queue somewhere and delays other commands too.

Koenkk, 2020-09-22:
@ndfred devices are never automatically updated. You always need to execute a command yourself to do this.

In the latest dev branch reporting will not be set up anymore for the devices with your softwareBuildID.

ndfred, 2020-09-23:
Just checked and it is all working well, thanks!