Koenkk/zigbee-herdsman #113: Feature request to take over an existing network of Zigbee devices.

TheNetStriker, 2019-12-21:
I have an existing network of 32 Zigbee devices and I don't wanted to re-pair all of the devices because I'm testing different coodinators and also different controller software to find the one combination that works the best for my network. So I looked for a way to take over existing Zigbee devices to the zigbee2mqtt database and I found a simple solution for this.

I simply made the following change to the onDeviceAnnounce method in the controller.js:
```
onDeviceAnnounce(payload) {
    debug.log(`Device announce '${payload.ieeeAddr}'`);
    const device = model_1.Device.byIeeeAddr(payload.ieeeAddr);

    if (!device) {
        this.onDeviceJoined(payload);
    }
```

The only thing I added are the last three lines. With this I only had to unplug and re-plug every lamp a single time and it was automatically added to the database. Also the interview of the lamps worked without any issues. Only sometimes the interview failed and in this case I removed the lamp from the database and added it again and then it worked.

Please check if this change could be added to the project. I've tested this with Philipps Hue and Innr lamps and with those devices it worked perfectly. I also tested this with the CC2531 and the CC1352P-2. With the CC1352P-2 I simply had to unplug every device before starting zigbee2mqtt because on first startup the device listens if there are any messages with the configured pan id and aborts if any messages are sent within 30 seconds. So this could also be used to switch between coordinators. Maybe this could also be enabled with an additional setting in the configuration.

timdonovanuk, 2019-12-22:
I'm about to upgrade to a CC1352P-2 and the thought of having to repair everything is puttinbg me off doing it! Are you telling me with this modification, devices will just rejoin the network but with a new coordinator? How does this work for other devices (not lamps/bulbs) like temperature probes, buttons, door sensors etc?

TheNetStriker, 2019-12-22:
@timdonovanuk Yes that' how it worked when I upgraded to the CC1352P-2. As far as I know the only difference when pairing a device is that the network key is exchanged with the device. The coordinator itself does not have a list of paired devices. So when the Zigbee device already has the correct network key re-pairing a device should not be necessary.

Sadly I only have light bulbs, but I guess this should work with all Zigbee devices. You can just add the 3 lines to the controller.js and test this. Just restart zigbee2mqtt after the change, unplug and re-plug the device to the mains power and the device should be added to the database.

dede34fr, 2019-12-29:
Were do u find this controller.js file please?
And is there any comment from the zigbee2mqtt dev team about this solution?

TheNetStriker, 2019-12-30:
@dede34fr You can find the file in the subdirectory "node_modules/zigbee-herdsman/dist/controller".

There is no comment from the dev team yet.

@Koenkk Can you please take a look at this? This could be very helpful.

Koenkk, 2020-01-05:
This workaround might indeed work for most devices, when starting your new coordinator you have to make sure all zigbee devices are powered down (it should not find an existing network). I think this can already be done without any code changes by making a backup of database.db, start and stop zigbee2mqtt with the new coordinator, restore database.db and start again

jarrah31, 2020-01-12:
I have a large 54 device network with a need to add more. I’m having stability issues and so would also like to upgrade to a CC1352-2, but as @timdonovanuk says, repairing puts me off. I would be very interested to hear if anyone with xiaomi door, motion, temp devices can swap to this new coordinator without repairing, and what steps you took please.

dede34fr, 2020-01-12:
I would love that too. Working with a CC2531 for dev/test but switching to production environment with CC1352P-2. Would like to secure and ease migration while still using both environments in parallel ideally.

HA-TB303, 2020-01-19:
Same for me. I got the cc1352p but I was mostly unable to repair my ikea bulbs.

I have another 23 sensors and the need to re-pair puts me off also.

In stead of adding new features propper and easy migration should get higher priority in my opinion.

jimtng, 2020-02-13:
> I think this can already be done without any code changes by making a backup of database.db, start and stop zigbee2mqtt with the new coordinator, restore database.db and start again

@Koenkk did you mean that the 3 additional lines aren't needed if we renamed database.db, start, stop, and rename back?

Koenkk, 2020-02-13:
yes

ginkel, 2020-03-05:
Is my assumption correct that this won't work when switching from Zigbee 1.2 to 3.0?

Koenkk, 2020-03-05:
@ginkel it won't work, in case it does it's a hack and may cause problems later on.

closed 2020-05-13 by stale[bot]