Koenkk/zigbee-herdsman #118: Pairing of Philips HUE bulbs not possible

turboproc, 2019-12-31:
Having done a factory reset of a Philips HUE bulb, the bulb refused to be paired. enabling the debug level and after adding some additional output it appeared that the pairing process failed due to timeout in the following code snippet in controller/model/device.tc

>             // Split into chunks of 3, otherwise some devices fail to respond.
            for (const chunk of ArraySplitChunks(Object.keys(Device.ReportablePropertiesMapping), 3)) {
                const result = await endpoint.read('genBasic', chunk);
                for (const [key, value] of Object.entries(result)) {
                    Device.ReportablePropertiesMapping[key].set(value, this);
                }

Getting through 2 itterations in the outer loop, the process would fail on the 3rd one and eventually timeout. Changing the value 3 into 2 (creating smaller chunks) solved the issue for now. Not sure about the root cause and whether this is the final solution.

Philips Hue bulb is of model "9290011370B"

Koenkk, 2020-01-06:
It works fine for me and many others, is this the root issue: https://github.com/Koenkk/zigbee2mqtt/issues/2624#issuecomment-569738558 ?

turboproc, 2020-01-06:
Well, to be honest I was expecting it would be working with others without issues otherwise it would have been noticed (and solved) before. At the same time it didn't in my case. I would have to go through the pairing process again to check whether it's related to the above issue. Let's close the issue for now.

closed 2020-01-06 by turboproc

Koenkk, 2020-01-17:
It actually seems to solve some issues, see https://github.com/Koenkk/zigbee2mqtt/issues/2759#issuecomment-575682168.   Will integrate ths