Koenkk/zigbee-herdsman #121: [Request] Include type definitions

anthonynichols, 2020-01-07:
First of all, thank you for the work on this library. I was considering picking and continuing work on the previously stale/abandoned libraries that influenced this one myself, so it is great to see active development from others in the community. Secondly, thank you for using Typescript! As a fellow Typescript user, I was glad to see `.ts` files in the `src` directory.

Could you generate type definitions and publish them along with your transpiled files? I would love to be able to use this in my Typescript project and have all of the type information.

Thanks!

Koenkk, 2020-01-08:
@anthonynichols I haven't looked into this but I don't know how to do it, probably you have more experience and can do this faster, could you make a pr?

anthonynichols, 2020-01-09:
Sure, I'd be happy to. PR incoming...

anthonynichols, 2020-01-09:
https://github.com/Koenkk/zigbee-herdsman/pull/123

closed 2020-01-09 by Koenkk