Koenkk/zigbee-herdsman #122: How to respond to `timeStatus` requests?

markuspf, 2020-01-07:
I have a device (https://github.com/Koenkk/zigbee2mqtt/issues/2648) that I am trying to hack up support for.

I asks for `timeStatus" in the `genTime` cluster. There is already a response to `time`, and I tried hacking in a response to `timeStatus` by extending the code for `time`.

I don't seem to be doing it right, as it doesn't seem the device is satisfied with the answer (and I can't really verify whether the answer is sent out correctly). I'd be grateful for any pointers how to do this correctly.

Koenkk, 2020-01-08:
You can respond to it here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/controller.ts#L509 and very if using sniffing https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

markuspf, 2020-01-10:
Ok, maybe lets first find out whether I am doing the code bit correctly, as that place is exactly where I was trying to respond before; 

I added:

```
if(data.includes("timeStatus")) {
  debug.log("responding to timeStatus: ");
  const resps = 5; // master, master for time zone and DST
  try { 
    yield endpoint.readResponse(frame.Cluster.ID, frame.Header.transactionSequenceNumber, {"timeStatus": { value: resps, type: 24 }} );
   }
  catch (error) {
    debug.log(error); 
    debug.error(`genTime response to ${device.ieeeAddr} failed`);
  }
}
```

In particular I am not sure what the structure of the response object passed to `readResponse` is supposed to be.

Koenkk, 2020-01-13:
It should be 

```js
    yield endpoint.readResponse(frame.Cluster.ID, frame.Header.transactionSequenceNumber, {timeStatus: resps} );
```

sjorge, 2020-01-21:
I think this is supported now?

Koenkk, 2020-01-21:
Yes

closed 2020-01-21 by Koenkk