Koenkk/zigbee-herdsman #127: New Livolo switch and malformed packet

kirovilya, 2020-01-16:
I try to use new Livolo switch https://aliexpress.ru/item/4000385782177.html

Wireshark log [livolo_switch.zip](https://github.com/Koenkk/zigbee-herdsman/files/4073227/livolo_switch.zip)

But in log I see malformed packet when Livolo Gateway send command to turn switch on/off.
I try to repeat this action in converters:
```
    await entity.command('genOnOff', 'toggle', {}, {});
    const payload = {0x0001: {value, type: 0}};
    await entity.write('genPowerCfg', payload, {manufacturerCode: 0x1ad2, disableDefaultResponse: true});
```
And got result logs [livolo_switch_herdsman.zip](https://github.com/Koenkk/zigbee-herdsman/files/4073240/livolo_switch_herdsman.zip)

I can not repeat payload data for command. What we can do for this?

kirovilya, 2020-01-16:
I can’t choose the type of value to send an array of zeros

Koenkk, 2020-01-17:
What is the ID in wireshark of the on off command?

kirovilya, 2020-01-17:
[image]
[image]

401 and down...
425, 448, 498

Koenkk, 2020-01-17:
I've created an example for the on and off: https://github.com/Koenkk/zigbee-herdsman/commit/dec27ed0b027671167d49b134945b4830bfbea24#diff-8158e442a03e3ab9574aeb1a2259d330R609 (requires latest herdsman)

so this would translate to:

```js
    await entity.command('genOnOff', 'toggle', {}, {});
    const payloadOn = {0x0001: {value: Buffer.from([1, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
    const payloadOff = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
    await entity.write('genPowerCfg', payloadOn OR payloadOff, {manufacturerCode: 0x1ad2, disableDefaultResponse: true});
```

kirovilya, 2020-01-18:
Thank you, I will try it tomorrow

kirovilya, 2020-01-19:
It is better now, but
[image]
frame header not equal...
switch not toggle :( 
[livolo_switch_herdsman2.zip](https://github.com/Koenkk/zigbee-herdsman/files/4083105/livolo_switch_herdsman2.zip)

Koenkk, 2020-01-20:
Can you change https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/endpoint.ts#L188 `Zcl.Direction.CLIENT_TO_SERVER` to `Zcl.Direction.SERVER_TO_CLIENT`, if this works I will provide a proper solution to pass it.

kirovilya, 2020-01-20:
I've done everything. Now even the leading bits match. But switching does not work ... :(
Livolo may have done something tricky at the pairing stage that the device does not accept commands...
[livolo_switch_herdsman3.zip](https://github.com/Koenkk/zigbee-herdsman/files/4087355/livolo_switch_herdsman3.zip)

kirovilya, 2020-01-20:
even reset the sequence numbers on the toggle command ...

Koenkk, 2020-01-20:
Strange, it's really annoying that Livolo doesn't respect the ZCL, this makes it very hard to integrate.

sjorge, 2020-01-21:
> Strange, it's really annoying that Livolo doesn't respect the ZCL, this makes it very hard to integrate.

So I guess the Livolo devices are also not zigbee certified?

Koenkk, 2020-01-21:
@sjorge definitely not, in terms of zigbee compliance, these are the worst devices I've ever seen.

closed 2020-03-28 by stale[bot]