Koenkk/zigbee-herdsman #134: Reflections on the interview process

kirovilya, 2020-01-31:
In the new herdsman, the interview process began to interrupt very often. In the previous (shepherd) version, there was no such thing, but there were other problems.

I think we need to do a renewable interview.
1. After a successful announcement, the device is already in network and online.
2. Now we need to read the model, manufacturer and type. This is enough to identify the device in the converters database and start working with the device.
3. If errors occur at the previous stage, then this is an occasion to send a message to the device about the need to leave the network and remove it from the database.

4. Now we can request endpoints and their parameters. If we expand the database of devices, then we can not request this data, but read from converters (this is an offer for the next versions of converters).
5. If errors occur in reading endpoints, the device may remain on the network and we need to schedule a survey of endpoints as soon as possible https://github.com/Koenkk/zigbee-herdsman/issues/104
6. Enroll IAS device - I think this is not a general operation, but individual for the device, like the configure block in converters. It seems to me that we can combine them or make configure at the same moment (or even earlier than this).
7. As in point 5, if errors occur, the device may remain on the network and we need to schedule a survey of endpoints at the earliest opportunity https://github.com/Koenkk/zigbee-herdsman/issues/104

In case of errors in the steps from the 4th, the device must be identified as working and receive messages, and can even be controlled.

Total:
It is necessary to allow several times to complete the steps from the 4th until all are completed. Configure should be part of the interview. Do schedule delayed actions for battery devices.

Koenkk, 2020-01-31:
> the interview process began to interrupt very often.

Can you provide the logs of this?

> After a successful announcement, the device is already in network and online.

What does it have to do with the interview procedure?

> Now we need to read the model, manufacturer and type. This is enough to identify the device in the converters database and start working with the device.

Note that we can only read the modelID, etc. **after** we know what endpoints the device has, so in my opinion the attributes can only be read after retrieving the descriptor and active endpoints.

> Enroll IAS device - I think this is not a general operation, but individual for the device, like the configure block in converters. It seems to me that we can combine them or make configure at the same moment (or even earlier than this).

What would be the advantage of having this in `configure`? I think it's actually handy to have it in herdsman as unsupported devices will immediately start working now.

kirovilya, 2020-01-31:
> Can you provide the logs of this?

No, I see this on emerging issues.

> What does it have to do with the interview procedure?
> Note that we can only read the modelID, etc. after we know what endpoints the device has, so in my opinion the attributes can only be read after retrieving the descriptor and active endpoints.

Hmm. You agree, yes, I forgot that we must first find out at least one endpoint from which to read the model ... :(

> What would be the advantage of having this in configure? I think it's actually handy to have it in herdsman as unsupported devices will immediately start working now.

sometimes it seems to me that this can not be applied to some devices, or can be delayed or optional, like "configure".
those, even without this block, we can assume that the interview has passed

Koenkk, 2020-01-31:
I think we should investigate per issue were it goes wrong in the interview procedure and how it can be fixed (I already fixed several of these issues last weeks).

closed 2020-02-05 by kirovilya