Koenkk/zigbee-herdsman #138: Cannot add Innr plug SP-120 to network

twsI, 2020-02-21:
I just bought a [Innr SP-120](https://www.zigbee2mqtt.io/devices/SP_120.html), which should be supported, thinking it was just an other plug and play object. But I can't manage to attach it to the network! There is simply no log whatsoever. Has anyone managed to do so without the proprietary gateway? If so, how did you do it?

Hypnos3, 2020-02-21:
The search mode is the join enabled mode of zigbee-herdsman.

For me the Innr SP-120 are working and CC2531 USB stick with Z-Stack 1.2 (default and source_routing) without any problem.

I only had to join them again last weekend and no problem. To get them joined several attempts (1-3) are necessary. Compared to Osram Plugs, that's great (there I hab one plug needed 8 attempts).

If the plug was joined already there is a factory reset necessary which will be done by long press the button.

twsI, 2020-02-22:
Alright, thats weird. Join was of course enabled and the plug was roughly 1m away from my CC2531, but I'll try moving it closer. My Osram plugs were by far less complicated...

twsI, 2020-02-22:
Grabbing an extension cord and moving the plug to the coordinator yielded the following log. Any further suggestions?
```
[32mzigbee2mqtt:info [39m 2020-02-22 02:21:30: Device '0xXXXXXXXXXXXXf291' joined
[32mzigbee2mqtt:info [39m 2020-02-22 02:21:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xXXXXXXXXXXXXf291"}}'
[32mzigbee2mqtt:info [39m 2020-02-22 02:21:30: Starting interview of '0xXXXXXXXXXXXXf291'
[32mzigbee2mqtt:info [39m 2020-02-22 02:21:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xXXXXXXXXXXXXf291"}}'
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: Received Zigbee message from '0xXXXXXXXXXXXXf291', type 'readResponse', cluster 'genBasic', data '{"modelId":"SP 120","manufacturerName":"innr"}' from endpoint 1 with groupID 0
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SP 120","manufacturerName":"innr"}'
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: Received Zigbee message from '0xXXXXXXXXXXXXf291', type 'readResponse', cluster 'genBasic', data '{"powerSource":1,"zclVersion":2}' from endpoint 1 with groupID 0
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1,"zclVersion":2}'
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: Received Zigbee message from '0xXXXXXXXXXXXXf291', type 'readResponse', cluster 'genBasic', data '{"appVersion":1,"stackVersion":2}' from endpoint 1 with groupID 0
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":1,"stackVersion":2}'
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: Received Zigbee message from '0xXXXXXXXXXXXXf291', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1,"dateCode":"20171027-100"}' from endpoint 1 with groupID 0
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1,"dateCode":"20171027-100"}'
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: Received Zigbee message from '0xXXXXXXXXXXXXf291', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.0"}' from endpoint 1 with groupID 0
[34mzigbee2mqtt:debug[39m 2020-02-22 02:21:31: No converter available for 'SP 120' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"2.0"}'
[32mzigbee2mqtt:info [39m 2020-02-22 02:21:31: Successfully interviewed '0xXXXXXXXXXXXXf291', device has successfully been paired
[32mzigbee2mqtt:info [39m 2020-02-22 02:21:31: Device '0xXXXXXXXXXXXXf291' is supported, identified as: Innr Smart plug (SP 120)
[32mzigbee2mqtt:info [39m 2020-02-22 02:21:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xXXXXXXXXXXXXf291","model":"SP 120","vendor":"Innr","description":"Smart plug","supported":true}}'
[32mzigbee2mqtt:info [39m 2020-02-22 02:21:31: Configuring '0xXXXXXXXXXXXXf291'
[31mzigbee2mqtt:error[39m 2020-02-22 02:21:41: Failed to configure '0xXXXXXXXXXXXXf291', attempt 2 (Error: Bind 0xXXXXXXXXXXXXf291/1 genOnOff from '0xXXXXXXXXXXXX752c/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:235:23)
at Generator.throw (<anonymous>)
at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```

twsI, 2020-02-22:
I don't know if there error above was expected, but after a dozen retries, I finally managed to pair them. For everyone else with the same problem: move them as close as possible and keep trying.

closed 2020-02-22 by twsI