Koenkk/zigbee-herdsman #151: i have a big timeout isse after merge

arteck, 2020-03-10:
https://github.com/Koenkk/zigbee-herdsman/commit/86e082392d456d1047cd05f193dd81a93a9b78f0

my hue bulbs and other devs are many timeouts with this version.. 
the parameter was too small "const DataConfirmTimeout = 3000;" and i set it too 6000..
thats ok... but i cant find it in new version...

Koenkk, 2020-03-10:
Data confirm timeout is this value (6000) https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/model/endpoint.ts#L497 minus 1000.

arteck, 2020-03-11:
the new .62 version is not better..sry

arteck, 2020-03-12:
other problem is also the he dimmer RWL021 lose the config/connection

arteck, 2020-03-17:
ok new info... I'm not alone with this problem and it's not only with hue .. 
the last correct version is 0.12.57 what works stable.. no connection lost or similar...

switch to version > 0.12.57 .. after some time, some devices are not be reached.. the hue dimmer has no configure and flashes red led

i have here 40 Devices .. i tested CC2538+CC2592 and cc26x2r also .. same problem ..

Koenkk, 2020-03-17:
Did you also check 0.12.63 with the latest cc26x2r firmware? https://github.com/Koenkk/Z-Stack-firmware/blob/develop/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20200312.zip

arteck, 2020-03-18:
what is different ?? to the last one ?? i have here >40 Devices and it's no fun to reconnect all

Koenkk, 2020-03-18:
Check of there is a coordinator_backup.json file in your data directory, if then yes

arteck, 2020-03-18:
yes i have... then reflash and check ...

arteck, 2020-03-19:
So.. very nice..works perfect...big thx

Koenkk, 2020-03-19:
Good!

closed 2020-03-19 by Koenkk

gev, 2020-03-24:
@arteck did you solve the timeout problem? I have the same! Tested on 2531, 2530 and 2538.
during issue no one answer from any device in a network and about 30 seconds later all works as well

samtherecordman, 2020-03-25:
@arteck I'd also like to know how you solved the problem. My device, in-wall dimmer, works but continues to throw this timeout error message.

samtherecordman, 2020-03-25:
@Koenkk Hi Koen, new user joining this great community. Thank you for your hard work!
Wonder if I should open a new issue for a similar timeout error in my setup.