Koenkk/zigbee-herdsman #160: Support for Control4 dimmers

samtherecordman, 2020-03-30:
In my home, I have a bunch of Control4 dimmer switches, but no Control4 hubs. From my research, it seems Control4 dimmers are based on Zigbee Pro protocols. I decided to tinker with these through Z2M, Node-red and CC2531 (coordinator).

I have not been able to have the dimmers join properly, device not supported and ModelID is not read during interview. Actually the interview fails for most of the dimmers, one of the interviews is successful but still no modelID. I bought an Ikea bulb to test and understand Z2M better, and ended up updating the database entry and config.yaml entries for all Control4 dimmers manually. I am now able to send basic state and brightness commands to the dimmers but I keep getting a timeout error after each and every command.

I have attached the (zigbee-herdsman=debug) log for one of the dimmer interviews. 
[interview.log](https://github.com/Koenkk/zigbee-herdsman/files/4405088/interview.log)

and also here is the error after any command:
  zigbee-herdsman:controller:endpoint Command 0x000fff00002c253e/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":20}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 57467 - 1 - 2 - 8 - 11 after 6000ms) +0ms
zigbee2mqtt:error 2020-03-30 15:43:45: Publish 'set' 'state' to 'C4D_TV' failed: 'Error: Command 0x000fff00002c253e/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":20}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 57467 - 1 - 2 - 8 - 11 after 6000ms)'
zigbee2mqtt:debug 2020-03-30 15:43:45: Error: Command 0x000fff00002c253e/1 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":20}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Timeout - 57467 - 1 - 2 - 8 - 11 after 6000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:339:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-03-30 15:43:45: MQTT publish: topic 'z2m/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'C4D_TV' failed: 'Error: Command 0x000fff00002c253e/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":0,\"transtime\":20}, {\"timeout\":6000,\"manufacturerCode\":null,\"disableDefaultResponse\":false}) failed (Error: Timeout - 57467 - 1 - 2 - 8 - 11 after 6000ms)'","meta":{"friendly_name":"C4D_TV"}}'
 

thank you for your help in advance and please let me know if you need other details. I'm still learning this stuff, but willing to try.

Koenkk, 2020-03-31:
- Does the interview always fail at `  zigbee-herdsman:controller:device:log Interview - failed for device '0x000fff000026bbe9' with error 'Error: AREQ - ZDO - simpleDescRsp after 10000ms` ?
- Regarding the timeout you might want to disableDefaultResponses for this device (example: https://github.com/Koenkk/zigbee-herdsman-converters/blob/a42193945e93abc1ce6e40e9190652a18a1a6dd9/devices.js#L2595)

samtherecordman, 2020-04-02:
- Yes, interview has failed for all dimmers at the same spot I believe, except for one dimmer which sometimes has success in finishing the interview process but without a modelID. I'm not familiar with the interview process, but I have removed two dimmers from the DB and re-paired them. The log is included here.  [interview_failed_two_dimmers_20200401.log](https://github.com/Koenkk/zigbee-herdsman/files/4421575/interview_failed_two_dimmers_20200401.log)

- I have set the disableDefaultResponses to true for the device and I do not get the timeout issue anymore. Thank you for that! Does this mean the dimmers never provide any response? I have not been able to issue a {"state":""} for these dimmers yet.

Koenkk, 2020-04-02:
- The device has 3 endpoints: `1,196,197`, for 1 simpleDescriptor is successfully retrieved, for 196 not. Replacing `node_modules/zigbee-herdsman/dist/controller/model/device.js` with https://gist.github.com/Koenkk/5178f686fc254807c7c0594f31dc4ccb should suppress the error.
- It means that they don't provide a ZCL default response (consult the ZCL specification for details about this)

samtherecordman, 2020-04-03:
Thank you for your support and feedback on this!

- I tried your suggestion, but I still get an error message during interview. I wasn't able to save the log as my pi had to be restarted for another reason. I will retry again today and report back. For this code, is it a simple file replace? I couldn't tell what has changed in the code.

- I was able to get state / status from the dimmers, it seems the Control4 Dimmers do follow some of the ZCL specifications. For example, I haven't been able to get group functionality to work with the dimmers.

Koenkk, 2020-04-03:
Yes just replace the file.

samtherecordman, 2020-04-06:
Sorry for the delay, here is the interview failed log for two of the dimmers. I believe it still failed to retrieve descriptors.
To test this, I stopped z2m, removed the dimmers entries from DB file, Reset the dimmers, then initiated announce/pairing on the dimmers. attached is the log. Let me know if the testing procedure was sufficient.
[interview_failed_dimmers_20200406.log](https://github.com/Koenkk/zigbee-herdsman/files/4440012/interview_failed_dimmers_20200406.log)

Koenkk, 2020-04-06:
The issue is that this device defines no `modelID` therefore interview cannot be completed. A workaround is to set this in `database.db` manually.

samtherecordman, 2020-04-07:
It's possible Control4 deliberately hides the modelID, so typical zigbee hubs cannot pair with their devices. You mentioned there are three endpoints, is there a way to check for modelID in the other endpoints?

Koenkk, 2020-04-08:
I've checked the log, but for the other 2 endpoints the device does not answer the simple descriptor request (so we can also not read attributes).

closed 2020-06-14 by stale[bot]