Koenkk/zigbee-herdsman #164: Group add IKEA TRADFRI remote control (E1524/E1810) not possible

ch90045, 2020-04-12:
Hi,

i did try this with 2 different RC's, one was known to zigbee2mqtt before, the other one i just added as new. Before the RC i successfully added 0x000b57fffe8ef780 via mqtt publish

- known one:
zigbee2mqtt:info  2020-04-12 10:45:43: Adding 'Ikea_Fernbedienung_Kueche' to 'Group_Kueche'
....
zigbee2mqtt:info  2020-04-12 10:46:01: MQTT publish: topic 'zigbee2mqtt/Ikea_Fernbedienung_Kueche', payload '{"battery":34,"linkquality":113,"update_available":false,"last_seen":"2020-04-12T10:46:01+02:00","action":"brightness_down_click"}'
zigbee2mqtt:info  2020-04-12 10:46:01: MQTT publish: topic 'zigbee2mqtt/Ikea_Fernbedienung_Kueche', payload '{"battery":34,"linkquality":120,"update_available":false,"last_seen":"2020-04-12T10:46:01+02:00","action":"toggle"}'
zigbee2mqtt:info  2020-04-12 10:46:03: MQTT publish: topic 'zigbee2mqtt/Ikea_Fernbedienung_Kueche', payload '{"battery":34,"linkquality":123,"update_available":false,"last_seen":"2020-04-12T10:46:03+02:00","action":"toggle"}'
zigbee2mqtt:error 2020-04-12 10:46:04: Failed to call 'Groups' 'onMQTTMessage' (Error: Command 0xd0cf5efffef5de72/1 genGroups.add({"groupid":2,"groupname":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 56421 - 1 - 12 - 4 - 0 after 6000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

- new one:
zigbee2mqtt:info  2020-04-12 10:47:36: Device '0x000b57fffec50e51' joined
zigbee2mqtt:info  2020-04-12 10:47:36: Starting interview of '0x000b57fffec50e51'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x000b57fffec50e51"}}'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x000b57fffec50e51"}}'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/0x000b57fffec50e51/availability', payload 'online'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/ZigbeeBridge', payload '{"state":true,"led_state":false,"linkquality":52,"description":"5645/0x680AE2FFFE3F822A","type":"NEIGHBOR","rssi":62,"last_seen":"2020-04-12T10:47:36+02:00"}'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/ZigbeeBridge', payload '{"state":true,"led_state":false,"linkquality":52,"description":"51874/0x680AE2FFFE8FF4A6","type":"NEIGHBOR","rssi":24,"last_seen":"2020-04-12T10:47:36+02:00"}'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/ZigbeeBridge', payload '{"state":true,"led_state":false,"linkquality":49,"description":"17114/0x000B57FFFE8EF780","type":"NEIGHBOR","rssi":52,"last_seen":"2020-04-12T10:47:36+02:00"}'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/ZigbeeBridge', payload '{"state":true,"led_state":false,"linkquality":49,"description":"17114/0x000B57FFFE8EF780","type":"NEIGHBOR","rssi":52,"last_seen":"2020-04-12T10:47:36+02:00"}'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/ZigbeeBridge', payload '{"state":true,"led_state":false,"linkquality":49,"description":"0/0x00124B0018ED1A14","type":"COORD","rssi":48,"last_seen":"2020-04-12T10:47:36+02:00"}'
zigbee2mqtt:info  2020-04-12 10:47:36: MQTT publish: topic 'zigbee2mqtt/ZigbeeBridge', payload '{"state":true,"led_state":false,"linkquality":49,"description":"4496/0x680AE2FFFE2F4E56","type":"CHILD_RFD","rssi":76,"last_seen":"2020-04-12T10:47:36+02:00"}'
zigbee2mqtt:info  2020-04-12 10:47:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x000b57fffec50e51"}}'
zigbee2mqtt:info  2020-04-12 10:48:15: Successfully interviewed '0x000b57fffec50e51', device has successfully been paired
zigbee2mqtt:info  2020-04-12 10:48:15: Device '0x000b57fffec50e51' is supported, identified as: IKEA TRADFRI remote control (E1524/E1810)
zigbee2mqtt:info  2020-04-12 10:48:15: Configuring '0x000b57fffec50e51'
zigbee2mqtt:info  2020-04-12 10:48:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x000b57fffec50e51","model":"E1524/E1810","vendor":"IKEA","description":"TRADFRI remote control","supported":true}}'
zigbee2mqtt:info  2020-04-12 10:48:21: Successfully configured '0x000b57fffec50e51'

I did not press any button on the new one and tried to add that one via:

zigbee2mqtt/bridge/group/Group_Kueche/add 0x000b57fffec50e51

zigbee2mqtt:info  2020-04-12 10:48:48: Adding '0x000b57fffec50e51' to 'Group_Kueche'
zigbee2mqtt:error 2020-04-12 10:48:53: Failed to call 'Groups' 'onMQTTMessage' (Error: Command 0x000b57fffec50e51/1 genGroups.add({"groupid":2,"groupname":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: AREQ - AF - dataConfirm after 5000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

after pressing the buttons i re-tried it and the behaviour did change slightly 

zigbee2mqtt:error 2020-04-12 10:50:22: Failed to call 'Groups' 'onMQTTMessage' (Error: Command 0x000b57fffec50e51/1 genGroups.add({"groupid":2,"groupname":""}, {"timeout":6000,"manufacturerCode":null,"disableDefaultResponse":true}) failed (Error: Timeout - 53344 - 1 - 36 - 4 - 0 after 6000ms)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

Here are the important infos from the config file:

homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.0.148:1884'
  client_id: zigbee_pi
serial:
  port: /dev/ttyACM0
advanced:
  availability_timeout: 600
  log_level: info
  report: true
  last_seen: ISO_8601_local
....
groups:
  '2':
    friendly_name: Group_Kueche
    retain: false
    transition: 2
    optimistic: true
    devices:
      - 0x000b57fffe8ef780/1

ch90045, 2020-04-12:
Just checked the dev branch as well - no change

Chrischi-, 2020-04-12:
@ch90045 
why do you want to put two remotes in one group?

ch90045, 2020-04-12:
I don't, I just tried 2 different ones to prove its not because one maybe faulty device. The group just should be a light and a RC. The light did work, none of the RCs were possible.,

Chrischi-, 2020-04-13:
It doesn't make sense for me to group remote-controls together.
As a binding I can understand it but not as a group.

Maybe you have another/wrong understanding of what a group should do.

I think what you really want is a binding, not a group.

Koenkk, 2020-04-13:
@ch90045 , @Chrischi-  is right, please check https://github.com/Koenkk/zigbee2mqtt/issues/3326#issuecomment-612577963

ch90045, 2020-04-14:
You are both right ;) i mixed up group and bind.  With the bind it was working although i was surprised that the color change was not included (but that was not even mentioned on the device page). Somehow same procedure still does not work for me with same kind of remote with a IKEA ICPSHC24-10EU-IL-1. The bind was confirmed in the messages but using the remote has no effect - any known issues?

closed 2020-06-20 by stale[bot]