Koenkk/zigbee-herdsman #168: [REQUEST] Support for Silicon Labs EFR32MG1 Series 1 Module (EFR32 family) based Zigbee radios using EZSP serial protocol?

Hedda, 2020-04-21:
It is maybe possible for you to port the EZSP (EmberZNet Serial Protocol) code from the bellows library to zigbee-herdsman (in a similar manner to how zigpy-cc was ported from zigbee-herdsman)?

- https://github.com/zigpy/bellows

As I understand bellows itself was based on UG100 official EZSP Reference Guide from Silicon Labs:

- https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

Reason why is that Ebyte is now making an inexpensive development board called "E180-ZG120B-TB" made for testing their E180-ZG120B module which is cheap yet have a powerful Zigbee 3.0 radio

- http://www.ebyte.com/en/product-view-news.aspx?id=896
  - http://www.ebyte.com/en/downpdf.aspx?id=896
- http://www.cdebyte.net/test-board/zigbee-3-0-module-usb-test-board-kits.html

This dev board is currently sold for less than $9 US-dollar on Aliexpress or about twice that on eBay UK:

- https://www.aliexpress.com/item/4000538003718.html
- https://www.ebay.co.uk/itm/EFR32-Zigbee-3-0-Test-Board-USB-Port-2-4GHZ-E180-ZG120B-TB-Test-Kit/233517910970

Looks like you can remove some jumpers to disable the USB converter if want to use serial directly.

Ebyte is now making that E180-ZG120B module based on the EFR32MG12 chip from Silicon Labs.

- http://www.ebyte.com/en/product-view-news.aspx?id=842
- http://www.ebyte.com/en/downpdf.aspx?id=842

Both E180-ZG120A and E180-ZG120B modules are now sold on eBay and Aliexpress at low prices. 

- https://www.aliexpress.com/wholesale?catId=0&SearchText=%22E180-ZG120B%22

An example is Ebyte E180-ZG120B module sold in Cojxu store on Aliexpress: 

- https://www.aliexpress.com/item/4000473346234.html

E180-ZG120B has 20dbm powerful Zigbee 3.0 radio capable based on an EFR32MG1 Series 1 MCU SoC / chip from Silicon Labs EFR32 family which I understand also be classified as an EmberZNet / Ember based radios using the EZSP serial protocol (UART bus) interface so could, therefore, be made compatible if ported the EZSP code from bellows library for zigpy if only the E180-ZG120B module is flashed with the right firmware?

- https://www.silabs.com/wireless/zigbee/efr32mg1-series-1-socs
- https://www.silabs.com/documents/public/data-sheets/efr32mg1-datasheet.pdf

I guess that my follow-up question will be which exact firmware to use on the E180-ZG120B module?

Hedda, 2020-04-21:
Understand adapter interface needs to be implemented in zigbee-herdsman for zigbee2mqtt support? 

https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts

This is how similar support for ConBee and RaspBee adapters are now being added to zigbee-herdsman

https://github.com/Koenkk/zigbee-herdsman/issues/72

@Koenkk if you do not already have one of these dev boards could I maybe offer to donate one to you?

Hedda, 2020-04-21:
FYI, Silicon Labs also has an official devkit called "EFR32 Mighty Gecko Wireless Starter Kit", but it cost almost $500 US-dollars ($499 to be precise), however, it then contains three complete dev boards.

- https://www.silabs.com/products/development-tools/wireless/mesh-networking/mighty-gecko-starter-kit
- https://www.silabs.com/wireless/zigbee/efr32mg1-series-1-socs/device.efr32mg1b132f256gm32#software
- https://www.digikey.com/catalog/en/partgroup/efr32-mighty-gecko-wireless-soc-starter-kit/59471
- https://www.mouser.se/new/silicon-labs/silabs-mighty-gecko-mesh-devkit/

Hedda, 2020-04-21:
Z-Smart Systems by @cdjackson also has code supporting latest Ember dongles at @zsmartsystems

- https://github.com/zsmartsystems
  - https://github.com/zsmartsystems/com.zsmartsystems.zigbee
    - https://github.com/zsmartsystems/com.zsmartsystems.zigbee.sniffer

The framework implements a package structure that allows efficient use of re-usable components in a number of different applications.

| Package                                           | Description                                          |
|---------------------------------------------------|------------------------------------------------------|
|com.zsmartsystems.zigbee                           |The main framework and cluster library implementation |
|com.zsmartsystems.zigbee.autocode                  |Code generator for the ZigBee cluster library classes |
|com.zsmartsystems.zigbee.dongle.cc2531             |Dongle driver for the Texas Instruments ZNP CC2531    |
|com.zsmartsystems.zigbee.dongle.conbee             |Dongle driver for the Dresden Electronics Conbee      |
|com.zsmartsystems.zigbee.dongle.ember              |Dongle driver for the Silabs EZSP Network Co-Processor|
|com.zsmartsystems.zigbee.dongle.ember.autocode     |Code generator for the Ember NCP dongle commands      |
|com.zsmartsystems.zigbee.dongle.telegesis          |Dongle driver for the Telegesis dongle                |
|com.zsmartsystems.zigbee.dongle.telegesis.autocode |Code generator for the Telegesis dongle commands      |
|com.zsmartsystems.zigbee.dongle.xbee               |Dongle driver for the Digi XBee dongle                | 
|com.zsmartsystems.zigbee.dongle.xbee.autocode      |Code generator for the XBee dongle commands           |
|com.zsmartsystems.zigbee.console                   |Console commands for the general framework            |
|com.zsmartsystems.zigbee.console.ember             |Console commands for the Silabs Ember NCP             |
|com.zsmartsystems.zigbee.console.main              |Main CLI console application                          |
|com.zsmartsystems.zigbee.serial                    |Serial driver implementation                          |
|com.zsmartsystems.zigbee.test                      |Overall tests and code coverage                       |

Koenkk, 2020-04-21:
Adding a new adapter indeed requires the implementation of the adapter.ts.

Note that adding support for a new adapter is a lot of work, given that we already support  2 powerful adapters (conbee and CC2652R/CC1352P-2 which will also available as USB in the future: https://electrolama.com/projects/zig-a-zig-ah/) I don't see a good a time/reward ratio at the moment.

Hedda, 2020-04-21:
That is understandable, hopefully someone else might be willing to take on such a project. My thinking is just that hardware-independence and improved compatibility with multiple Zigbee 3.0 coordinator adapters/dongles from different manufacturers would be beneficial to everyone in the long run. That is, support for more Zigbee 3.0 coordinator adapters/dongles could democratize to the platform, meaning it makes it more accessible to uses around the world, which in turn help drive development with more developers contributing to projects like Zigbee2mqtt in the future. Zigbee2mqtt has certainly already achieved a lot in regards to democratizing the entry point to Zigbee Home Automation with the CC2531 USB-sticks (which I see Itead now even sells pre-flashed compatible coordinator firmware at a low price from China), but improved accessibility in my head also means improved support for enhanced hardware adapters with Zigbee 3.0 at lower prices and perhaps more importantly easier to find pre-flashed with compatible coordinator firmware at low prices all over the world. IMHO all Zigbee based projects will benefit from lower prices for Zigbee 3.0 coordinator hardware and support for multiple adapters/dongles from different manufacturers, both directly and indirectly.

SillyDay, 2020-04-24:
@Hedda Ebyte E180-ZG120A(B) is a Serial port transparent transmission module, running Ebyte private  firmware,not EZSP. (However, it can act as a standard ZigBee3.0 router in the Network) For example, I sent AABBCCDDEEFF(not HEX) to the serial port, then I captured [image]
  In other words, the moudule running a Ebyte offical firmware cant't act as a standard coordinator like CC2652R, while it can connect your serial device to ZigBee network easily.(zigbee-herdsman & 
 zigbee-herdsman-converters modification needed)

Hedda, 2020-04-27:
@SillyDay Sorry if I was unclear. I suspected that something like would be the case, though I believed that it did not come with any application firmware at all, only a bootloader firmware from the factory. 

I am of course suggesting that 'someone' first also need to build a "standard" EmberZNet Zigbee coordinator firmware image for Silicon Labs EFR32MG12 SoCs using the official Simplicity Studio software from Silabs so that we are able to reflash E180-ZG120B modules and E180-ZG120B-TB boards before they will be able to used with zigbee-herdsman and zigbee2mqtt as coordinators.

This is the same principle how zigbee-herdsman work with Texas Instruments chips when you flash those with firmware from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator as the all the CC2531 sniffer USB sticks and CC2652R boards can not be used before you flash them either unless you first flash them with a Z-Stack coordinator firmware.

The main issue regarding building a such "standard" Silicon Labs EmberZNet Zigbee coordinator firmware image for an EFR32MG12 SoC is that in order to to get access to the EmberZNet Zigbee stack from the official Simplicity Studio software you first need to buy and register an official development kit for Zigbee from Silicon Labs, to gain license rights to build a firmware with their Zigbee stack so to speak, like example their official devkit called "EFR32 Mighty Gecko Wireless Starter Kit" (article number: SLWSTK6000B) which cost almost $499 US-dollars, though in their defence that kit does contain three (3) complete dev boards/mainboards with six (6) matching EFR32MG12 Zigbee radio modules designed for development.

- https://www.silabs.com/products/development-tools/wireless/mesh-networking/mighty-gecko-starter-kit

I would have thought that the Ebyte module would most likely only come with standalone bootloader firmware preloaded for boot. Such standalone bootloader firmware usually is just enough to allow you to flash and upgrade the main application firmware via USB (without having to use a JTAG connector). That is, just enough bootloader firmware to allow USB flashing of an application firmware like Silicon Labs EmberZNet Zigbee coordinator firmware image for EFR32MG12 SoCs.

- https://www.silabs.com/documents/public/user-guides/ug103-06-fundamentals-bootloading.pdf

I would think you have to build/compile Zigbee stack (Zigbee EmberZNet) project with coordinator device type config in their SDK ("Simplicity Studio") and then flash application bootloader firmware with Zigbee stack yourself with either "Simplicity Studio" or "Simplicity Commander" or other flashing tools. 

You can use Simplicity Studio (Silicon Labs SDK) for non-Zigbee development without buying an official devkit, but I do not think that you can create an EmberZNet Zigbee coordinator firmware image with it without registering an official Zigbee devkit from Silabs as you will not have access to the EmberZNet? 

- https://www.silabs.com/mcu/programming-options

If and when you do have registered an official Zigbee devkit from Silabs so you have access to the EmberZNet Zigbee stack in Simplicity Studio (Silabs SDK) then I believe that these might be a couple of relatively easier to follow how-to guide for building EmberZNet for EFR32 and EM35x firmware, (as I understand they call protocol such as their EmberZNet Zigbee stack for "apps"). 

- https://www.silabs.com/documents/public/quick-start-guides/qsg106-efr32-zigbee-pro.pdf

- https://www.silabs.com/support/mesh-networking-training/zigbee/zigbee-3-0-dyi-building-a-zigbee-3-0-switch-and-light-from-scratch

I don't own one myself but would imagine that once you bought their official devkit they have made it simple enough start using their latest SDK or other firmware tools (like "Simplicity Studio" and "Simplicity Commander") and it will even detect the hardware once it is plugged in to then start a new project. 

- https://www.silabs.com/products/software

Guessing that you first might need to register a serial number together with dev kit article-number to your account for full SDK access, again like for the official "EFR32 Mighty Gecko Wireless Starter Kit" (that kit part number is "SLWSTK6000B") as that official dev board in that dev kit is kind similar to the Ebyte E180-ZG120B-TB eval board as they both use the same type of EFR32MG12 SoC chip in Silicon Labs EFR32 family.

- https://siliconlabs.force.com/KitRegistration

Simplicity Studio
- https://www.silabs.com/products/development-tools/software/simplicity-studio
  - https://www.silabs.com/documents/public/training/simplicity-studio/simplicity-studio-introduction.pdf
  - https://www.silabs.com/documents/public/application-notes/AN0822-simplicity-studio-user-guide.pdf
  - https://www.silabs.com/support/training/simplicity-studio

Simplicity Commander
- https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2016/07/22/simplicity_commander-65kS
  - https://www.silabs.com/documents/public/user-guides/ug162-simplicity-commander-reference-guide.pdf

Silicon Labs has its community knowledge-base here:

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base

Silicon Labs also have different user forums here:

- https://www.silabs.com/community/wireless/zigbee-and-thread/forum

- https://www.silabs.com/community/software/simplicity-studio

Silicon Labs IoT also has a community on Slack:

https://silabsiot.slack.com/

Silicon Labs IoT also has a community on Gitter:

https://gitter.im/Silabs-IoT

Hedda, 2020-04-27:
Should be noted that Silicon Labs makes it clear that it is OK for multiple developers within a company, organization or team share a single Silabs account for use with their Simplicity Studio SDK. 

So if example "team zigbee2mqtt" or "team zigbee-herdsman" collectively bought an official Silicon Labs Zigbee devkit and register that to a shared account then all developers on that team could,  as long as they login with that same account, use the Simplicity Studio SDK with access to EmberZNet Zigbee stack to build Zigbee coordinator firmware images for the EFR32MG12 SoC chip.

Perhaps Silicon Labs / Silabs would be willing to sponsor Zigbee2mqtt with an official EFR32 devkit?

It would be to their benefit if Zigbee2mqtt via zigbee-herdsman had support for Ember based coordinators as that would both sell more hardware as well as make more developers use/know their software.

Hedda, 2020-04-27:
Another tip is this "Zigbee Boot Camp Course" wiki for Silicon Labs @MarkDing has put on GitHub :

- https://github.com/MarkDing/IoT-Developer-Boot-Camp/wiki/Zigbee-Boot-Camp

- https://github.com/MarkDing/IoT-Developer-Boot-Camp

He has also put together this PDF with Silicon Labs ZigBee Onboarding Roadmap for beginners:

- https://github.com/MarkDing/IoT-Developer-Boot-Camp/wiki/files/ZB-Zigbee-Boot-Camp/Silicon-Labs-ZigBee-Onboarding-Roadmap.pdf

Hedda, 2020-04-27:
@cdjackson from @zsmartsystems also mention two main NCP images in their README.md

- https://github.com/zsmartsystems/com.zsmartsystems.zigbee

  "Currently there are two main NCP images - one that supports hardware flow control with a baud rate of 115200, and one that supports software flow control with a rate of 57600."

Perhaps he would be willing to tip how to build or make EmberZNet Zigbee coordinator firmware for EFR32MG12 SoC chip like the official EFR32 devkit or Ebyte E180-ZG120B / E180-ZG120B-TB?

@cdjackson at least keeps adding new EZSP support from UG100 to com.zsmartsystems.zigbee

- https://github.com/zsmartsystems/com.zsmartsystems.zigbee/pulls?q=ezsp

- https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf

- https://www.silabs.com/documents/public/application-notes/an706-ezsp-uart-host-interfacing-guide.pdf

cdjackson, 2020-04-27:
This statement is probably a little old now - it was made when Silabs provided pre-build firmware images which they no longer do. You now need to compile them for your hardware, so data rates etc are configurable.

It is relatively easy to compile the firmware if you have the SDK etc. I do have a document that I provide to customers but don’t really want to publish it on the net but can answer questions if needed.

Hedda, 2020-04-27:
@cdjackson So in your opinion, once someone has registered an official EFR32 devkit to their account for Silicon Labs and installed + login to Simplicity Studio SDK it should be relatively easy for them to build and compile an EmberZNet PRO Zigbee coordinator firmware image file for an EFR32MG12 based module like the E180-ZG120B-TB evaluation board which then could be shared to everyone so that anyone could flash their own Ebyte E180-ZG120B module based hardware with more freely available software?  ...again similar to how Koenkk share Z-Stack coordinator firmware images?

cdjackson, 2020-04-27:
I’m not familiar with the EByte module so can’t really comment on that. If it is a standard EFR32 under the hood, and doesn’t do anything too non-standard, then yes, it’s reasonably straight forward to create firmware. You will need to know the hardware configuration of the module though (ie what pins of the EFR32 are connected where in the module) - I’m not sure if this information is available?

Hedda, 2020-04-27:
Ebyte does indeed have public documentation with detailed layouts of the modules pins available.

E180-ZG120B module
- http://www.ebyte.com/en/product-view-news.aspx?id=842
  - User Manual PDF with PIN layout & listing: http://www.ebyte.com/en/downpdf.aspx?id=842
  - Alternative URL: http://www.ebyte.com/en/pdf-down.aspx?id=1618

E180-ZG120B-TB evaluation board
- http://www.ebyte.com/en/product-view-news.aspx?id=896
  - User Manual PDF with PIN layout & listing: http://www.ebyte.com/en/downpdf.aspx?id=896

E180-ZG120B

SillyDay, 2020-04-27:
@Hedda I can access the EmberZNet Zigbee stack and try to build a coordinator firmware for EFR32MG1B232F256, but someone have to add support for EmberZNet stack

SillyDay, 2020-04-27:
@Hedda Also, programming and read back the firmware is available for E180 series via JLink. I just  flashed a customized firmware from EBYTE using my JLink V11.

MPM1107, 2020-04-27:
Well Sonoff made this even more interesting since their new bridge uses the EFR32MG21, so implementing this would add it to the list of supported coordinators as well. @SillyDay could you try and build a firmware file for that chip as well? I don't have one of these Ebyte modules but my Sonoff bridge is about to arrive in the mail, so I could start tinkering with it. I do have some access to prebuilt files myself but they're kinda old. If it's of any importance I think the exact chip they use is the EFR32MG21A020F768IM32-B.

Hedda, 2020-04-28:
@SillyDay I understand. 

I would great if you or someone could only build Zigbee coordinator firmware images with latest EmberZNet Zigbee stack for the EFR32MG1 Series 1 SoC inside E180-ZG120B and distribute those image files in a GitHub repo or similar location so that everyone could download it without hunting.  

Maybe similar to how Koenkk distribute Z-Stack coordinator (and router) firmware but in its own repo?

- https://github.com/Koenkk/Z-Stack-firmware/

That would be the first hurdle so that anyone who wants can just buy a cheap E180-ZG120B module or the E180-ZG120B-TB board, download and flash an existing firmware then start developing EZSP interfaces for it (without having to buy the official EFR32 devkit for Zigbee which is relatively expensive). 

Is it possible to build separate "debug" and "production" versions of latest EmberZNet firmware?

Anyway, as @MPM1107 mention sharing EmberZNet coordinator firmware could possibly be a great start for latest EmberZNet with Zigbee 3.0 support in more projects than just zigbee-herdsman (Zigbee2mqtt and IoBroker) as other DIY home automation projects could possible also start to develop their own EZSP interface, as long as both inexpensive hardware and compatible firmware is available.

SillyDay, 2020-04-28:
@Hedda I have finished building a Customized NCP firmware following AN1010: Building a Customized NCP Application, and posted it on https://github.com/SillyDay/EFR32/blob/master/E180-NCP.bin You can flash this firmware via JLink to E180. In addition, this firmware is untested!

Hedda, 2020-04-28:
@MPM1107 FYI, E180-ZG120B is based on EFR32MG12 (EFR32MG1x) making it an EFR32 Series 1.

Sonoff ZBBridge (Sonoff Zigbee Bridge) is based on EFR32MG21 (EFR32MG2x) so it's an EFR32 Series 2.

Confusing naming scheme but as I understand they should still use the same SDK & Zigbee stack, see: 

https://www.silabs.com/documents/public/application-notes/an0918.2-efm32-to-efr32xg2x-migration-guide.pdf

PS: EZR32 is referred to as Series 0 and it also uses older EmberZNet and older EZSP interface version.

Hedda, 2020-04-28:
@SillyDay you should not need to use a JTAG flasher for the E180-ZG120B-TB evaluation board, right?

EFR32 should come with a bootloader from the factory that should allow software flashing over USB?

The previous generation (Silicon Labs ETRX357) adapters, like example Telegesis ETRX357 USB dongles, have plenty of instructions out there on how to flash firmware over USB with only software.

https://github.com/zigpy/bellows/blob/dev/README.md

- Telegesis ETRX357USB (Note! This first have to be flashed with other EmberZNet firmware)
- Telegesis ETRX357USB-LRS (Note! This first have to be flashed with other EmberZNet firmware)
- Telegesis ETRX357USB-LRS+8M (Note! This first have to be flashed with other EmberZNet firmware)

https://community.home-assistant.io/t/eu-usb-sticks-for-the-new-zigbee-component/16718/10

Quote bellow originally posted by Frank Aug '17 in linked thread:

I can confirm that the ETRX357USB-LRS+8M can be flashed with an appropriate EZSP firmware without the use of any programming hardware.

So, without any guarantees (you might -but probably won’t- brick your device), here’s how:

Download the firmware. I used some random blob someone put on github that at least seemed to have the right name. What could possibly go wrong, right? :smile: https://github.com/yqyunjie/Zigbee-Project/blob/master/firmware/EmberZNet/EM35x-EZSP/build/em35x-ezsp-images/EM357/em357-ncp-uart-xon-xoff-use-with-serial-uart-bl-500.ebl?raw=true 282

(Install USB-to-serial drivers for the device. Linux will load the driver automatically when you plug in; not sure about other OSes.)

Install a serial port communication app that supports X-MODEM. (debian/ubuntu: sudo apt-get install minicom) Run it and configure it (sudo minicom -s) to use the fake serial port (/dev/ttyUSB0) at 19200 baud 8N1. Disable both hardware (RTS/CTS) and software (XON/XOFF) flow control.

Type ‘AT’ , you should get OK in response. Now type ‘AT+BLOAD’ . The device will reboot into the bootloader.

Change the baud rate in the serial port communication app setting to 115200 baud. (exit minicom using CTRL-A Q, run ‘sudo minicom -s’ again)

Pressing enter in the terminal should now show you a three-option boot loader menu. Choose option 1.

‘C’ characters will start showing. Don’t wait for this to finish, but start an X-MODEM upload of the firmware you downloaded earlier (use CTRL-A S in minicom). You have 60 seconds to start the upload.

After the upload finished, you should return to the menu. Now select option 2, to reboot into the new firmware. You’re done.

SillyDay, 2020-04-28:
@Hedda No, the bootloader in E180-ZG120 is not a UART XMODEM Bootloader, so you have to use SWD Interface at least once. BTW, the Bootloader is not included in my firmware.

MPM1107, 2020-04-28:
@SillyDay Yeah I checked the migration guide as well and it says somewhere in there that series 1 and 2 are not binary compatible. But you should still be able to run the exact same code/stack on them, so once implemented zigbee2mqtt shouldn't care which of the two it's using. That's why I specifically asked whether you could compile the firmware for that chip as well. It would also be interesting to then test the range/performance difference between the two series.

SillyDay, 2020-04-28:
@MPM1107 Pls let me know the Pin definition of your device, especially TX&RX and you need a NCP firmware for EFR32MG21A020F768IM32 is that right?

MPM1107, 2020-04-29:
@SillyDay Yes, that is the right chip. I did some reversing of the circuitry today, I'm pretty sure I got all of the pins right. Getting hold of documentation for these chinese modules sure is a pain though.

TX: PB01 (connected to ESP8266 U0RXD)
RX: PB00 (connected to ESP8266 U0TXD)
PA00 connected to ESP8266 GPIO5
DRX: PA06
DTX: PA05
SWDIO: PA02
SWCLK: PA01

So long story short, TX is on PB01 and RX is on PB00. SWDIO and SWCLK are the default pins for them so that's no surprise. I'm not sure what DRX and DTX are for, these are just two pads that are broken out of the module as test pads and not actually used for communication with the main processor. Maybe as a debugging uart console? And then there's PA00, which is connected to one of the GPIO pins of the host processor. I'm not sure what that could be useful for, if you have any ideas feel free to think of a function for that pin.

stevew817, 2020-04-29:
@MPM1107 Do you have the possibility to attach a logic analyser to TX, RX and PA0? If there's only one GPIO signal between the two, it might be any of:
* a flow control signal for TX/RX (to avoid buffer overruns in the serial communication)
* a coexistence signal (making sure the two 2.4GHz radios don't try to transmit at the same time)
* a bootloader entry signal, to allow the ESP8266 to upload new firmware to the EFR32

MPM1107, 2020-04-29:
Little update: No changes on that PA0 line in normal operation (even while sending/receiving data), so I'm pretty sure it's for entering the bootloader mode for over the air updates from the host mcu. This is further supported by the fact that there's a physical resistor in place that's pulling the data line up to Vcc which I didn't notice before. Thanks for the suggestions @stevew817

s-hadinger, 2020-06-01:
@MPM1107 @SillyDay I did hooked a logic analyzer to the Sonoff Zigbee bridge and I confirm Sonoff does not use EmberZNet firmware.
https://github.com/arendst/Tasmota/issues/8583

I will try to flash the firmware, as there are clearly 5 programming pads: `Z_RX`, `Z_TX`, `SWDIO`, `SWCLK` and `Z_RST`.
[image]

@MPM1107 I'm not familiar for EFR32 flashing, what do I need? Is JLink through RaspberyPI enough?

s-hadinger, 2020-06-01:
Contrary to my previous message, the Sonoff Zigbee Bridge does come with Ember firmware already loaded, as it shows when listening to `Z_TX`

```
Reset info: 0x03 (EXT)
Extended Reset info: 0x0301 (PIN)
init pass
Build Time Apr 29 2020 17:12:29
EEPROM init: 0x00
EMBER_NETWORK_UP 0x0000
[Z3Coord_callbacks.c:emberAfStackStatusCallback]
status = 0x90
Send Data:
FE 0D 01 80 02 00 00 00 00 01 01 04 74 

Enter password:

InitNetworkEventHandler
PANID = D136
```

Hedda, 2020-06-01:
@s-hadinger It would be absolutely great if that SONOFF ZBBridge (SONOFF Zigbee Bridge) already have Ember firmware already based on EmberZNet Zigbee Stack with their standard EZSP (EmberZNet Serial Protocol) serial protocol interface / API over UART bus.

If it already has Ember firmware with EZSP interfaces then will hopefully not need to worry about any firmware upgrade of in the first place and instead just focus on hacking its ESP8266 instead.

However to answer the question. Yes, If the Zigbee SoC module used in the SONOFF ZBBridge (SONOFF Zigbee Bridge) is at all similar to other EFR32 modules, like the E180-ZG120B by Ebyte, then yes at least according to the Ebyte manual can flash the firmware on EFR32 module with J-Link

- http://www.ebyte.com/en/product-view-news.aspx?id=842
  - http://www.ebyte.com/en/downpdf.aspx?id=842
  - http://www.ebyte.com/en/pdf-down.aspx?id=1618

In the Ebyte E180-ZG120B manual, the burning program sais that GPIO pins for J-Link interface is:
* VCC (Positive Supply Voltage)
* SWCLK (Clock Pin)
* SWDIO (Serial Wire Debug IO)
* GND (Ground)

FYI, there's also a related discussion about flashing firmware on these type of EFR32 modules here:

- https://github.com/zigpy/bellows/issues/24

If still are serious about flashing that firmware then checkout a Silicon Labs document called AN136

- https://silabs-prod.adobecqms.net/documents/public/application-notes/AN136-production-programming-options.pdf

Maybe it is possible to just use the Command-Line Interface (CLI) of Simplicity Commander as it supports using Serial Wire Debug (SWD) or Joint Test Action Group (JTAG) as the target interface?

- https://www.silabs.com/documents/public/user-guides/ug162-simplicity-commander-reference-guide.pdf

AN136 otherwise refers to other documents for many different methods, like SWD, JTAG, and C2:

- AN0062 https://www.silabs.com/documents/public/application-notes/an0062.pdf
- AN105 https://www.silabs.com/documents/public/application-notes/an105.pdf
- AN127https://www.silabs.com/documents/public/application-notes/AN127.pdf

MarkDing who work at Silicon Labs summarized their Serial Wire Debug (SWD) in their forum here:

 - https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2014/10/21/serial_wire_debugs-qKCT

The same @MarkDing also maintains a wiki about development with Silicon Labs IoT MCU SoCs:

- https://github.com/MarkDing/IoT-Developer-Boot-Camp/wiki

Hedda, 2020-06-01:
All, by the way, here are a few of related discussions specifically about hacking SONOFF ZBBridge:

- https://github.com/arendst/Tasmota/issues/8583 (started by @s-hadinger  )
- https://github.com/zigpy/zigpy/issues/385
- https://github.com/xoseperez/espurna/issues/2224
- https://github.com/esphome/feature-requests/issues/688
- https://github.com/1technophile/OpenMQTTGateway/issues/205
- https://github.com/letscontrolit/ESPEasy/issues/3021

s-hadinger, 2020-06-01:
After more digging, the EFR32MG21 on the Sonoff Zigbee Bridge does have Ember software, but unfortunately it does not have EZSP. It looks like Sonoff implemented its own application on the EFR32 and has its proprietary protocol to the ESP. Trying to connect a non-SONOFF zigbee device to the coordinator what caught by the EFR, but was filtered out and never surfaced to the ESP8266.

I'm afraid we need to re-flash a complete EmberZNet + EZSP software anyways.

Hedda, 2020-06-01:
@s-hadinger Good to know, but too bad. Maybe @SillyDay can build a firmware if get pins definition?

Hedda, 2020-06-01:
Makes me wonder though if they really removed/disabled all the standard EZSP APIs in the firmware for the EFR32 MCU when they added proprietary app with its own protocol? As I understand it, the original EmberZNet app has several EZSP APIs, so maybe they just added an additional app and left the original with the EZSP APIs without even removing/disabling those.

s-hadinger, 2020-06-01:
I tried to connect with a JLink but I'm getting an error:

```
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: J-Link EDU Mini V1 compiled May 26 2020 15:04:19
 - Device "EFR32MG21AXXXF1024" selected.
 - Target interface speed: 1000 kHz (Fixed)
 - VTarget = 3.293V
 - Found SW-DP with ID 0x6BA02477
 - DPIDR: 0x6BA02477
 - Scanning AP map to find all available APs
 - AP[3]: Stopped AP scan as end of AP map has been reached
 - AP[0]: AHB-AP (IDR: 0x84770001)
 - AP[1]: APB-AP (IDR: 0x54770002)
 - AP[2]: AHB-AP (IDR: 0x84770001)
 - Iterating through AP map to find AHB-AP to use
 - AP[0]: Skipped. Could not read CPUID register
 - AP[1]: Skipped. Not an AHB-AP
 - AP[2]: Skipped. Could not read CPUID register
 - Found SW-DP with ID 0x6BA02477
 - DPIDR: 0x6BA02477
 - Scanning AP map to find all available APs
 - AP[3]: Stopped AP scan as end of AP map has been reached
 - AP[0]: AHB-AP (IDR: 0x84770001)
 - AP[1]: APB-AP (IDR: 0x54770002)
 - AP[2]: AHB-AP (IDR: 0x84770001)
 - Iterating through AP map to find AHB-AP to use
 - AP[0]: Skipped. Could not read CPUID register
 - AP[1]: Skipped. Not an AHB-AP
 - AP[2]: Skipped. Could not read CPUID register
 - ERROR: Could not find core in Coresight setup
 - ERROR: Failed to connect.
```

It looks similar to error here: https://forum.segger.com/index.php/Thread/6521-SOLVED-Can-t-connect-to-Silicon-Labs-EFR32MG21A010F1024/

Any idea from someone who knows JLink?

stevew817, 2020-06-01:
Debug interface might be locked down.

s-hadinger, 2020-06-01:
Confirmed, Simplicity Studio gives more details. The device is debug-locked, but not erased lock. It means that we can erase it completely and push a new firmware (secure boot is not enabled so firmware doesn't need to be signed):
[image]

MPM1107, 2020-06-01:
I'm guessing that itead has implemented a way to run ota firmware updates, so the gecko bootloader is probably available, right? Correct me if I'm wrong but that should make it possible for anyone to flash new firmware over the esp8266 instead of a debugger probe.

s-hadinger, 2020-06-01:
Yes probably. Interestingly there is a 24C512 chip attached to the ESP8266, which is a **I2C 512KB EEPROM** with 2 POGO pads to flash it. So I guess it's goal is to store a copy of the EFR32 firmware so that the ESP can flash it.

What I'm concerned with is that it requires an unlock key. If we erase the chip, does it erase the bootloader as well?

MPM1107, 2020-06-01:
Oh that's interesting. From what i can see from the pictures the second chip is a FM25Q16A, so flash for the esp8266. That would indeed mean that the EEPROM has soem zigbee function. Another guess would be that it stores information about all of the devices that are allowed to connect to the bridge.
I don't know a lot about these chips, but from the screenshot you posted above I can see that there's no sign key written to the device. My guess is that this means we don't actually have to erase the chip, right? If there's no key to validate the firmware signature we should be able to just push new firmware using the bootloader, without erasing the chip first.

s-hadinger, 2020-06-01:
512KB for Zigbee information would be huge, considering that the SPI Flash is already 4MB. It is surprising that Sonoff didn't use the 4MB Flash instead.

This document describes the Debug-lock behavior: https://www.silabs.com/documents/public/application-notes/an1190-efr32-secure-debug.pdf

Indeed they don't mention that debug-lock prevents flashing new software.

Also erasing the EFR32 does not erase the bootloader as described here: https://www.silabs.com/documents/public/application-notes/an1190-efr32-secure-debug.pdf

Looking at the PCB, there are only 4 wires connected between EFR32 and ESP8266:
- `TX` + `RX` connected to ESP-GPIO1/3 (no surprise)
- `Z_RST` connected to ESP-GPIO4
- another pin connected to ESP-GPIO5. I didn't located which pin on the EFR32 it is connected to, but I hope it triggers the bootloader to enter Flash mode.

MPM1107, 2020-06-01:
Looking at this doc, page 3: [https://www.silabs.com/documents/public/application-notes/an706-ezsp-uart-host-interfacing-guide.pdf](https://www.silabs.com/documents/public/application-notes/an706-ezsp-uart-host-interfacing-guide.pdf) it's almost certain that the GPIO5 connection is for entering bootloader mode. That fact is confirmed even further by the pullup resistor that I can see on that line, and the fact that the only other way of entering bootloader mode is through EZSP communication as far as I can tell from the documentation, and we have already determined that EZSP is not used in this firmware. So itead _has_ to have a data line for entering the bootloader if they want to execute remote firmware updates.

s-hadinger, 2020-06-02:
It makes sense. Next step, I will flash a Tasmota on the ESP and try to enter bootloader mode.

stevew817, 2020-06-02:
> Also erasing the EFR32 does not erase the bootloader as described here: https://www.silabs.com/documents/public/application-notes/an1190-efr32-secure-debug.pdf

Fair warning, erasing the EFR32 series 2 does also erase the bootloader, since the bootloader would be a part of main flash that is mass-erased as part of the 'standard debug unlock' or 'device erase' commands.

Hedda, 2020-06-02:
So could also be possible unlock debug access on EFR32 via CLI from ESP8266 with custom firmware?

That is if debug unlock don't need a custom key to unlock or you can get that key from ITead/Sonoff?

Hedda, 2020-06-02:
Do you think it will be possible to flash from a custom ESP8266 firmware over UART or OTA?

- https://www.silabs.com/mcu/programming-options#programming

Do you think it's maybe possible to access the EFR32's SWD interface from ESP8266 over UART?

- https://www.silabs.com/documents/public/application-notes/an958-mcu-stk-wstk-guide.pdf

And are there any open-source tools that could flash EFx32 over UART or OTA from ESP8266?
 
I only found this discussion which talks about debugging and flashing with OpenOCD and GDB:

- https://www.silabs.com/community/software/simplicity-studio/forum.topic.html/efm32_developmentde-LqeK

Guess I'm getting way ahead but nice to know if anyone has experience with EFR32 or EFM32. Just very excited to see a few DIY hackers interest in this ESP8266-based Zigbee bridge so soon!

s-hadinger, 2020-06-02:
> Do you think it will be possible to flash from a custom ESP8266 firmware over UART or OTA?

That's what we hope with the Gecko bootloader.

> Do you think it's maybe possible to access the EFR32's SWD interface from ESP8266 over UART?

Definitely no. The SWD interface is not electrically connected to the ESP8266. It is only available through the pads.

> And are there any open-source tools that could flash EFx32 over UART or OTA from ESP8266?

The protocol is XMODEM which is fairly simple. If the Gecko bootloader would allow us.

s-hadinger, 2020-06-03:
Sadly I'm currently stuck. I wasn't able to flash Tasmota to the ESP8266EX. After many tentatives and trial/error, it looks like the Flash SPI chip has an issue with sector erase. Some bits are not properly erased and keep flipping. I didn't find anything in the data sheet that could explain it, beyond a defective chip.

I will probably need to unsolder it and replace it. Unless someone has a better idea.

MPM1107, 2020-06-03:
Are you getting any errors while erasing? Also, you say that it has issues with sector erase, did you try a full chip erase as well?

s-hadinger, 2020-06-03:
I'm not getting any error when erasing, but reading back shows that although most bytes are 0xFF, some bits are 0. And among them some bits are flipping between reads.

When I dumped the original Sonoff firmware, I did it twice and got exactly the same result. This makes me think that the corruption is internal to the SPI Flash chip.

When I write only 0x00 to a sector, the result is good. This makes we think the erasing phase is not working well.

Full flash erase completes in 2.1 seconds without error and does nothing. The data sheet says it should last 7 seconds. This is weird too.

MPM1107, 2020-06-03:
What about just trying to flash a "blank" bin file then? Stupid and hacky, but maybe the chip just has some weird ways to erase it that the flash tool doesn't quite understand? If you flash a 0xFF-filled bin and no bits are flipped that should be a good sign that at least the chip is able to hold data properly.

s-hadinger, 2020-06-03:
I tried too. I flashed a 900KB file full of 0xFF. But the problem remains the same which is understandable. Flashing 0xFF means erasing and writing nothing. Only flashing 0x00 means actually writing bits after erasing.

I also tried flashing a 4KB file with only 0xFF. Since esptoool.py zips the content before sending to UART, I had the proof that it was not a UART data corruption during the file transfer.

MPM1107, 2020-06-03:
Ah, so it's a chinese clone of the Winbond flash modules used by many of the little esp8266 boards. Source: [https://chromium.googlesource.com/chromiumos/third_party/flashrom/+/master/flashchips.h](https://chromium.googlesource.com/chromiumos/third_party/flashrom/+/master/flashchips.h)

I'm not quite sure what to say at this point. It would be weird if you got unlucky enough to receive a module with dead flash. It does seem like an error somewhere but who knows where it comes from.

larena1, 2020-06-03:
Maybe try asking Sonoff?

s-hadinger, 2020-06-04:
The number do match with FM25Q16A data sheet:
[image]

Hedda, 2020-06-04:
Again, WeMos D1 (all models?) should have the same type of "ESP8266EX" as Sonoff ZBBridge so could possibly be the perfect hardware for creating alternative development kit platforms for such a project?

As for the Zigbee MCU SoC part it seems to currently be a bit harder to find inexpensive alternative hardware modules with Silicon Labs EFR32 Series 2 (EFR32MG21) MCU SoC if you do not want to by Silabs official EFR32xG21 Wireless Gecko Starter Kit (SLWSTK6006A) for $479 US-dollars on your own:

- https://www.silabs.com/products/development-tools/wireless/efr32xg21-wireless-starter-kit

Though three developers could co-purchase it as that kit contains three complete development boards and EFR32MG21 Zigbee radio modules, plus a bonus is that purchasing the starter kit also it gives your Silabs registered account access to the Zigbee stack which is not publicly available otherwise.

However, if it would be good enough to use a Silicon Labs EFR32 Series 1 (EFR32MG12) module in a DIY development kit instead, if so then check out this Ebyte E180-ZG120B module and the E180-ZG120B-TB (EFR32 Test Board) which can both be found relatively inexpensive from ex. Aliexpress or eBay.

- http://www.ebyte.com/en/product-view-news.aspx?id=842

- http://www.ebyte.com/en/product-view-news.aspx?id=896

At least Silicon Labs seem to use the same SDK and Zigbee stack for all EFR32; EFR32 Mighty Gecko Wireless SoC Series 1 = EFR32MG1/EFR32MG12/EFR32MG13/EFR32MG14, and EFR32 Wireless Gecko Series 2 =EFR32MG21 "Mighty Gecko". Same Zigbee API too however a newer version API.

- http://wless.ru/files/ZigBee/EFR32MG21/EFR32xG21_migration_guide.pdf

Hardware-wise all Wireless Gecko Series 1 MCUs obviously has less RAM and FLASH memory as well as being based on Arm Cortex-M4 instead of an Arm Cortex M33 CPU that Wireless Gecko Series 2 has. 

- https://www.cnx-software.com/2020/04/27/ebyte-e180-zg120b-tb-zigbee-3-0-evaluation-board-features-silicon-labs-efr32mg1b-zigbee-thread-soc/

FYI, alternative hardware platforms for development platforms was also suggested/discussed here:

- https://github.com/arendst/Tasmota/issues/8197

MPM1107, 2020-06-04:
I don't really see a need for an alternative development board, since we're not doing any hardware development here. Having access to the zigbee stack would be nice, but the price of that kit is really high and as far as I know we already have members here that have access to it.

Hedda, 2020-06-05:
FYI, @mtx512 posted now in https://github.com/arendst/Tasmota/issues/8583 saying that he also has access to Silicon Labs SDK and EmberZNet Zigbee Stack and has offered his help to Tasmota developers to build compatible Zigbee coordinator firmware image for the EFR32MG21 NCP that is in the Sonoff ZBBridge hardware. Hopefully, he might also willing to publicly share those firmware images.

s-hadinger, 2020-06-05:
Thanks to @grobasoz we now have the complete pinout map (cross-posting from Tasmota issue):

The complete EFR32 module pinout and ESP8266 connection is as follows:
- `PB01-TX` + `PB00-RX`, pads `ETX`+`ERX`, connected to `GPIO1+3`
- `PA00`, no pad, connected to `GPIO5`
- `PA01`, pad `SWCLK`, not connected to ESP
- `PA02`, pad `SWDIO`, not connected to ESP
- `PA05`, pad `Z_TX/PA05`, not connected to ESP (EFR32 console)
- `PA06`, pad `Z_RX/PA06`, not connected to ESP (EFR32 console)
- `nRST`, pad `Z_RST, connected to `GPIO4`
- no other EFR32 pin connected

Additional pads for ESP8266:
- `IO0` connected to `GPIO0`
- `LOG` connected to `GPIO13`
- `IO15` connected to `GPIO15`
- `SDA` connected to `GPIO12`, I2C connection to 512KB EEPROM
- `SCL` connected to `GPIO14`, I2C connection to 512KB EEPROM
- `E_RST` connected to `EXT_RSTB`
- `KEY` connected to (probably) `GPIO16` (external button)

Gamester17, 2020-06-05:
> Maybe try asking Sonoff?

ITead (Sonoff) as a company has benefited a lot from alternative firmware from the community in the past and there are today more aware of that fact so would not hurt for you to ask.

Suggest that you sent a mail to "Jerry Shi" <jerry.shi 'at' itead.cc> and "Stan Li" <stan.li 'at' itead.cc> and explain that you are an open-source developer from the Tasmota community that is looking into developing an alternative firmware for the Sonoff ZBBridge and are therefore wondering if they could share the key to disable the debug lock. Jerry Shi and Stan Li

I have no affiliations to ITead/Sonoff myself however in the past I have sent a few questions or suggestions to them and got relatively quick replies from Jerry Shi and Stan Li. I think that Jerry is in marketing at ITead so he has to pass on the technical questions to the developers, however, I know he helped the community in the past and since he is in marketing I assume he would especially helpful if it directly or indirectly helps ITead sell more products, which is something Tasmota surely does.

[134 more replies]

closed 2020-07-23 by Koenkk