Koenkk/zigbee-herdsman #170: out-of-bounds access for routing request for Osram Smart+ plugs

Kosta-Github, 2020-05-01:
I am getting the following out-of-bounds errors when doing a routing request via zigbee2mqtt `zigbee2mqtt/bridge/networkmap/routes` for my 2 [Osram Smart+ plugs](https://www.zigbee2mqtt.io/devices/AB3257001NJ.html)

```
zigbee2mqtt    | zigbee2mqtt:info  2020-05-01 23:02:57: Starting network scan (includeRoutes 'true')
zigbee2mqtt    | 2020-05-01T21:02:57.147Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtLqiReq - {"dstaddr":0,"startindex":0}
zigbee2mqtt    | 2020-05-01T21:02:57.149Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,37,49,0,0,0,23]
zigbee2mqtt    | 2020-05-01T21:02:57.160Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,49,0,85]
zigbee2mqtt    | 2020-05-01T21:02:57.160Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,49,0,85]
zigbee2mqtt    | 2020-05-01T21:02:57.161Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 49 - [0] - 85
zigbee2mqtt    | 2020-05-01T21:02:57.162Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtLqiReq - {"status":0}
zigbee2mqtt    | 2020-05-01T21:02:57.163Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-05-01T21:02:57.173Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,72,69,177,0,0,0,3,0,3,221,221,221,221,221,221,221,221,237,229,119,15,0,111,13,0,162,66,18,2,1,53,221,221,221,221,221,221,221,221,117,103,9,10,170,62,176,124,141,181,21,2,1,5,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,21,2,1,1,185]
zigbee2mqtt    | 2020-05-01T21:02:57.174Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,72,69,177,0,0,0,3,0,3,221,221,221,221,221,221,221,221,237,229,119,15,0,111,13,0,162,66,18,2,1,53,221,221,221,221,221,221,221,221,117,103,9,10,170,62,176,124,141,181,21,2,1,5,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,21,2,1,1,185]
zigbee2mqtt    | 2020-05-01T21:02:57.174Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 72 - 2 - 5 - 177 - [0,0,0,3,0,3,221,221,221,221,221,221,221,221,237,229,119,15,0,111,13,0,162,66,18,2,1,53,221,221,221,221,221,221,221,221,117,103,9,10,170,62,176,124,141,181,21,2,1,5,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,21,2,1,1] - 185
zigbee2mqtt    | 2020-05-01T21:02:57.177Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtLqiRsp - {"srcaddr":0,"status":0,"neighbortableentries":3,"startindex":0,"neighborlqilistcount":3,"neighborlqilist":[{"extPandId":"0xdddddddddddddddd","extAddr":"0x000d6f000f77e5ed","nwkAddr":17058,"deviceType":2,"rxOnWhenIdle":0,"relationship":1,"permitJoin":2,"depth":1,"lqi":53},{"extPandId":"0xdddddddddddddddd","extAddr":"0x7cb03eaa0a096775","nwkAddr":46477,"deviceType":1,"rxOnWhenIdle":1,"relationship":1,"permitJoin":2,"depth":1,"lqi":5},{"extPandId":"0xdddddddddddddddd","extAddr":"0x7cb03eaa0a09632f","nwkAddr":55401,"deviceType":1,"rxOnWhenIdle":1,"relationship":1,"permitJoin":2,"depth":1,"lqi":1}]}
zigbee2mqtt    | 2020-05-01T21:02:57.178Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-05-01T21:02:57.186Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtRtgReq - {"dstaddr":0,"startindex":0}
zigbee2mqtt    | 2020-05-01T21:02:57.187Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,37,50,0,0,0,20]
zigbee2mqtt    | 2020-05-01T21:02:57.196Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,50,0,86]
zigbee2mqtt    | 2020-05-01T21:02:57.197Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,50,0,86]
zigbee2mqtt    | 2020-05-01T21:02:57.198Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 50 - [0] - 86
zigbee2mqtt    | 2020-05-01T21:02:57.199Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtRtgReq - {"status":0}
zigbee2mqtt    | 2020-05-01T21:02:57.200Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-05-01T21:02:57.204Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,41,69,178,0,0,0,7,0,7,23,176,0,105,216,141,181,0,141,181,105,216,0,105,216,162,246,0,141,181,55,119,0,141,181,233,184,0,105,216,169,106,0,141,181,199]
zigbee2mqtt    | 2020-05-01T21:02:57.205Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,41,69,178,0,0,0,7,0,7,23,176,0,105,216,141,181,0,141,181,105,216,0,105,216,162,246,0,141,181,55,119,0,141,181,233,184,0,105,216,169,106,0,141,181,199]
zigbee2mqtt    | 2020-05-01T21:02:57.206Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 41 - 2 - 5 - 178 - [0,0,0,7,0,7,23,176,0,105,216,141,181,0,141,181,105,216,0,105,216,162,246,0,141,181,55,119,0,141,181,233,184,0,105,216,169,106,0,141,181] - 199
zigbee2mqtt    | 2020-05-01T21:02:57.209Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtRtgRsp - {"srcaddr":0,"status":0,"routingtableentries":7,"startindex":0,"routingtablelistcount":7,"routingtablelist":[{"destNwkAddr":45079,"routeStatus":"ACTIVE","nextHopNwkAddr":55401},{"destNwkAddr":46477,"routeStatus":"ACTIVE","nextHopNwkAddr":46477},{"destNwkAddr":55401,"routeStatus":"ACTIVE","nextHopNwkAddr":55401},{"destNwkAddr":63138,"routeStatus":"ACTIVE","nextHopNwkAddr":46477},{"destNwkAddr":30519,"routeStatus":"ACTIVE","nextHopNwkAddr":46477},{"destNwkAddr":47337,"routeStatus":"ACTIVE","nextHopNwkAddr":55401},{"destNwkAddr":27305,"routeStatus":"ACTIVE","nextHopNwkAddr":46477}]}
zigbee2mqtt    | 2020-05-01T21:02:57.209Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-05-01T21:02:57.215Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtLqiReq - {"dstaddr":46477,"startindex":0}
zigbee2mqtt    | 2020-05-01T21:02:57.216Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,37,49,141,181,0,47]
zigbee2mqtt    | 2020-05-01T21:02:57.228Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,49,0,85]
zigbee2mqtt    | 2020-05-01T21:02:57.229Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,49,0,85]
zigbee2mqtt    | 2020-05-01T21:02:57.230Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 49 - [0] - 85
zigbee2mqtt    | 2020-05-01T21:02:57.230Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtLqiReq - {"status":0}
zigbee2mqtt    | 2020-05-01T21:02:57.231Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-05-01T21:02:57.267Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,141,181,0,186]
zigbee2mqtt    | 2020-05-01T21:02:57.268Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,141,181,0,186]
zigbee2mqtt    | 2020-05-01T21:02:57.268Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [141,181,0] - 186
zigbee2mqtt    | 2020-05-01T21:02:57.269Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":46477,"relaycount":0,"relaylist":[]}
zigbee2mqtt    | 2020-05-01T21:02:57.270Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | 2020-05-01T21:02:57.293Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,72,69,177,141,181,0,3,0,3,221,221,221,221,221,221,221,221,242,91,233,11,0,75,18,0,0,0,36,1,1,26,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,37,1,1,42,221,221,221,221,221,221,221,221,228,146,191,15,0,111,13,0,23,176,18,1,1,36,52]
zigbee2mqtt    | 2020-05-01T21:02:57.294Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,72,69,177,141,181,0,3,0,3,221,221,221,221,221,221,221,221,242,91,233,11,0,75,18,0,0,0,36,1,1,26,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,37,1,1,42,221,221,221,221,221,221,221,221,228,146,191,15,0,111,13,0,23,176,18,1,1,36,52]
zigbee2mqtt    | 2020-05-01T21:02:57.294Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 72 - 2 - 5 - 177 - [141,181,0,3,0,3,221,221,221,221,221,221,221,221,242,91,233,11,0,75,18,0,0,0,36,1,1,26,221,221,221,221,221,221,221,221,47,99,9,10,170,62,176,124,105,216,37,1,1,42,221,221,221,221,221,221,221,221,228,146,191,15,0,111,13,0,23,176,18,1,1,36] - 52
zigbee2mqtt    | 2020-05-01T21:02:57.296Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtLqiRsp - {"srcaddr":46477,"status":0,"neighbortableentries":3,"startindex":0,"neighborlqilistcount":3,"neighborlqilist":[{"extPandId":"0xdddddddddddddddd","extAddr":"0x00124b000be95bf2","nwkAddr":0,"deviceType":0,"rxOnWhenIdle":1,"relationship":2,"permitJoin":1,"depth":1,"lqi":26},{"extPandId":"0xdddddddddddddddd","extAddr":"0x7cb03eaa0a09632f","nwkAddr":55401,"deviceType":1,"rxOnWhenIdle":1,"relationship":2,"permitJoin":1,"depth":1,"lqi":42},{"extPandId":"0xdddddddddddddddd","extAd [... cut ...]
zigbee2mqtt    |     at Znp.onUnpiParsed (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:76:48)
zigbee2mqtt    |     at Parser.emit (events.js:310:20)
zigbee2mqtt    |     at Parser.parseNext (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:47:26)'
zigbee2mqtt    | 2020-05-01T21:02:57.381Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt    | zigbee2mqtt:error 2020-05-01 23:03:07: Failed to execute routing table for 'plug_floor_2'
```
see: https://github.com/Koenkk/zigbee2mqtt/issues/3455

Koenkk, 2020-05-02:
It seems that it returns no routes and therefore parsing fails. I would like to confirm this via sniffing (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html), if that also shows no routes then I can implement a fix.

Can you make a sniff?

Kosta-Github, 2020-05-02:
Unfortunately, I don't have a sniffer or flashing equipment at home right now.

Koenkk, 2020-05-02:
I can ignore this case but that wont really help much (dont know if Im doing the correct thing). Anyway I dont expect much benefit from fixing this, after it the routing table will still be empty as the device returns an empty one.

closed 2020-07-10 by stale[bot]

zen2, 2020-08-06:
I got accuratelly the same issue with the same device.
Since I recently migrate from CC2531 to CC1352-P2 coordinator I have sniff the traffic with CC2531.

The corresponding pcap-ng file is for this part of zigbee2mqtt log:
```
info  2020-08-06 14:12:58: Starting network scan (includeRoutes 'true')
error 2020-08-06 14:13:15: Failed to execute routing table for 'plug.aquarium_light'
info  2020-08-06 14:13:27: Network scan finished
info  2020-08-06 14:13:27: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G
{
        ***  full graph here but no copy/paste because it's really long for 51 devices ***
}
[zb-network-scan_osram-route-table-error.pcapng.gz](https://github.com/Koenkk/zigbee-herdsman/files/5034728/zb-network-scan_osram-route-table-error.pcapng.gz)

```

sammyke007, 2020-09-02:
Today I received two Osram Smart+ plugs. Both work fine, but I get the same failed: routingtable error with these devices. Any idea? Running zigbee2mqtt v1.14.3

Kosta-Github, 2020-09-02:
To be honest, it looks like the OSRAM smart plugs are crappy; I replaced them all by [Innr SmartPlugs](https://smile.amazon.de/gp/product/B084B1VX3T) a couple of weeks/months ago and those are working flawlessly since then without a single hic-up...

sammyke007, 2020-09-02:
I understand, but the Osram ones costed me only 11 euros each :-D

Kosta-Github, 2020-09-02:
but you get what you pay for... really, I tried them for a couple of weeks and just nothing worked reliably: I spent so many hours re-pairing all my sensors again and again; since switching to `Innr` no more re-pairing sessions...

zen2, 2020-09-02:
> It seems that it returns no routes and therefore parsing fails. I would like to confirm this via sniffing (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html), if that also shows no routes then I can implement a fix.
> 
> Can you make a sniff?

I've done it on my side:  https://github.com/Koenkk/zigbee-herdsman/issues/204
If you need more, I will be happy to help debug.

docbobo, 2020-10-19:
Anything new on this?

Koenkk, 2020-10-19:
@docbobo it's an issue in the firmwares of the plugs (see https://github.com/Koenkk/zigbee-herdsman/issues/204#issuecomment-687760554)

docbobo, 2020-10-19:
Thanks, @Koenkk