Koenkk/zigbee-herdsman #172: Feature Request: NRF52840-Dongle support

itavero, 2020-05-04:
I reckon this has been mentioned before, but I think it would be interesting to see if the NRF52840-Dongle from Nordic Semiconductors can be supported.
I noticed some work going on to support the deconz adapter, so I reckon some abstraction is already (being put) in place.

The dongle is cheap (about 11 euro from Mouser) and has a DFU option for programming it via USB (Nordic also provides a free cross-platform tool for uploading firmware).
It also has more RAM then the TI alternatives, I believe.

They appear to have a text based CLI available for using the chip as a Network Co-Processor.
Also see: https://infocenter.nordicsemi.com/topic/sdk_tz_v4.1.0/zigbee_architectures.html?cp=7_3_3_2_1_1#zigbee_architectures_designs_soc_designs

I'm currently on my smartphone, but I'll have another look at the nRF Thread and Zigbee SDK tomorrow to see if the binary is actually included.
I'll also have a look at the code of this component soon to get a better understanding on what adding support would actually mean and what is required for it to be feasible.

PS: I have a couple of these dongles (as well as the DK) at home already.

Koenkk, 2020-05-05:
To add a new adapter the following interface has to be implemented: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts. Implementations of deconz/zstack can be found here: https://github.com/Koenkk/zigbee-herdsman/tree/master/src/adapter

danergo, 2020-06-22:
@itavero: Have you started working on this already? :)

TI's multiprotocol ICs only support BLE peripherial features. Do you know by any chance if this Nordic can do BLE central while also playing ZB Coordinator role?

itavero, 2020-06-22:
@danergo: I did some preliminary investigation. The API from Nordics co-processor software seemed some what limited and for sure very different than the TI one. It will still be possible to use this as an adapter, however there seems to be more work than I anticipated/was hoping for.

With regards to the simultaneous use of BLE and Zigbee: I'm not sure, but I can imagine this is not possible, due to the difference in the physical layer. I'd suggest to check the docs from Nordic. They do support switching between them for sure.

closed 2020-07-30 by github-actions[bot]