Koenkk/zigbee-herdsman #179: Failed to pair devices: Xiaomi WXKG01LM / Xiaomi Cube MFKZQ01LM

YehorFedorov, 2020-05-19:
Hi!

I'm trying to implement zigbee bridge in my setup using your zigbee-hersdman module but I'm facing some issues.

My setup:
CC2530 (Zigbee 1.2 firmware) esp8266 (virtual serial port)
CC2530 (Zigbee 3.0.2 firmware) esp8266 (virtual serial port)
Zigbee bridge is running in Docker and connects to controller via TCP

Time to time I'm not able to pair Xiaomi WXKG01LM / Xiaomi Cube MFKZQ01LM devices. And here is logs:
```
##################### Xiaomi WXKG01LM

  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":15061,"extaddr":"0x00158d0002b88d5e","parentaddr":0} +12s
  zigbee-herdsman:controller:log Device '0x00158d0002b88d5e' joined +12s
  zigbee-herdsman:controller:log New device '0x00158d0002b88d5e' joined +0ms
  zigbee-herdsman:controller:log Creating device '0x00158d0002b88d5e' +0ms
  zigbee-herdsman:controller:log Interview '0x00158d0002b88d5e' start +1ms
  zigbee-herdsman:controller:device:log Interview - start device '0x00158d0002b88d5e' +12s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":15061,"nwkaddrofinterest":15061} +16s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,213,58,213,58,35] +16s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0] +555ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0] +264ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,254,27,68,129,0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> error Error: Invalid checksum
  zigbee-herdsman:adapter:zStack:unpi:parser     at Function.fromBuffer (/Users/efedorov/2smart/zigbee-bridge/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/frame.js:33:19)
  zigbee-herdsman:adapter:zStack:unpi:parser     at Parser.parseNext (/Users/efedorov/2smart/zigbee-bridge/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:45:51)
  zigbee-herdsman:adapter:zStack:unpi:parser     at Parser._transform (/Users/efedorov/2smart/zigbee-bridge/node_modules/zigbee-herdsman/dist/adapter/z-stack/unpi/parser.js:26:14)
  zigbee-herdsman:adapter:zStack:unpi:parser     at Parser.Transform._read (_stream_transform.js:190:10)
  zigbee-herdsman:adapter:zStack:unpi:parser     at Parser.Transform._write (_stream_transform.js:178:12)
  zigbee-herdsman:adapter:zStack:unpi:parser     at doWrite (_stream_writable.js:410:12)
  zigbee-herdsman:adapter:zStack:unpi:parser     at writeOrBuffer (_stream_writable.js:394:5)
  zigbee-herdsman:adapter:zStack:unpi:parser     at Parser.Writable.write (_stream_writable.js:294:11)
  zigbee-herdsman:adapter:zStack:unpi:parser     at Socket.ondata (_stream_readable.js:666:20)
  zigbee-herdsman:adapter:zStack:unpi:parser     at Socket.emit (events.js:182:13)
  zigbee-herdsman:adapter:zStack:unpi:parser     at addChunk (_stream_readable.js:283:12)
  zigbee-herdsman:adapter:zStack:unpi:parser     at readableAddChunk (_stream_readable.js:264:11)
  zigbee-herdsman:adapter:zStack:unpi:parser     at Socket.Readable.push (_stream_readable.js:219:10)
  zigbee-herdsman:adapter:zStack:unpi:parser     at TCP.onStreamRead [as onread] (internal/stream_base_commons.js:94:17) +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [27,68,129,0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,6,0,213,58,1,1,0,117,0,44,225,5,0,0,7,24,3,10,0,0,16,1,213,58,29,127] +45ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [27,68,129,0,0,6,0,213,58,1,1,0,117,0,44,225,5,0,0,7,24,3,10,0,0,16,1,213,58,29,127] +0ms
  zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '15061' failed (Error: SRSP - ZDO - nodeDescReq after 6000ms), retry +2m
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":15061,"options":0,"radius":30} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,213,58,0,30,149] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,69,0,33] +5s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [27,68,129,0,0,6,0,213,58,1,1,0,117,0,44,225,5,0,0,7,24,3,10,0,0,16,1,213,58,29,127,254,1,101,69,0,33] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0} +22s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":15061,"nwkaddrofinterest":15061} +3s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,213,58,213,58,35] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +2ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d0002b88d5e', attempt 1 +19s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":15061,"nwkaddrofinterest":15061} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,213,58,213,58,35] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:adapter Node descriptor request for '15061' failed (Error: AREQ - ZDO - nodeDescRsp after 10000ms), retry +24s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extRouteDisc - {"dstAddr":15061,"options":0,"radius":30} +10s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,69,213,58,0,30,149] +10s
[... 166 lines cut ...]
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,114,192,114,192,35] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [102] +11ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00158d00029aaa46', attempt 6 +24s
  zigbee-herdsman:controller:device:log Interview - failed for device '0x00158d00029aaa46' with error 'Error: Interview failed because can not get node descriptor ('0x00158d00029aaa46')
  zigbee-herdsman:controller:device:log     at Device.<anonymous> (/Users/efedorov/2smart/zigbee-bridge/node_modules/zigbee-herdsman/dist/controller/model/device.js:330:23)
  zigbee-herdsman:controller:device:log     at Generator.throw (<anonymous>)
  zigbee-herdsman:controller:device:log     at rejected (/Users/efedorov/2smart/zigbee-bridge/node_modules/zigbee-herdsman/dist/controller/model/device.js:6:65)' +0ms
  zigbee-herdsman:controller:error Interview failed for '0x00158d00029aaa46 with error 'Error: Interview failed because can not get node descriptor ('0x00158d00029aaa46')' +3m
```

Interesting fact, that Xiaomi button (WXKG01LM) is pairing time to time, but it happens in 2 out of 10 times.
I also have Xiaomi ZNLDP12LM and Xiaomi WSDCGQ11LM and I don't have any problems with this devices.

When I'm pairing devices I remove all entries in database.db by ieeeAddr of devices that I want to pair.

2 weeks earlier I was using zigbee-herdsman 0.12.58 and I didn't have any problems with pairing. I can also check downgrade and share results with you but maybe you have any thoughts?

Thanks!
[file.log](https://github.com/Koenkk/zigbee-herdsman/files/4652835/pairing-bug.log)

Koenkk, 2020-05-20:
`error Error: Invalid checksum` -> this error can occur when the remote connection to the adapter is not stable.

Does this still occur when the CC2530 is connected directly? (through USB)

YehorFedorov, 2020-05-20:
No i haven't tried it yet. I could try but I need to use it in this setup to be able to connect via TCP.
But thanks for explanation)

Any thoughts about Xiaomi cube? I'll try in few hours with old version of zigbee-herdsman

YehorFedorov, 2020-05-20:
Okey I think I should close this issue because my Xiaomi cube is broken I guess. Now I don't getting any data from it after any action even clicking reset button.

Thank you again!

closed 2020-05-20 by YehorFedorov