Koenkk/zigbee-herdsman #180: Strange behaviour when all devices stop communicate

YehorFedorov, 2020-05-20:
Hi!

I'm trying to find a reason when all paired devices goes offline in my system and here is my thoughts.

I have 5 setups of my system with the same configuration:
CC2530 (Zigbee 1.2 firmware) esp8266 (virtual serial port)
Zigbee bridge is running in Docker and connects to controller via TCP

Devices: Xiaomi wireless switch WXKG01LM, Xiaomi Cube MFKZQ01LM, Xiaomi bulb ZNLDP12LM, Xiaomi Aqara temperature, humidity and pressure sensor WSDCGQ11LM, Xiaomi Mi power plug ZNCZ02LM

On each setup I have different variations of zigbee network:
coordinator + 5 end devices in one room
coordinator + 4 routers + 5 end devices in 4 rooms
coordinator + 2 routers + 2 end devices in one room

In all of variations I'm struggling from problem when all devices suddenly stop communication.
I think that problem is in CC2530 and it's firmware so yesterday I flashed one with a Zigbee 3.0.2 firmware for testing.

Logs:
```
2020-05-16T23:15:58.949Z zigbee-herdsman:controller:log Permit joining\n
2020-05-16T23:15:58.950Z zigbee-herdsman:adapter:zStack:znp:SREQ --\u003e ZDO - mgmtPermitJoinReq - {\"addrmode\":15,\"dstaddr\":65532,\"duration\":254,\"tcsignificance\":0}\n
2020-05-16T23:15:58.950Z zigbee-herdsman:adapter:zStack:unpi:writer --\u003e frame [254,5,37,54,15,252,255,254,0,228]\n
2020-05-16T23:15:59.099Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [254,1,101,54,0,82]\n
2020-05-16T23:15:59.099Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]\n
2020-05-16T23:15:59.100Z zigbee-herdsman:adapter:zStack:unpi:parser --\u003e parsed 1 - 3 - 5 - 54 - [0] - 82\n
2020-05-16T23:15:59.100Z zigbee-herdsman:adapter:zStack:znp:SRSP \u003c-- ZDO - mgmtPermitJoinReq - {\"status\":0}\n
2020-05-16T23:15:59.100Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []\n
2020-05-16T23:15:59.102Z zigbee-herdsman:adapter:zStack:znp:SREQ --\u003e AF - dataRequestExt - {\"dstaddrmode\":2,\"dstaddr\":\"0x000000000000fffd\",\"destendpoint\":242,\"dstpanid\":0,\"srcendpoint\":242,\"clusterid\":33,\"transid\":36,\"options\":0,\"radius\":30,\"len\":6,\"data\":{\"type\":\"Buffer\",\"data\":[25,17,2,11,254,0]}}\n
2020-05-16T23:15:59.102Z zigbee-herdsman:adapter:zStack:unpi:writer --\u003e frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,36,0,30,6,0,25,17,2,11,254,0,222]\n
2020-05-16T23:15:59.113Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [254,3,69,182,0,0,0,240]\n
2020-05-16T23:15:59.114Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]\n
2020-05-16T23:15:59.114Z zigbee-herdsman:adapter:zStack:unpi:parser --\u003e parsed 3 - 2 - 5 - 182 - [0,0,0] - 240\n
2020-05-16T23:15:59.114Z zigbee-herdsman:adapter:zStack:znp:AREQ \u003c-- ZDO - mgmtPermitJoinRsp - {\"srcaddr\":0,\"status\":0}\n
2020-05-16T23:15:59.115Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []\n
2020-05-16T23:15:59.199Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [254,1,100,2,0,103]\n
2020-05-16T23:15:59.199Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]\n
2020-05-16T23:15:59.199Z zigbee-herdsman:adapter:zStack:unpi:parser --\u003e parsed 1 - 3 - 4 - 2 - [0] - 103\n
2020-05-16T23:15:59.200Z zigbee-herdsman:adapter:zStack:znp:SRSP \u003c-- AF - dataRequestExt - {\"status\":0}\n
2020-05-16T23:15:59.200Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []\n
2020-05-16T23:15:59.213Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [254,3,68,128,0,242,36,17]\n
2020-05-16T23:15:59.213Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,36,17]\n
2020-05-16T23:15:59.214Z zigbee-herdsman:adapter:zStack:unpi:parser --\u003e parsed 3 - 2 - 4 - 128 - [0,242,36] - 17\n
2020-05-16T23:15:59.214Z zigbee-herdsman:adapter:zStack:znp:AREQ \u003c-- AF - dataConfirm - {\"status\":0,\"endpoint\":242,\"transid\":36}\n
2020-05-16T23:15:59.214Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []\n
2020-05-16T23:16:47.340Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [254,3,69,196,144,177]\n
2020-05-16T23:16:47.340Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,144,177]\n
2020-05-16T23:16:47.351Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [0,163]\n
2020-05-16T23:16:47.352Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,144,177,0,163]\n
2020-05-16T23:16:47.352Z zigbee-herdsman:adapter:zStack:unpi:parser --\u003e parsed 3 - 2 - 5 - 196 - [144,177,0] - 163\n
2020-05-16T23:16:47.353Z zigbee-herdsman:adapter:zStack:znp:AREQ \u003c-- ZDO - srcRtgInd - {\"dstaddr\":45456,\"relaycount\":0,\"relaylist\":[]}\n
2020-05-16T23:16:47.353Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []\n
2020-05-16T23:16:47.381Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [254,68,68,129,0,0,0]\n
2020-05-16T23:16:47.381Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,68,68,129,0,0,0]\n
2020-05-16T23:16:47.391Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [0,144,177,1,1,0,63,0,7,96,159,0,0,48,28,95,17,92,10,1,255,66,39,3,40,38,5,33,52,0,7,39,0,0,0,0,0,0,0,0,8,33,23,1,9,33,1,2,10,33,0,0,100,32,0,101,32,255,102,33,153,0,144,177,29,214]\n
2020-05-16T23:16:47.392Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,68,68,129,0,0,0,0,144,177,1,1,0,63,0,7,96,159,0,0,48,28,95,17,92,10,1,255,66,39,3,40,38,5,33,52,0,7,39,0,0,0,0,0,0,0,0,8,33,23,1,9,33,1,2,10,33,0,0,100,32,0,101,32,255,102,33,153,0,144,177,29,214]\n
2020-05-16T23:16:47.392Z zigbee-herdsman:adapter:zStack:unpi:parser --\u003e parsed 68 - 2 - 4 - 129 - [0,0,0,0,144,177,1,1,0,63,0,7,96,159,0,0,48,28,95,17,92,10,1,255,66,39,3,40,38,5,33,52,0,7,39,0,0,0,0,0,0,0,0,8,33,23,1,9,33,1,2,10,33,0,0,100,32,0,101,32,255,102,33,153,0,144,177,29] - 214\n
2020-05-16T23:16:47.393Z zigbee-herdsman:adapter:zStack:znp:AREQ \u003c-- AF - incomingMsg - {\"groupid\":0,\"clusterid\":0,\"srcaddr\":45456,\"srcendpoint\":1,\"dstendpoint\":1,\"wasbroadcast\":0,\"linkquality\":63,\"securityuse\":0,\"timestamp\":10444807,\"transseqnumber\":0,\"len\":48,\"data\":{\"type\":\"Buffer\",\"data\":[28,95,17,92,10,1,255,66,39,3,40,38,5,33,52,0,7,39,0,0,0,0,0,0,0,0,8,33,23,1,9,33,1,2,10,33,0,0,100,32,0,101,32,255,102,33,153,0]}}\n
2020-05-16T23:16:47.395Z zigbee-herdsman:controller:log Received 'zcl' data '{\"frame\":{\"Header\":{\"frameControl\":{\"frameType\":0,\"manufacturerSpecific\":true,\"direction\":1,\"disableDefaultResponse\":true},\"transactionSequenceNumber\":92,\"manufacturerCode\":4447,\"commandIdentifier\":10},\"Payload\":[{\"attrId\":65281,\"dataType\":66,\"attrData\":{\"3\":38,\"5\":52,\"7\":\"0x0000000000000000\",\"8\":279,\"9\":513,\"10\":0,\"100\":0,\"101\":255,\"102\":153}}]},\"address\":45456,\"endpoint\":1,\"linkquality\":63,\"groupID\":0}'\n
2020-05-16T23:16:47.396Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []\n
2020-05-16T23:19:18.951Z zigbee-herdsman:controller:log Permit joining\n
2020-05-16T23:19:18.952Z zigbee-herdsman:adapter:zStack:znp:SREQ --\u003e ZDO - mgmtPermitJoinReq - {\"addrmode\":15,\"dstaddr\":65532,\"duration\":254,\"tcsignificance\":0}\n
2020-05-16T23:19:18.952Z zigbee-herdsman:adapter:zStack:unpi:writer --\u003e frame [254,5,37,54,15,252,255,254,0,228]\n
(node:17) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms\n
    at Timeout._onTimeout (/node_modules/zigbee-herdsman/dist/utils/waitress.js:47:35)\n
    at listOnTimeout (internal/timers.js:549:17)\n
    at processTimers (internal/timers.js:492:7)\n
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)\n
(node:17) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.\n
2020-05-16T23:19:40.519Z zigbee-herdsman:adapter:zStack:unpi:parser \u003c-- [0]\n
[... 62 lines cut ...]
2020-05-16T23:25:59.407Z zigbee-herdsman:adapter:zStack:znp:AREQ \u003c-- ZDO - mgmtPermitJoinRsp - {\"srcaddr\":0,\"status\":0}\n
2020-05-16T23:25:59.407Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []\n
[... cut ...]
1st error is related to permit join and I think this is a first problem that caused by some problems with controller.
2d error is more interesting. Please take a look on `dataRequestExt` req/res before permit join error and after. Could you explain please why before error I receive `dataRequestExt: { status: 0 }` but after `dataRequestExt: { status: 2 }`?

After this all devices stop communicate. Any thoughts?

Thanks!
[offline.log](https://github.com/Koenkk/zigbee-herdsman/files/4656210/offline.log)

z9th3, 2020-06-04:
I am having same issue on a fresh setup with latest zigbee2mqtt firmware on CC2531.

- Xiaomi wireless switch WXKG01LM
- Xiaomi Temp, Humid, Pressure sensor WSDCGQ11LM
- Xiaomi Movement and Illuminance sensor RTCGQ11LM
- Xiaomi Contact sensor MCCGQ11LM
- 4x IKEA TRADFRI LED bulb GU10 400 lm LED1650R5
- IKEA FLOALT LED light panel L1528
- IKEA TRADFRI remote control E1524
- Innr E27 bulb RB265

Attached the logs and zigbee2mqtt config file. If any other info can be used to troubleshoot this problem please let me know.

[Openhabian_20200602044155.log](https://github.com/Koenkk/zigbee-herdsman/files/4729380/Openhabian_20200602044155.log)

[configuration.yaml.txt](https://github.com/Koenkk/zigbee-herdsman/files/4729396/configuration.yaml.txt)

Koenkk, 2020-06-06:
The adapter crashed. Try using the source routing firmware which has increased stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

z9th3, 2020-06-06:
Thanks, just flashed the source routing firmware and will report back with results.

YehorFedorov, 2020-06-06:
Thanks @Koenkk! I'll try in few days

YehorFedorov, 2020-06-19:
I had a problem with pairing after flashing with source routing firmware but it's another issue I guess.
Re-flashed with default fw(https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default) and everything works fine now

closed 2020-06-19 by YehorFedorov