Koenkk/zigbee-herdsman #187: Senic switch support

kuersche1337, 2020-06-09:
Hello, 

I'm working with iobroker and try to pair the senic smart switch. 

Unfortunately, whenever I press a button I see following in the log: 

```
2020-06-09 18:45:25.019 - error: zigbee.0 (17111) uncaught exception: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received undefined
2020-06-09 18:45:25.021 - error: zigbee.0 (17111) TypeError [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received undefined
at Cipheriv.update (internal/crypto/cipher.js:155:11)
at GreenPower.encryptSecurityKey (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/greenPower.js:50:34)
at GreenPower. (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/greenPower.js:57:34)
at Generator.next ()
at /opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/greenPower.js:8:71
at new Promise ()
at __awaiter (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/greenPower.js:4:12)
at GreenPower.onZclGreenPowerData (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/greenPower.js:54:16)
at Controller. (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/controller.js:405:37)
at Generator.next ()

```

It looks like something with the zigbee-herdsman, so I'm opening this issues here. 
Or is it missing implementation in iobroker.zigbee?

Best regards and thanks
Alex

Koenkk, 2020-06-09:
Can you provide the complete debug logging. It can be achieved by setting the environment variable.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

kuersche1337, 2020-06-09:
Thank you for your fast response!

I tried to set the debug logging, but had no success. 
As I said, I have an iobroker running with iobroker.zigbee, which is using the zigbee-herdsman. Or so I think. 

I had a look at the linked website and tried to run the command 
`DEBUG=zigbee-herdsman* npm start`
in one of my folders in iobroker on my raspberry. As it seems, I didn't find the correct folder, cause I always got various error messages. 

So either you've got a more detailed or specific documentation for me or you've got to work with what I provided so far. 

I'm very willing to help, if you tell me what to do in more detail.

kuersche1337, 2020-06-12:
fyi, I think this post describes a similiar (or even the same?) error as I have. But He never answered your question.
https://github.com/Koenkk/zigbee-herdsman/issues/22#issuecomment-622993649

t-o-o-m, 2020-06-20:
Same error for me. 

Starting with `DEBUG=zigbee-herdsman* npm start` exposes the following log entries:

```zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,42,68,129,0,0,33,0] +6s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [85,195,242,242,1,48,0,116,42,102,0,0,22,17,83,4,32,11,85,195,113,1,98,0] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0,85,195,242,242,1,48,0,116,42,102,0,0,22,17,83,4,32,11,85,195,113,1,98,0] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,19,0,223,111,215,100] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0,85,195,242,242,1,48,0,116,42,102,0,0,22,17,83,4,32,11,85,195,113,1,98,0,0,0,19,0,223,111,215,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [157,227,104,223,111,29,5] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0,85,195,242,242,1,48,0,116,42,102,0,0,22,17,83,4,32,11,85,195,113,1,98,0,0,0,19,0,223,111,215,100,157,227,104,223,111,29,5] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 42 - 2 - 4 - 129 - [0,0,33,0,85,195,242,242,1,48,0,116,42,102,0,0,22,17,83,4,32,11,85,195,113,1,98,0,0,0,19,0,223,111,215,100,157,227,104,223,111,29] - 5 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":50005,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":48,"securityuse":0,"timestamp":6695540,"transseqnumber":0,"len":22,"data":{"type":"Buffer","data":[17,83,4,32,11,85,195,113,1,98,0,0,0,19,0,223,111,215,100,157,227,104]}} +6s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":83,"manufacturerCode":null,"commandIdentifier":4},"Payload":{"options":2848,"srcID":24232789,"frameCounter":98,"commandID":19,"payloadSize":0,"commandFrame":{"raw":{"type":"Buffer","data":[223,111,215,100,157,227,104]}}}},"address":50005,"endpoint":242,"linkquality":48,"groupID":0}' +6s
  zigbee-herdsman:controller:log 'zcl' data is from unknown device with address '50005', skipping... +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +18ms
(node:1946) UnhandledPromiseRejectionWarning: TypeError [ERR_INVALID_ARG_TYPE]: The "data" argument must be of type string or an instance of Buffer, TypedArray, or DataView. Received undefined
    at Cipheriv.update (internal/crypto/cipher.js:155:11)
    at GreenPower.encryptSecurityKey (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/greenPower.js:50:34)
    at GreenPower.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/greenPower.js:57:34)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/greenPower.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/greenPower.js:4:12)
    at GreenPower.onZclGreenPowerData (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/greenPower.js:54:16)
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:405:37)
    at Generator.next (<anonymous>)
(node:1946) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:1946) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,42,68,129,0,0,33,0] +195ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [85,195,242,242,1,48,0,201,95,102,0,0,22,17,84,4] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0,85,195,242,242,1,48,0,201,95,102,0,0,22,17,84,4] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [32,11,85,195,113,1,99,0,0,0,23,0,223,111,214,221] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0,85,195,242,242,1,48,0,201,95,102,0,0,22,17,84,4,32,11,85,195,113,1,99,0,0,0,23,0,223,111,214,221] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [131,28,230,223,111,29,24] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,42,68,129,0,0,33,0,85,195,242,242,1,48,0,201,95,102,0,0,22,17,84,4,32,11,85,195,113,1,99,0,0,0,23,0,223,111,214,221,131,28,230,223,111,29,24] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 42 - 2 - 4 - 129 - [0,0,33,0,85,195,242,242,1,48,0,201,95,102,0,0,22,17,84,4,32,11,85,195,113,1,99,0,0,0,23,0,223,111,214,221,131,28,230,223,111,29] - 24 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":33,"srcaddr":50005,"srcendpoint":242,"dstendpoint":242,"wasbroadcast":1,"linkquality":48,"securityuse":0,"timestamp":6709193,"transseqnumber":0,"len":22,"data":{"type":"Buffer","d [... cut ...]
    at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:405:37)
    at Generator.next (<anonymous>)
(node:1946) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)```

kuersche1337, 2020-06-20:
Tried this, bus as soon as I press the energy bar, the error appears and the adapter restarts. No matter how many and which button I press. Tried to factory reset, but don't know if it worked. 

@Koenkk could this problem be caused not by the herdsman but by the iobroker.zigbee? 
or with my firmware on my zigbee stick? 
Would reflashing help? (then I would have to repair all my devices -.-)

Koenkk, 2020-06-23:
@t-o-o-m  

> Upon releasing the device is recognised immediately.

What do you mean by this, do you see device paired in the log?

Can you provide me the complete log starting from the initial pairing? (with herdsman debug log like your first log)

kuersche1337, 2020-07-06:
Has nobody an idea here? 
At the moment I have an 70 € zigbee-adapter-restart-switch ;-)

So ANY help is appreciated -.-

kuersche1337, 2020-07-10:
I'm closing this, because I finally could pair the device. 
The solution for me was to pair the switch to a hue bridge first und then deleting it from there. 

Afterwards I could pair it with the zigbee adapter. I followed the advices here: https://www.zigbee2mqtt.io/devices/GreenPower_7.html
Now the support in the iobroker.zigbee adpater is missing, but this has nothing to do with this project.

Thanks for the help
kuersche

closed 2020-07-10 by kuersche1337

hb0nes, 2021-02-03:
> I'm closing this, because I finally could pair the device.
> The solution for me was to pair the switch to a hue bridge first und then deleting it from there.
>
> Afterwards I could pair it with the [...]

Hey man. I am pressing the buttons from the guide you linked but literally nothing is happening.
I removed the caps, hold the upper right contact and click the energy bar for 7 seconds.
I see nothing in the logs. What did you do to get it working? The Senic switch just seems like a broken piece of plastic now.