Koenkk/zigbee-herdsman #204: Failed to execute routing table for OSRAM Smart+ Plug

zen2, 2020-08-06:
Following the closed issue #https://github.com/Koenkk/zigbee-herdsman/issues/170

I got accuratelly the same issue with the same device.
Since I recently migrate from CC2531 to CC1352-P2 coordinator I have sniffed traffic with CC2531.

**zigbee2mqtt log:**

info  2020-08-06 14:12:58: Starting network scan (includeRoutes 'true')
error 2020-08-06 14:13:15: Failed to execute routing table for 'plug.aquarium_light'
info  2020-08-06 14:13:27: Network scan finished
info  2020-08-06 14:13:27: MQTT publish: topic 'zigbee2mqtt/bridge/networkmap/graphviz', payload 'digraph G
{
        ***  full graph here but no copy/pasted because it's really long for 51 devices ***
}

**sniffer pcpapng file:**

[zb-network-scan_osram-route-table-error.pcapng.gz](https://github.com/Koenkk/zigbee-herdsman/files/5036257/zb-network-scan_osram-route-table-error.pcapng.gz)

Koenkk, 2020-09-02:
Please provide the following:
- The network address of the plug (can be found in `data/database.db` `networkAddress`)
- The network key (you can share it privately to me on telegram @koenkk)

zen2, 2020-09-03:
I got this database.db:

`{"id":14,"type":"Router","ieeeAddr":"0x7cb03eaa00a75f11","nwkAddr":31560,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01","epList":[3],"endpoints":{"3":{"profId":49246,"epId":3,"devId":16,"inClusterList":[4096,0,3,4,5,6,2820,64527],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Plug 01","manufacturerName":"OSRAM","powerSource":1,"zclVersion":1,"appVersion":9,"stackVersion":2,"hwVersion":1,"dateCode":"20140331DEOS****","swBuildId":"V1.05.09"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca1af6c","endpointID":1}]}},"appVersion":9,"stackVersion":2,"hwVersion":1,"dateCode":"20140331DEOS****","swBuildId":"V1.05.09","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1599118380813}`

I've added you on telegram but it seems that I need to be in your contact to send message. So @Zentoo.

Koenkk, 2020-09-04:
Is this you? 
[image]

zen2, 2020-09-05:
> Is this you?
No ! at all ! I'm new on telegram so I've creat my t.me name only yesterday evening: @Zento0 (zero at end).

Koenkk, 2020-09-06:
I've checked your sniff and found the issue. The OSRAM plug sends a corrupted response: 

[image]

Not something that we can fix unfortunately (have to be fixed in the device firmware by OSRAM)

tbludau, 2020-09-07:
There are some solutions described in the net that an older firmware could fix the problem. Is there a way to downgrade the OSRAM Smart Plugs? (But i'm not really sure, if this will work)
I'm using V1.05.11 from the smart Gateway and think i have the same problems. (For a longer time)

zen2, 2020-09-08:
How could I know the firmware version of the device ?

tbludau, 2020-09-08:
@zen2 this is your firmware version: "swBuildId":"V1.05.09"

zen2, 2020-09-08:
Ho I figure out, I need to RTFM ^^
I got this firmware revision V1.04.12

zen2, 2020-09-08:
> @zen2 this is your firmware version: "swBuildId":"V1.05.09"

Actually I've two OSRAM plug installed that I bough as a pack of 4 and I got two different firmware version:
- V1.04.12
- V1.05.09
Anyway both of them present the symptom.

tbludau, 2020-09-08:
I've mailed yesterday night the ledvance support and mentioned this ticket about there firmware problem. If i will get an answer i will give some additional informations

closed 2020-10-16 by github-actions[bot]